ivmarkov/rust-esp32-std-demo #1: building `rust` component code fails with error: could not read vendor path

lucifer8851, 2020-12-04:
Steps to reproduce:
* Build the Rust Compiler (https://github.com/ivmarkov/rust)
* toolchain setup + set `XARGO_RUST_SRC` with ```export XARGO_RUST_SRC=`rustc --print sysroot`/lib/rustlib/src/rust/library```
* go to `rust-esp32-std-hello/rust` folder and execute `xargo build --release`

It fails with the following error
`error: could not read vendor path`

ivmarkov, 2020-12-04:
Can you create an empty `vendor` directory in the root of the `rust` folder where you checked out `https://github.com/ivmarkov/rust`?

Let me know the outcome.

AdityaHPatwardhan, 2020-12-05:
Thanks for the reply.
I tried that and the `could not read vendor path` error is gone.

but still it was not able to compile the `rust` component and failed with multiple errors like
```
error[E0432]: unresolved import `core::iter`
  --> /Users/lucifer/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:43:13
   |
43 |         use core::iter;
   |             ^^^^^^^^^^ no `iter` in the root

error[E0432]: unresolved import `core::ops`
  --> /Users/lucifer/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:45:13
   |
45 |         use core::ops;
   |             ^^^^^^^^^ no `ops` in the root

error[E0432]: unresolved import `core::option`
  --> /Users/lucifer/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:47:13
   |
47 |         use core::option;
   |             ^^^^^^^^^^^^ no `option` in the root

error[E0432]: unresolved import `core::ffi`
........
```

AdityaHPatwardhan, 2020-12-05:
Hey @ivmarkov Thanks for the suggestion, I was able to resolve the `could not read vendor path` error by creating a pseudo `vendor` directory
 But after resolving that the build still fails with a few unresolved imports.

```
error[E0432]: unresolved import `core::iter`
  --> /Users/flying_raijin/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:43:13
   |
43 |         use core::iter;
   |             ^^^^^^^^^^ no `iter` in the root

error[E0432]: unresolved import `core::ops`
  --> /Users/flying_raijin/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:45:13
   |
45 |         use core::ops;
   |             ^^^^^^^^^ no `ops` in the root

error[E0432]: unresolved import `core::option`
  --> /Users/flying_raijin/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:47:13
   |
47 |         use core::option;
   |             ^^^^^^^^^^^^ no `option` in the root

error[E0432]: unresolved import `core::ffi`
  --> /Users/flying_raijin/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:55:13
   |
55 |         use core::ffi;
   |             ^^^^^^^^^ no `ffi` in the root

error[E0432]: unresolved import `core::fmt`
  --> /Users/flying_raijin/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:57:13
   |
57 |         use core::fmt;
   |             ^^^^^^^^^ no `fmt` in the root

error[E0432]: unresolved import `core::hash`
  --> /Users/flying_raijin/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.80/src/lib.rs:59:13
```
Any suggestion on what I might do to resolve them ?
thanks.

ivmarkov, 2020-12-05:
That's strange. Looks as if the core crate that xargo compiles is broken.

In any case, I found an issue in `.cargo/config` I've fixed now.
Please try the following:
1) Pull latest  from the repo
2) `xargo clean`
3) `rm -rf ~/.cargo/registry/`
4) `rm -rf ~/.cargo/git/`
5) `xargo build --release`

Also: looks as if you are building on Mac? Haven't tried there, only Linux. In theory, there should be no difference, but then who knows. Let me try on Mac, but mind you - rebuilding the whole toolchain & all might take a while...

ivmarkov, 2020-12-05:
OK, so on a completely pristine Mac (only brew and Xcode installed):
- The rust xtensa+STD toolchain compiles just fine by following the instructions (only had to do `brew install cmake` and `brew install ninja` for the LLVM to build).
- The rust-esp32-std-hello app *also* compiles fine. 

Disclaimers:
- I've only checked with the latest `rust-esp32-std-hello` version (with the fixes to .cargo/config applied)
- Everything (rustup and then xargo) were installed prior to/after the toolchain build
- Obviously, i've started with completely empty `~/.cargo/registry` and `~./cargo/git` directories. So perhaps wiping those out as I suggested above might help

ivmarkov, 2020-12-05:
Huh. Kind of strange. On Mac it also did not even complain about the "vendor" directory. 
Even stranger that the stable compiler on Mac (the one I used for building the xtensa+STD toolchain) is the same as the one on Lunux:
`stable-x86_64-apple-darwin unchanged - rustc 1.48.0 (7eac88abb 2020-11-16)`

On Linux, it is:
`stable-x86_64-unknown-linux-gnu (default)
rustc 1.48.0 (7eac88abb 2020-11-16)`

ivmarkov, 2020-12-05:
More on the "vendor" topic. If you want to use regular cargo instead of xargo:
1. Uncomment the "unstable" lines in `.cargo/config` in the repo of this demo app
2. Apply the workaround to your rust clone of my repo, as described [here](https://github.com/rust-lang/cargo/issues/8945), at the bottom of the issue description.

AdityaHPatwardhan, 2020-12-06:
@ivmarkov Thanks for the fix, I was able to build the `rust-esp32-std-hello` example successfully. I was able to build that with `cargo` as well after applying your workaround.

lucifer8851, 2020-12-06:
Thank you @ivmarkov for providing the solution. It has fixed the issue for me. 
Closing down the issue.

closed 2020-12-06 by lucifer8851