ivmarkov/rust-esp32-std-demo #3: bootloop when running demo app on ESP32-S2-DevKitM-1 board

mykmelez, 2021-06-07:
When I try running this on my ESP32-S2-DevKitM-1 board, it bootloops:

```
I (493) cpu_start: Starting scheduler on PRO CPU.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008f994: 3c910a0c 130ca0c6 e9b202bd
Core  0 register dump:
PC      : 0x4008f999  PS      : 0x00060f30  A0      : 0x800876cb  A1      : 0x3ffd0378  
A2      : 0x00000001  A3      : 0x3f00189c  A4      : 0x00004e20  A5      : 0x383f8000  
A6      : 0x3ffd1a60  A7      : 0x00000000  A8      : 0x3f001864  A9      : 0x3ffc981c  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060b20  A13     : 0x0000000c  
A14     : 0x3ffd1a60  A15     : 0x3ffd1a68  SAR     : 0x00000017  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00060b20  LEND    : 0x0000000c  LCOUNT  : 0x40023c65  

Backtrace:0x4008f996:0x3ffd0378 |<-CORRUPTED

ELF file SHA256: 1d83d964a1181dba

Rebooting...
```

The board isn't listed in `pio boards`, but I've tried it with the three ESP32S2 boards on that list (featheresp32-s2, esp32-s2-kaluga-1, and esp32-s2-saola-1), and they all bootloop, seemingly in the same way.

ivmarkov, 2021-06-07:
First of all, thanks for trying and congrats for getting that far!

Boot-looping on an ESP32-S2 board is not to be unexpected, because the ESP-IDF bindings in my esp-idf-sys crate are currently *only* for ESP32 (non-S2 version). (And for ESP-IDF 4.1 for that matter.)

This (or more likely next) weekend, I'll update them to support ESP32-S2. And the latest ESP-IDF in PlatformIO. The process is a bit involved, as I first need to update my Rust fork with the latest changes of @mabezdev , as they introduce ESP32-S2 MCU support, as a new Rust target. Then I need to push to github all changes I have lying around, like a minimal support for embedded-hal on top of ESP-IDF.

On a positive note, if you happen to have an ESP32S2 with an LCD screen, we might be able to get it going given that I now have just-enough SPI support to drive an LCD screen with the embedded-graphics crate!

So, sorry for the troubles and I'll let you know once these changes are in.
 
In the meantime, if you have a pure ESP32 board, the current demo should work with it.

mykmelez, 2021-06-08:
Thanks for the quick response! Looking forward to trying this out again with the ESP32-S2 once you've updated the bindings. In the meantime, I've ordered an ESP32-DevKitC-32E, which should arrive tomorrow, and I'll try it out on that board once it arrives!

Rosabis, 2021-06-15:
I would like to ask some digressions. Can you elaborate on the environment required to compile PlugSorbet? I encountered many errors when compiling this project.

ivmarkov, 2021-06-17:
@mykmelez Small update: support for Rust STD on ESP32S2 might have to wait a bit due to lack of atomics support on the ESP32S2 (and ESP32C3) chip. The issue is being worked on however. More info here:
https://github.com/espressif/rust-esp32-example/issues/3

mykmelez, 2021-06-30:
@Rosabis, were you asking me that question, and was it a question about PluotSorbet? If so, I haven't compiled it in years, but as I recall, you need at least the TypeScript compiler, emscripten, and a Java JRE to compile it.

ivmarkov, 2021-06-30:
@mykmelez @Rosabis You are commenting in the wrong repo, and in the wrong issue.

ivmarkov, 2021-06-30:
@mykmelez But thanks for clarifying the situation anyway! :)

ivmarkov, 2021-07-12:
@mykmelez :

Update:

There is a very, very, very preliminary support for ESP32-S2 and ESP32-C3 (the riscv chip) with STD.

To test it, you'll need to pull the latest rust-esp32-std-hello demo, but most importantly, you'll have to compile from source the new Rust STD compiler fork, which is in branch [stable_V1.53.0](https://github.com/ivmarkov/rust/tree/stable_V1.53.0) by following the instructions in the README.md file of that branch.

A few notes:
- I have not even tested running the demo on ESP32-S2, because I forgot my S2 board - sorry about that (the ESP32-C3 does work - tested). I'll try it on the S2 tomorrow.
- You might not need the new compiler fork for S2 specifically, but you do need it for C3, as the new fork uses a patched libc which "knows" about the riscv32 architecture on newlib. In any case I have not tried building for S2 with the old compiler fork.
- The support is "preliminary" because the whole approach might change, as described [here](https://github.com/espressif/rust-esp32-example/issues/3).

Let me know if you decide to give it a try and whether it does / does not work for you!

ivmarkov, 2021-07-13:
Confirmed: the demo crate is working on S2.

ivmarkov, 2021-07-13:
By the way I can even draw on the kaluga-1 LED screen using a (slightly modified) version of the LED code in this demo crate. However, the colors seem off. (Might be using the wrong display driver altogether, as depending on your exact kaluga-1 board, it is either ili9341v or st7789 - I'm using st7789 but have to try with ili9341.) Once I fix that, I'll push my changes to the LED code so that the demo works with kaluga-1 too.

ivmarkov, 2021-07-15:
I'll close this. Let's open separate bug reports for specific issues with ESP32-S2 when these are found.

closed 2021-07-15 by ivmarkov