ivmarkov/rust-esp32-std-demo #6: esp-idf-hal: cannot find type `UART2` in module `serial`

wassup-, 2021-07-27:
I know this is the wrong repository to post this issue, but the esp-idf-hal repository does not have an Issues section.
When compiling for ESP32-C3 (`target = "riscv32imac-esp-espidf"`), I get the following error:

```➜  rust-esp32-std-hello git:(main) ✗ cargo build --release
   Compiling esp-idf-hal v0.16.2
error[E0433]: failed to resolve: could not find `UART2` in `serial`
  --> /Users/tomk/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.16.2/src/peripherals.rs:61:28
   |
61 |             uart2: serial::UART2::new(),
   |                            ^^^^^ could not find `UART2` in `serial`

error[E0412]: cannot find type `UART2` in module `serial`
   --> /Users/tomk/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.16.2/src/peripherals.rs:24:24
    |
24  |     pub uart2: serial::UART2,
    |                        ^^^^^ help: a struct with a similar name exists: `UART0`
    | 
   ::: /Users/tomk/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.16.2/src/serial.rs:588:9
    |
588 |         pub struct $uart;
    |         ----------------- similarly named struct `UART0` defined here

Some errors have detailed explanations: E0412, E0433.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `esp-idf-hal` due to 2 previous errors
```

I fixed this by replacing `#[cfg(any(esp32, not(feature = "ulp")))]` with `#[cfg(all(not(esp32c3), not(feature = "ulp")))]` on line 23 and line 60 (in peripherals.rs). However I'm not sure if this is the _right_ fix.

ivmarkov, 2021-07-27:
Hi there. I noticed this one as well. It should be fixed now:
* Please pull latest of `rust-esp32-std-hello`
* `cargo update`
* `cargo build`

If it still does not work after that, please let me know.

ivmarkov, 2021-07-27:
(If that's not clear, I've republished newer versions of a bunch of crates, including `esp-idf-hal`. The pull + cargo updqate should bring these new versions.)

ivmarkov, 2021-07-27:
I've asked the Issues tab to be opened for `esp-idf-hal` and a few others, so that should be resolved soon too:
https://github.com/esp-rs/rust/issues/59

wassup-, 2021-07-27:
Thanks for the quick response! That does indeed solve the issue.

closed 2021-07-27 by wassup-