ivmarkov/rust-esp32-std-demo #7: App doesn't run after flashing to NodeMCU ESP32-C3M-Kit

wassup-, 2021-07-27:
After flashing the example app to my NodeMCU ESP32-C3M-Kit, I get the following output:
```
$ cargo espmonitor --release --speed 115200 /dev/cu.usbserial-1410
Opening /dev/cu.usbserial-1410 with speed 115200
WARNING: Flash image /Users/tomk/Projects/Git/3rdparty/rust-esp32-std-hello/target/xtensa-esp32-none-elf/release/rust-esp32-std-hello does not exist (you may need to build it)
Resetting device... done
ESP-ROM:e�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x1758
load:0x403ce000,len:0x894
load:0x403d0000,len:0x2c5c
entry 0x403ce000
I (53) boot: ESP-IDF v4.3-beta3 2nd stage bootloader
I (53) boot: compile time 18:22:41
I (53) boot: chip revision: 3
I (56) boot.esp32c3: SPI Speed      : 40MHz
I (60) boot.esp32c3: SPI Mode       : DIO
I (65) boot.esp32c3: SPI Flash Size : 4MB
I (69) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (101) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (108) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (116) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (123) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (131) boot: End of partition table
E (135) boot: ota data partition invalid and no factory, will try all partitions
I (143) esp_image: segment 0: paddr=00100020 vaddr=3c0e0020 size=22da0h (142752) map
I (183) esp_image: segment 1: paddr=00122dc8 vaddr=3fc8e600 size=02ea4h ( 11940) load
I (186) esp_image: segment 2: paddr=00125c74 vaddr=40380000 size=0a3a4h ( 41892) load
I (198) esp_image: segment 3: paddr=00130020 vaddr=42000020 size=d7b18h (883480) map
I (389) esp_image: segment 4: paddr=00207b40 vaddr=4038a3a4 size=041d8h ( 16856) load
I (393) esp_image: segment 5: paddr=0020bd20 vaddr=50000000 size=00004h (     4) load
I (395) esp_image: segment 6: paddr=0020bd2c vaddr=50000008 size=00010h (    16) load
I (408) boot: Loaded app from partition at offset 0x100000
I (409) boot: Disabling RNG early entropy source...

################################################
arch:ESP32C3, 3
compile_time:Ai-Thinker|B&T
ble_mac:7cDFA1a444C1
wifi_mac:7cDFA1a444C0
sdk_version:v4.3-beta3
firmware_version:2.1.0(MINI-1)
compile_time:May 18 2021 18:23:12

ready

################################################
factory_parameter partition missed
2.1.0
```

As you can see, even the simple_playground output (`Hello, world from Rust!`) is missing, which leads me to believe the app isn't loading / executing correctly.

I compiled both Rust and LLVM from scratch (as explained at https://github.com/esp-rs/rust).

I compiled and flashed the image using the following script:
```
#!/usr/bin/env sh

CHIP=esp32c3
TARGET=riscv32imac-esp-espidf

# build
cargo build --release

# create image
esptool.py --chip "$CHIP" \
    elf2image \
    "target/$TARGET/release/rust-esp32-std-hello"

# write image
esptool.py --chip "$CHIP" \
    -p /dev/cu.usbserial-1410 \
    -b 460800 \
    --before=default_reset \
    --after=hard_reset \
    write_flash \
    --flash_mode dio \
    --flash_freq 40m \
    --flash_size 4MB 0x10000 \
    "target/$TARGET/release/rust-esp32-std-hello.bin"
```

Is there something that I'm missing?

ivmarkov, 2021-07-27:
Nothing works because the logs say that they cannot even find the image you have flashed.  I think the reason for that is that the partition table that you have on that chip that the logs are listing is a non-standard:
```
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (101) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (108) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (116) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (123) boot:  5 ota_1            OTA app          00 11 00280000 00180000
```

So you are flashing at address `0x10000`. However in your partition table this is not a valid address for the binary.
You should instead be using the address of `ota_0` or `ota_1` which - for your case - are `0x100000` (notice the extra zero) and `0x280000`. Flashing to any of these two addresses (or both of them) should work.

Once you are beyond your first flash, I suggest you change the partition table to something more meaningful for your use cases (say, a single "factory style" app), but please try the suggestion from above first.

P.S. Here's how my (also non-standard) partition table of my esp32c3-devkitm-1 looks:
```
I (86) boot: ## Label            Usage          Type ST Offset   Length
I (93) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (101) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (108) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (116) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (123) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (131) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (138) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (146) boot:  7 coredump         Unknown data     01 03 00350000 00010000
```

Since I also flashed at the wrong address first, every time now I boot the chip, I'm greeted with:
```
E (158) boot: ota data partition invalid and no factory, will try all partitions
```

I couldn't care less. I'll fix my own partition table one of these days... :)

wassup-, 2021-07-27:
Once again, thank you for the quick response!

I have now flashed to `0x100000`, and now everything is working as expected. Your explanation makes total sense, it might be a good idea to put this in the README of the project?

Thank you for your great work!

ivmarkov, 2021-07-27:
Yes, this I'll do.

ivmarkov, 2021-07-28:
README.md updated.

closed 2021-07-28 by ivmarkov