ivmarkov/rust-esp32-std-demo #12: long running task

Kethku, 2021-09-10:
I'm trying to use an esp32 feather to communicate with other micro controllers using the dwm1000 for ultra wideband communication and ranging. To do so I'm using a hal crate for the dwm1000 which requires calling some functions on a loop. Generally things seem to be working except for the fact that the cpu seems to reset periodically with:
```
Task watchdog got triggered. The following tasks did not reset the watchdog in time
```
I couldn't find any details about this in the repo so I was wondering if there was an example for interacting with the watchdog so as to allow for long running tasks like what I'm trying to do

ivmarkov, 2021-09-10:
There is no example yet, but in principle, most of the ESP-IDF API should be available via the `esp-idf-sys` crate. So in theory whatever you can do from "C" to interact with the watchdog, you can also do from Rust (with unsafe calls).

If you figure it out how it works with C calls, we might want to create safe Rust API wrappers around the native unsafe ESP-IDF watchdog APIs. But that would be a next step.
 
*With that said* - are you busy-waiting in that loop? If so, the easiest fix might be to just not do this. You can use Rust's `thread::sleep` for example, and this way the watchdog will not be triggered anymore.

ivmarkov, 2021-09-16:
Closing - this is more of a help request than an actual bug. Feel free to reopen with clarifications if you think otherwise - thanks!

closed 2021-09-16 by ivmarkov