ivmarkov/rust-esp32-std-demo #13: Build issue

maelp, 2021-09-15:
I'm on osX, and I've followed your installation guide. I know my Rust extensa toolchain works because I've been able to compile https://github.com/espressif/rust-esp32-example and make it work

when running `cargo build` I get:

```
[...]
warning: cargo:rustc-link-arg requires -Zextra-link-arg flag
warning: cargo:rustc-link-arg requires -Zextra-link-arg flag
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
   Compiling rust-esp32-std-hello v0.17.4 (/Users/primet/work/gouach/embedded-nobackup/rust-esp32-std-hello)
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
error: linking with `ldproxy` failed: exit status: 101
[...]
  = note: Running ldproxy
          thread 'main' panicked at 'Cannot locate argument '--ldproxy-linker <linker>'', /Users/primet/.cargo/registry/src/github.com-1ecc6299db9ec823/ldproxy-0.1.3/src/main.rs:43:10
          stack backtrace:
             0: _rust_begin_unwind
             1: core::panicking::panic_fmt
             2: core::option::expect_failed
             3: ldproxy::main
          note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

ivmarkov, 2021-09-15:
`warning: cargo:rustc-link-arg requires -Zextra-link-arg flag`

That's not a good indication. Can you run run `cargo --version` and `rustc --version` and let me know the outcome?

(the `'+memctl' is not a recognized feature for this target (ignoring feature)` you can safely ignore)

maelp, 2021-09-15:
```
❯ cargo --version                                                    ±[●][main]
cargo 1.54.0-dev
❯ rustc --version                                                    ±[●][main]
rustc 1.54.0-dev
```

closed 2021-09-15 by ivmarkov