ivmarkov/rust-esp32-std-demo #14: Tiny "Hello world" does not run properly.

FrozenDroid, 2021-09-16:
When I comment out all of the code and just print something, the ESP32C3 board I'm using prints: 
```
E (174) esp_image: invalid segment length 0xffee
E (174) boot: Factory app partition is not bootable
E (174) boot: No bootable app partitions in the partition table
```
My partition table looks like:
```
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
```

I've been looking into it for a bit but it seems that the padding is not aligned to 4-byte, which is causing this error to pop up. I'm using `esptool.py` to generate an app image from the compiled ELF.

ivmarkov, 2021-09-16:
I do not quite understand what you are saying:
- Are you saying that the full demo works, but **stops** to work the moment you turn it into a "Hello world print" and then flash?
- Or are you saying that it doesn't work in principle and you commenting out everything, to see why it does not work?

Most importantly, the log from the board actually says that the app you flashed *does not even boot*.

So it would be nice to see how exactly you are flashing. As in your command line.

ivmarkov, 2021-09-16:
Also please provide the `elf2bin` command line. Basically your workflow from `cargo build` onwards.

ivmarkov, 2021-09-16:
I don't have any paddings inserted by the `esptool` when the .bin file is created:
```
esptool.py v3.1
Image version: 1
Entry point: 403802ca
6 segments

Segment 1: len 0x2a088 load 0x3c0c0020 file_offs 0x00000018 [DROM]
Segment 2: len 0x03510 load 0x3fc8e200 file_offs 0x0002a0a8 [DRAM,BYTE_ACCESSIBLE]
Segment 3: len 0x02a50 load 0x40380000 file_offs 0x0002d5c0 [IRAM]
Segment 4: len 0xb5508 load 0x42000020 file_offs 0x00030018 [IROM]
Segment 5: len 0x0b70c load 0x40382a50 file_offs 0x000e5528 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x000f0c3c [RTC_IRAM,RTC_DRAM]
Checksum: f5 (valid)
Validation Hash: 5ad3c59e4137c5e938beb30241cc46f2192c9ae8dd775a05ec5c37d8055f1cda (valid)
```

Why are you using a dev version of `esptool`? Yours says `esptool.py v3.1-dev` (whatever `dev` means), mine says simply `esptool.py v3.1`.

Also `-b 921600` seems a bit high, but perhaps that's not the main issue.

ivmarkov, 2021-09-16:
*Sorry*, now *I* messed up and did not read your explanations about the original example, etc. Let me read that...

FrozenDroid, 2021-09-16:
I gave both the latest dev version and 3.1 of esptool a try, but the resulting image is the same.

ivmarkov, 2021-09-16:
Ha! You are completely right!

I just reproduced it. Full log just in case:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d11a0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x17a8
load:0x403ce000,len:0x894
load:0x403d0000,len:0x2bf8
entry 0x403ce000
I (57) boot: ESP-IDF v4.3-beta2-2-g9a2d251912 2nd stage bootloader
I (57) boot: compile time 19:26:26
I (57) boot: chip revision: 3
I (60) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (68) boot.esp32c3: SPI Speed      : 80MHz
I (72) boot.esp32c3: SPI Mode       : DIO
I (77) boot.esp32c3: SPI Flash Size : 4MB
I (82) boot: Enabling RNG early entropy source...
I (87) boot: Partition Table:
I (91) boot: ## Label            Usage          Type ST Offset   Length
I (98) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (105) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (113) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (121) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (128) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (136) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (143) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (151) boot:  7 coredump         Unknown data     01 03 00350000 00010000
I (158) boot: End of partition table
E (163) boot: ota data partition invalid and no factory, will try all partitions
I (171) boot_comm: chip revision: 3, min. application chip revision: 0
I (178) esp_image: segment 0: paddr=00020020 vaddr=3c030020 size=11148h ( 69960) map
I (197) esp_image: segment 1: paddr=00031170 vaddr=3fc8ae00 size=01bf8h (  7160) load
I (199) esp_image: segment 2: paddr=00032d70 vaddr=40380000 size=0aca4h ( 44196) load
I (212) esp_image: segment 3: paddr=0003da1c vaddr=50000000 size=00010h (    16) load
I (212) esp_image: segment 4: paddr=0003da34 vaddr=00000000 size=025e4h (  9700) 
I (222) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=2a2e4h (172772) map
E (255) esp_image: invalid segment length 0xffee
E (256) boot: OTA app partition slot 0 is not bootable
E (256) esp_image: image at 0x1b0000 has invalid magic byte
W (262) esp_image: image at 0x1b0000 has invalid SPI mode 255
W (268) esp_image: image at 0x1b0000 has invalid SPI size 15
E (274) boot: OTA app partition slot 1 is not bootable
E (280) boot: No bootable app partitions in the partition table
``` 

Segment info, just in case:
```
esptool.py v3.1
Image version: 1
Entry point: 403802ca
8 segments

Segment 1: len 0x11148 load 0x3c030020 file_offs 0x00000018 [DROM]
Segment 2: len 0x01bf8 load 0x3fc8ae00 file_offs 0x00011168 [DRAM,BYTE_ACCESSIBLE]
Segment 3: len 0x0aca4 load 0x40380000 file_offs 0x00012d68 [IRAM]
Segment 4: len 0x00010 load 0x50000000 file_offs 0x0001da14 [RTC_IRAM,RTC_DRAM]
Segment 5: len 0x025e4 load 0x00000000 file_offs 0x0001da2c [PADDING]
Segment 6: len 0x2a2e4 load 0x42000020 file_offs 0x00020018 [IROM]
Segment 7: len 0x0ffee load 0x00000000 file_offs 0x0004a304 [PADDING]
Segment 8: len 0x009e4 load 0x4202a302 file_offs 0x0005a2fa [IROM]
Checksum: b4 (valid)
Validation Hash: 5ced93819a4a522162f07e7a2d227dfe91df0998b8c80ad89411c73ced59a5e2 (valid)
```

FrozenDroid, 2021-09-16:
Awesome!
Yeah, I'm really not sure what is causing this but I feel like it might be a bug in esptool. All segments need to be divisible by 4, but this padding segment clearly isn't. 

https://github.com/espressif/esptool/blob/master/esptool.py#L2538

Would you agree? Maybe we should open an issue there as well.

ivmarkov, 2021-09-16:
I agree - if you can open a bug at their repo that would be great.

Additional info:
- Tried with an older compiler (Rust 1.54), and the problem is still there. Different padding section, but again not divisible by 4
- Tried generating an image for the ESP32 (same sourcecode). **No** padding section there

That might be esp32c3 related somehow. Otherwise, I would imagine people would've complained long ago...

igrr, 2021-09-21:
@ivmarkov @FrozenDroid May I ask you to attach an ELF file which triggers this issue over at https://github.com/espressif/esptool/issues/661? This would help the folks working on esptool pick up the issue report quicker, since they won't have to set up rust-esp32-std-hello to reproduce. Thanks in advance!

radimkarnis, 2021-09-22:
^^^ I would appreciate that!

FrozenDroid, 2021-09-22:
> @ivmarkov @FrozenDroid May I ask you to attach an ELF file which triggers this issue over at [espressif/esptool#661](https://github.com/espressif/esptool/issues/661)? This would help the folks working on esptool pick up the issue report quicker, since they won't have to set up rust-esp32-std-hello to reproduce. Thanks in advance!

Done! Check the esptool issue.

icewind1991, 2021-09-23:
I think the issue is not with esptool but with the generated elf/linker script

```
Segment 8: len 0x00980 load 0x4202a51a file_offs 0x0005a512 [IROM]
```

`0x4202a51a` is not aligned to 32bit so looks like an invalid address for the segment

ivmarkov, 2021-09-23:
The error message in my log above `E (255) esp_image: invalid segment length 0xffee`
... seems to be for segment 7, not for segment 8 (because segment 7 is having length 0xfee, not segment 8)?

Or are you saying that both segment 7 and segment 8 are problematic?
Btw, I don't see a message that segment 6 is loaded either...

icewind1991, 2021-09-23:
The padding (segment 7) is the wrong size because it's trying to pad to a non aligned address (segment 8)

AlyoshaVasilieva, 2022-01-07:
I'm also running into this error with the default PIO build system*. However, I can't get `native` building to work. The last log lines are

```
  Cloning into 'K:/Code/rust/esp32-std/.embuild/espressif/esp-idf-v4.3.1/components/nghttp/nghttp2/third-party/neverbleed'...
  From https://github.com/tatsuhiro-t/neverbleed
   * branch            b967ca054f48a36f82d8fcdd32e54ec5144f2751 -> FETCH_HEAD
  From https://github.com/espressif/tinyusb
   * branch              334e95fac52a607150157ae5199a19e11f843982 -> FETCH_HEAD
  WARNING: You are using pip version 20.3.1; however, version 21.3.1 is available.
  You should consider upgrading via the 'K:\Code\rust\esp32-std\.embuild\espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
  error: patch failed: components/xtensa/stdatomic.c:133
  error: components/xtensa/stdatomic.c: patch does not apply
  thread 'main' panicked at '
  command did not execute successfully, got: exit code: 1
```

I get this error about "patch does not apply" on both Windows 10 and WSL Ubuntu 20.04.

I attempted to use a local ESP-IDF v4.3.1 and that didn't work either, but I'm not sure I used the right combination of env vars. Is there something I'm doing wrong? [Log here](https://gist.githubusercontent.com/AlyoshaVasilieva/b41257e7d0a065d977b33eb19220349f/raw/86961e4684158e5bdd80291b09674a41001a9037/compile%2520log.txt), it seems to fail in `openeth.h`.

Compiling for an ESP32-C3 rev 3.

*: Without touching anything other than telling it to target C3, I can compile and it works. It's when I start experimenting that it breaks.

ivmarkov, 2022-01-10:
> I'm also running into this error with the default PIO build system*. However, I can't get `native` building to work. The last log lines are
>
> ```
>   Cloning into [...]

Forget about using the native build on Windows currently. See below.

> I attempted to use a local ESP-IDF v4.3.1 and that didn't work either, but I'm not sure I used the right combination of env vars. Is there something I'm doing wrong? [Log here]

This is not (yet) supported, but is work in progress (i.e. supplying your own **local** copy of ESP-IDF). You can still supply your own Git clone of ESP-IDF, but then see below - that's not necessary.

> *: Without touching anything other than telling it to target C3, I can compile and it works. It's when I start experimenting that it breaks.

Yes, the issue does not always happen.

To workaround the issue, you **have to** use a version > ESP-IDF 4.3.X. Using version > 4.3.X is not possible with the PIO build, hence why you **also** need to switch to the native build. This leaves you with two options:
* Use ESP-IDF master. Unfortunately this one cannot be built at the moment. We have not yet looked into this issue, might be an issue in ESP-IDF itself
* Use ESP-IDF release/v4.4 branch <-- This is what I would recommend

So to build, please use `export ESP_IDF_VERSION=release/v4.4; cargo build --features native`.

Unfortunately, the above **will not work on Windows** (i.e. on Windows currently there is simply no workaround for that problem), because the native build generates too long command lines on Windows. We are aware of that, and the fix should actually be implemented inside ESP-IDF itself (to be more precise, on how they build the ESP-IDF GCC toolchains for Windows). We are waiting for the next GCC Windows toolchains release from them.

dobairoland, 2022-01-10:
Hi @FrozenDroid, @fkohlgrueber and others who experience this issue with the ESP-IDF master branch.

Could you please try to build your application with and without https://github.com/espressif/esp-idf/pull/8219?

I've noticed that the ELF file posted here https://github.com/espressif/esptool/issues/661#issuecomment-924979468 has a weird `.rwtext` section. That is that evil "Segment 8" which is causing the unaligned PADDING before it. 

I could not found a pure ESP-IDF project which produces this and it probably should be in `.flash.text`.

ivmarkov, 2022-01-10:
> Hi @FrozenDroid, @fkohlgrueber and others who experience this issue with the ESP-IDF master branch.
> 
> Could you please try to build your application with and without [espressif/esp-idf#8219](https://github.com/espressif/esp-idf/pull/8219)?
> 
> I've noticed that the ELF file posted here [espressif/esptool#661 (comment)](https://github.com/espressif/esptool/issues/661#issuecomment-924979468) has a weird `.rwtext` section. That is that evil "Segment 8" which is causing the unaligned PADDING before it.
> 
> I could not found a pure ESP-IDF project which produces this and it probably should be in `.flash.text`.

But did anybody actually had the problem happening with anything else than the combo ESP32C3 + ESP-IDF V4.3.X in the first place?

E.g. if I rebuild against ESP-IDF V4.4 or against ESP-IDF V5.0, **or** I use a different chip from ESP32C3 (even with ESP-IDF V4.3.X!), the problem never happens?

ivmarkov, 2022-01-10:
@dobairoland Wooow, I might have finally caught it and it just might have been *me* who introduced this oddity! (blush)
You are on to something!

The strange `.rwtext` section is likely coming from... *my* code, which indeed only triggers for ESP V34.3.X *and* ESP32C3.

It is a patch code in `esp-idf-sys` written in Rust which "backports" to ESP-IDF V4.3.X the ESP32C3 atomic syscalls.
All I wanted is to place them in the IRAM memory, and for some reason I thought I should put them in `rwtext` section. (blush)
See here for the gory details: https://github.com/esp-rs/esp-idf-sys/blob/master/src/patches/atomics_esp32c3.rs#L40

So... what is the correct section I should use, which is equivalent to the `__IRAM` attribute in C?

dobairoland, 2022-01-10:
@ivmarkov I have only one MAP file where this happens:

```
.rwtext         0x0000000042079fea      0x646
 .rwtext        0x0000000042079fea      0x646 /Users/XY/Work/embedded-interface-demo/server/target/riscv32imc-esp-espidf/release/deps/server-bc90bce522a312ad.server.73e9cc15-cgu.6.rcgu.o
                0x0000000042079fea                __atomic_load_1
                0x000000004207a016                __atomic_store_1
                0x000000004207a044                __atomic_exchange_1
                0x000000004207a07e                __atomic_compare_exchange_1
                0x000000004207a0d6                __atomic_fetch_add_1
                0x000000004207a114                __atomic_fetch_sub_1
                0x000000004207a152                __atomic_fetch_and_1
                0x000000004207a190                __atomic_fetch_or_1
                0x000000004207a1ce                __atomic_fetch_xor_1
                0x000000004207a20c                __atomic_load_2
                0x000000004207a238                __atomic_store_2
                0x000000004207a266                __atomic_exchange_2
                0x000000004207a2a0                __atomic_compare_exchange_2
                0x000000004207a2f8                __atomic_fetch_add_2
                0x000000004207a336                __atomic_fetch_sub_2
                0x000000004207a374                __atomic_fetch_and_2
                0x000000004207a3b2                __atomic_fetch_or_2
                0x000000004207a3f0                __atomic_fetch_xor_2
                0x000000004207a42e                __atomic_load_4
                0x000000004207a458                __atomic_store_4
                0x000000004207a484                __atomic_exchange_4
                0x000000004207a4bc                __atomic_compare_exchange_4
                0x000000004207a50e                __atomic_fetch_add_4
                0x000000004207a548                __atomic_fetch_sub_4
                0x000000004207a582                __atomic_fetch_and_4
                0x000000004207a5bc                __atomic_fetch_or_4
                0x000000004207a5f6                __atomic_fetch_xor_4
```

I assume you have something similar. I assume that those atomic functions could be in `.flash.text`. Please try to replace `.rwtext`  to `.flash.text` in you project.

ivmarkov, 2022-01-10:
But .flash.text <> IRAM?

To repeat:
> So... what is the correct section I should use, which is equivalent to the __IRAM attribute in C?

ivmarkov, 2022-01-10:
> So... what is the correct section I should use, which is equivalent to the `__IRAM` attribute in C?

`iram0.text`?

ivmarkov, 2022-01-10:
Any section that has a name of type `iram1+` or `iram1.+`?

dobairoland, 2022-01-10:
@ivmarkov What about `irom0.text`? It would put it right at the place I was trying with my PR: https://github.com/espressif/esp-idf/blob/846fdba4f7e65de913484fd4ec7d1acf5d5c2642/components/esp_system/ld/esp32c3/sections.ld.in#L222

Why do you want it to be in iram? Code doesn't have to be in read-write area.

igrr, 2022-01-10:
It probably does have to be in IRAM since atomics may be emitted for code that is already in IRAM. That could be because the code should run while the Flash cache is disabled. IDF code places atomics into IRAM here: https://github.com/espressif/esp-idf/blob/df1f9ec26e81138494462451889e7cb05dfc4ad4/components/newlib/newlib.lf#L7

igrr, 2022-01-10:
@ivmarkov I think any section with name matching `.iram1.*` should work. Ideally you'd want unique `.iram.N` sections for each of the functions (with different values of N), so that the linker can eliminate unused code if not all functions from this object file need to be linked.

ivmarkov, 2022-01-11:
> @ivmarkov I think any section with name matching `.iram1.*` should work. Ideally you'd want unique `.iram.N` sections for each of the functions (with different values of N), so that the linker can eliminate unused code if not all functions from this object file need to be linked.

@igrr Thanks a lot, that's the info - that's exactly what I wanted to confirm. 
I'll replace the custom Rust-based atomics code for ESP32C3+V4.3.1 with a build-time patch in C against the ESP-IDF, because actually only two atomic primitives are missing for riscv (atomic_load and atomic_store - the rest is in-place), and the two missing atomics primitives for the ESP32S2 chip and V4.3.1 are also implemented with a [build-time patch](https://github.com/esp-rs/esp-idf-sys/blob/master/patches/missing_xtensa_atomics_fix.diff), so that will bring some symmetry.

However how to place Rust code in IRAM is still useful e.g. for [this](https://github.com/esp-rs/esp-idf-hal/blob/master/src/interrupt.rs) and [this](https://github.com/esp-rs/esp-idf-hal/blob/master/src/cpu.rs) utility code in esp-idf-hal, as that functionality is here to stay and will soon graduate from its "experimental" status.

ivmarkov, 2022-01-11:
With the above ^^^ commits I would consider this issue finally fixed!

closed 2022-01-11 by ivmarkov

FrozenDroid, 2022-01-11:
Amazing work!

dobairoland, 2022-01-11:
Thanks @ivmarkov for the fix. Thanks @igrr for the help.