ivmarkov/rust-esp32-std-demo #15: Cargo build with pio failure

pfesenmeier, 2021-10-01:
Hello, thank you for this example, really exciting!
I am a bit a beginner with these non-rust build concepts, and curious if anyone could point me in a good direction:

cargo 1.57.0-nightly (0121d66aa 2021-09-22)
Ubuntu clang version 12.0.0-3ubuntu1~20.04.4 (sudo mv /usr/bin/clang /usr/bin/clang-moved; ln -s /usr/bin/clang-12 /usr/bin/clang)
In .cargo/config.toml: `target = "riscv32imc-esp-espidf"`
build output:

```
... shortened ...
   Compiling esp-idf-sys v0.20.2
   Compiling esp-idf-hal v0.20.0
   Compiling esp-idf-svc v0.20.1
   Compiling rust-esp32-std-hello v0.18.2 (/home/pfes/rust-esp32-std-hello)
error: failed to run custom build command for `esp-idf-sys v0.20.2`

Caused by:
  process didn't exit successfully: `/home/pfes/rust-esp32-std-hello/target/debug/build/esp-idf-sys-57948bb6b1a5e6e3/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.0 -> /home/pfes/.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=./patches/master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=/home/pfes/rust-esp32-std-hello/./sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32-c3-devkitm-1; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
  PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32-C3-DevKitM-1
  HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES:
   - framework-espidf 3.40300.0 (4.3.0)
   - tool-cmake 3.16.4
   - tool-esptoolpy 1.30100.210531 (3.1.0)
   - tool-ninja 1.7.1
   - toolchain-riscv-esp 1.80400.0 (8.4.0)
  Reading CMake configuration...
  Generating assembly for certificate bundle...
  LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Compiling .pio/build/debug/src/main.o
  Generating LD script .pio/build/debug/esp32c3_out.ld
  Generating partitions .pio/build/debug/partitions.bin
  Compiling .pio/build/debug/app_trace/app_trace.o
  Compiling .pio/build/debug/app_trace/app_trace_util.o
  Compiling .pio/build/debug/app_trace/host_file_io.o
  Compiling .pio/build/debug/app_trace/gcov/gcov_rtio.o
  Compiling .pio/build/debug/app_update/esp_ota_ops.o
... etc ...
  Compiling .pio/build/debug/lwip/port/esp32/netif/wlanif.o
  Compiling .pio/build/debug/lwip/port/esp32/netif/ethernetif.o
  Compiling .pio/build/debug/lwip/port/esp32/vfs_lwip.o
  Compiling .pio/build/debug/mbedtls/esp_crt_bundle/esp_crt_bundle.o

  --- stderr
  CMake Error at /home/pfes/.platformio/packages/framework-espidf/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
    file failed to open for reading (No such file or directory):

      /home/pfes/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-9073ee0a67e8cf49/out/esp-idf/.pio/build/debug/x509_crt_bundle

  *** Error 1
  *** [.pio/build/debug/.pio/build/debug/x509_crt_bundle.o] Source `.pio/build/debug/x509_crt_bundle.S' not found, needed by target `.pio/build/debug/.pio/build/debug/x509_crt_bundle.o'.
  ========================= [FAILED] Took 60.48 seconds =========================
  Environment    Status    Duration
  -------------  --------  ------------
  debug          FAILED    00:01:00.480
  ==================== 1 failed, 0 succeeded in 00:01:00.480 ====================
  Error: No such file or directory (os error 2)
```

georgik, 2021-10-01:
The problem is caused by Python Cryptography v35 released 29th September.
Workaround:
```
source ~/.platformio/penv/bin/activate
pip install "cryptography<4"
cargo +esp build
```
The topic is being discussed in https://matrix.to/#/#esp-rs:matrix.org

pfesenmeier, 2021-10-05:
That did it! Thank you.

closed 2021-10-05 by pfesenmeier