ivmarkov/rust-esp32-std-demo #16: Failed to build project: Failed to read: sdkconfig.debug

StefansProjects, 2021-10-01:
I followed your guide to build the std-hello example for a esp32 on Windows 10 with `cargo build`.  When invoking `cargo build --release`, the missing file is `sdkconfig.release`.  Rust build seems to run okay but in the final build step PlatformIO is invoked and does not find the file. I have ensured that my PlatformIO installation is up to date.

```
...
error: failed to run custom build command for `esp-idf-sys v0.20.2`

Caused by:
  process didn't exit successfully: `C:\projekte\rust-esp32-std-hello\target\debug\build\esp-idf-sys-dc637b5a811cf755\build-script-build` (exit code: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.0 -> C:\Users\stefa\.platformio\penv\Scripts\platformio.exe
  cargo:rerun-if-changed=.\patches\master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=.\patches\missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=.\patches\ping_setsockopt_fix.diff
  cargo:rerun-if-changed=.\patches\pthread_destructor_fix.diff
  cargo:rerun-if-changed=C:\projekte\rust-esp32-std-hello\.\sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
  PLATFORM: Espressif 32 (3.2.0) > AI Thinker ESP32-CAM
  HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES:
   - framework-espidf 3.40200.210118 (4.2.0)
   - tool-cmake 3.16.4
   - tool-esptoolpy 1.30000.201119 (3.0.0)
   - tool-idf 1.0.1
   - tool-mconf 1.4060000.20190628 (406.0.0)
   - tool-ninja 1.9.0
   - toolchain-esp32ulp 1.22851.191205 (2.28.51)
   - toolchain-xtensa32 2.80400.210211 (8.4.0)
  Reading CMake configuration...
  LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 1 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Retrieving maximum program size .pio\build\debug\firmware.elf
  Checking size .pio\build\debug\firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.2% (used 10436 bytes from 327680 bytes)
  Flash: [=         ]  14.3% (used 150364 bytes from 1048576 bytes)
  ========================= [SUCCESS] Took 15.08 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  debug          SUCCESS   00:00:15.080
  ========================= 1 succeeded in 00:00:15.080 =========================

  --- stderr
  Error: Failed to read '"C:\\projekte\\rust-esp32-std-hello\\target\\xtensa-esp32-espidf\\debug\\build\\esp-idf-sys-22b339291a313812\\out\\esp-idf\\sdkconfig.debug"'
```

ivmarkov, 2021-10-01:
This is really strange, because - indeed - the PlatformIO output contains
`Building in debug mode`

.. while if you really invoked `cargo build --release`, it should contain `Building in release mode` or something.

The thing is, I cannot reproduce it.
The following things work fine for me, in that order:
`cargo clean; cargo build --release`
`cargo build`
`cargo build --release`

Can you try to invoke `cargo clean` first, and then again `cargo build --release` and see if that still fails?
Also, have you changed the sample project in any way?

StefansProjects, 2021-10-01:
Thank you for your help. I have just modified `SSID` and `PASS` (to connect to my Wifi) in `main.rs` as described in the instructions. To further explain my first issue, the error happens reproduable in both debug and release builds (with a different but similar file missing: sdkconfig.[debug, release].).

Here is the full output from beginning when invoking `cargo clean; cargo build --release`. I`m using the PowerShell.

```
PS C:\projekte\rust-esp32-std-hello> cargo clean;cargo build --release
warning: unused config key `unstable.extra-link-arg` in `C:\projekte\rust-esp32-std-hello\.cargo\config.toml`
warning: unused config key `unstable.configurable-env` in `C:\projekte\rust-esp32-std-hello\.cargo\config.toml`
warning: unused config key `unstable.extra-link-arg` in `C:\projekte\rust-esp32-std-hello\.cargo\config.toml`
warning: unused config key `unstable.configurable-env` in `C:\projekte\rust-esp32-std-hello\.cargo\config.toml`
   Compiling compiler_builtins v0.1.49
   Compiling core v0.0.0 (C:\Users\stefa\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core)
   Compiling libc v0.2.98 (https://github.com/rust-lang/libc.git#67562637)
   Compiling cc v1.0.69
   Compiling std v0.0.0 (C:\Users\stefa\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std)
   Compiling memchr v2.4.1
   Compiling proc-macro2 v1.0.28
   Compiling autocfg v1.0.1
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.75
   Compiling winapi v0.3.9
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.101
   Compiling log v0.4.14
   Compiling serde v1.0.130
'+memctl' is not a recognized feature for this target (ignoring feature)
'+memctl' is not a recognized feature for this target (ignoring feature)
   Compiling serde_derive v1.0.130
   Compiling lazy_static v1.4.0
   Compiling crossbeam-utils v0.8.5
   Compiling fnv v1.0.7
   Compiling ryu v1.0.5
   Compiling crossbeam-epoch v0.9.5
   Compiling untrusted v0.7.1
   Compiling regex-syntax v0.6.25
   Compiling spin v0.5.2
   Compiling rayon-core v1.9.1
   Compiling version_check v0.9.3
   Compiling anyhow v1.0.43
   Compiling tinyvec_macros v0.1.0
   Compiling serde_json v1.0.67
   Compiling glob v0.3.0
   Compiling scopeguard v1.1.0
   Compiling matches v0.1.9
   Compiling once_cell v1.8.0
   Compiling unicode-width v0.1.8
   Compiling unicode-bidi v0.3.6
   Compiling ppv-lite86 v0.2.10
   Compiling either v1.6.1
   Compiling adler v1.0.2
   Compiling gimli v0.25.0
   Compiling percent-encoding v2.1.0
   Compiling bitflags v1.3.2
   Compiling strsim v0.8.0
   Compiling rustc-demangle v0.1.21
   Compiling vec_map v0.8.2
   Compiling base64 v0.13.0
   Compiling humantime v2.1.0
   Compiling bindgen v0.57.0
   Compiling shlex v0.1.1
   Compiling itoa v0.4.8
   Compiling lazycell v1.3.0
   Compiling chunked_transfer v1.4.0
   Compiling rustc-hash v1.1.0
   Compiling zero v0.1.2
   Compiling peeking_take_while v0.1.2
   Compiling shlex v1.1.0
   Compiling futures-core v0.3.17
   Compiling strsim v0.10.0
   Compiling ident_case v1.0.1
[... 1373 lines cut ...]
  Linking .pio\build\release\bootloader.elf
  Building .pio\build\release\bootloader.bin
  esptool.py v3.0
  __action_dump([".pio\build\release\firmware.elf"], [".pio\build\release\src\main.o"])
  Linking .pio\build\release\firmware.elf
  Retrieving maximum program size .pio\build\release\firmware.elf
  Checking size .pio\build\release\firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.2% (used 10436 bytes from 327680 bytes)
  Flash: [=         ]  14.3% (used 150364 bytes from 1048576 bytes)
  Building .pio\build\release\firmware.bin
  esptool.py v3.0
  ========================= [SUCCESS] Took 62.74 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  release        SUCCESS   00:01:02.736
  ========================= 1 succeeded in 00:01:02.736 =========================

  --- stderr
  Error: Failed to read '"C:\\projekte\\rust-esp32-std-hello\\target\\xtensa-esp32-espidf\\release\\build\\esp-idf-sys-c4b1b1bc67c76a84\\out\\esp-idf\\sdkconfig.release"'

  Caused by:
      Das System kann die angegebene Datei nicht finden. (os error 2)
```

ivmarkov, 2021-10-01:
OK clear. You are running a too old ESP-IDF (4.2) inside your PlatformIO. The easiest fix is to just wipe out (or rename) your `~/.platformio` directory, so that it can be re-downloaded during the build.

Unfortunately, this will bring another error that we have since today and which is totally unrelated to this project, and which will require upstream bugfix in ESP-IDF itself, as well as in PlatformIO. Once you do what I suggest above and invoke `cargo build [--release]`, the build will fail with something like:
```
CMake Error at /Users/georgik/.platformio/packages/framework-espidf/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
    file failed to open for reading (No such file or directory):

      /Users/georgik/projects/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-e0d5dac67e377451/out/esp-idf/.pio/build/debug/x509_crt_bundle

  *** Error 1
  *** [.pio/build/debug/.pio/build/debug/x509_crt_bundle.o] Source `.pio/build/debug/x509_crt_bundle.S' not found, needed by target `.pio/build/debug/.pio/build/debug/x509_crt_bundle.o'.
```

To fix it, do:
```
source ~/.platformio/penv/bin/activate
pip install "cryptography<4"
cargo +esp build [--release]
```

More info in the chat: https://app.element.io/#/room/#esp-rs:matrix.org

StefansProjects, 2021-10-06:
After cleaning the `.platformio` dir I had issues around missing `CMAKE_ROOT` which I couldn`t resolve so I decided to move to a Ubuntu VM, where the build finally worked.
Thanks for your help.

closed 2021-10-07 by ivmarkov