ivmarkov/rust-esp32-std-demo #17: Warning rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000

pekkah, 2021-10-05:
Got a simple hello world app running on [LILYGO® TTGO T8 ESP32-S2 V1.1 ST77789](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1321) but I'm getting a warning to the bootlog about invalid value RTC_XTAL_FREQ_REG value: 0x00000000.

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17dc
load:0x4004c000,len:0xa48
load:0x40050000,len:0x2e44
entry 0x4004c204
I (21) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (21) boot: compile time 13:43:49
I (21) boot: chip revision: 0
I (25) boot.esp32s2: SPI Speed      : 80MHz
I (30) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0bce8h ( 48360) map
I (101) esp_image: segment 1: paddr=0001bd10 vaddr=3ffbd960 size=028f0h ( 10480) load
I (103) esp_image: segment 2: paddr=0001e608 vaddr=40022000 size=01a10h (  6672) load
I (110) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=20bc4h (134084) map
I (143) esp_image: segment 4: paddr=00040bec vaddr=40023a10 size=09f4ch ( 40780) load
I (153) esp_image: segment 5: paddr=0004ab40 vaddr=50000000 size=00010h (    16) load
I (160) boot: Loaded app from partition at offset 0x10000
I (160) boot: Disabling RNG early entropy source...
I (174) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (174) cpu_start: Pro cpu up.
W (175) r����f���������xf���`�f�fff��f`ffff�f`~~�ffx���f��f����`��~�x�~f����~�f���������xf���`�f�fff��f`ffff�f`~~�ffx���f��f����`��~�x�~f��f���~c_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
W (387) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
W (199) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
W (212) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
W (213) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
�`��~�x�~f��x���~c_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
W (460) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
I (245) cpu_start: Pro cpu start user code
I (245) cpu_start: cpu freq: 160000000
I (245) cpu_start: Application information:
I (250) cpu_start: Project name:     esp-idf
I (255) cpu_start: App version:      1
I (259) cpu_start: Compile time:     Oct  4 2021 19:57:36
I (265) cpu_start: ELF file SHA256:  0000000000000000...
I (271) cpu_start: ESP-IDF:          4.3.0
I (276) heap_init: Initializing. RAM available for dynamic allocation:
I (283) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (289) heap_init: At 3FFC0B98 len 0003B468 (237 KiB): DRAM
I (296) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (302) spi_flash: detected chip: boya
I (306) spi_flash: flash io: dio
W (310) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (327) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
I (330) cpu_start: Starting scheduler on PRO CPU.
Hello world
```

ivmarkov, 2021-10-07:
I don't think this is related to this demo crate, because the error is happening **before** the code of the crate even starts to run.

Are you sure you are powering your board properly?

ivmarkov, 2021-10-07:
Another reason could be that simply esp32-s2 support is just too new, and it is not always working properly in the ESP-IDF V4.3, which is used by default when compiling this crate.

As an experiment, you can do the following:
* in `.cargo/config.toml` uncomment the line `ESP_IDF_VERSION = { value = "master" }`
* rebuild the project with `cargo clean; cargo build --features native`

The above will use the latest ESP-IDF master (V4.4.dev).

ivmarkov, 2021-10-12:
Yes - that's because Espressif committed to ESP-IDF master precisely this [fix](https://github.com/espressif/esp-idf/commit/b86fe0c66ca17d5907ae26ecdffc9d4eb79cec07) ~ 3 hours ago. :)

This patch needs to be removed from `esp-idf-sys` now. Let me quickly do that...

ivmarkov, 2022-01-14:
I'm closing this because it is very unlikely for this problem to come from the demo itself. For me, inability to operate / calibrate the xtal quartz is either a hardware problem of your board, or a software issue in the esp idf itself.

closed 2022-01-14 by ivmarkov