ivmarkov/rust-esp32-std-demo #18: Build failure

pekkah, 2021-10-06:
No idea what happened but today I'm getting this error when trying to build the project. Tried also cloning this repo again and getting the same error.

```
error: failed to run custom build command for `esp-idf-sys v0.20.4`

Caused by:
  process didn't exit successfully: `C:\projects\pekkah\smoker-ai-esp32\target\debug\build\esp-idf-sys-2121b71c259a5271\build-script-build` (exit code: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.0 -> C:\Users\pekka\.platformio\penv\Scripts\platformio.exe
  cargo:rerun-if-changed=.\patches\master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=.\patches\missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=.\patches\ping_setsockopt_fix.diff
  cargo:rerun-if-changed=.\patches\pthread_destructor_fix.diff
  cargo:rerun-if-changed=C:\projects\pekkah\smoker-ai-esp32\.\sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: featheresp32-s2; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32-s2.html
  PLATFORM: Espressif 32 (3.3.2) > Adafruit ESP32-S2 Feather Development Board
  HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES:
   - framework-espidf 3.40300.0 (4.3.0)
   - tool-cmake 3.16.4
   - tool-esptoolpy 1.30100.210531 (3.1.0)
   - tool-idf 1.0.1
   - tool-mconf 1.4060000.20190628 (406.0.0)
   - tool-ninja 1.9.0
   - toolchain-esp32s2ulp 1.22851.191205 (2.28.51)
   - toolchain-riscv-esp 1.80400.0 (8.4.0)
   - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
  Reading CMake configuration...
  LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Retrieving maximum program size .pio\build\debug\firmware.elf
  Checking size .pio\build\debug\firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.7% (used 12136 bytes from 327680 bytes)
  Flash: [=         ]  14.0% (used 146934 bytes from 1048576 bytes)
  ========================= [SUCCESS] Took 10.68 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  debug          SUCCESS   00:00:10.681
  ========================= 1 succeeded in 00:00:10.681 =========================
  cargo:rerun-if-changed=C:\Users\pekka\.cargo\registry\src\github.com-1ecc6299db9ec823\esp-idf-sys-0.20.4\src\include\esp-idf\bindings.h

  --- stderr
  Output: "C:\\projects\\pekkah\\smoker-ai-esp32\\target\\xtensa-esp32s2-espidf\\debug\\build\\esp-idf-sys-a16082cedf72ba1e\\out\\bindings.rs"
  Bindgen builder flags: ["C:\\Users\\pekka\\.cargo\\registry\\src\\github.com-1ecc6299db9ec823\\esp-idf-sys-0.20.4\\src\\include\\esp-idf\\bindings.h", "--rust-target", "1.40", "--blacklist-function", "strtold", "--blacklist-function", "_strtold_r", "--blacklist-function", "esp_eth_mac_new_esp32", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_pm/include", "-IC:/projects/pekkah/smoker-ai-esp32/target/xtensa-esp32s2-espidf/debug/build/esp-idf-sys-a16082cedf72ba1e/out/esp-idf/.pio/build/debug/config", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/newlib/platform_include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/freertos/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/freertos/port/xtensa/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_hw_support/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32s2/.", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_hw_support/port/esp32s2/private_include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/heap/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/log/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/lwip/include/apps", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/lwip/include/apps/sntp", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/lwip/lwip/src/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/lwip/port/esp32/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/soc/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/soc/esp32s2/.", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/soc/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/hal/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/hal/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_rom/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_rom/esp32s2", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_common/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_system/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/driver/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/driver/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_ringbuf/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/efuse/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/efuse/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/xtensa/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/xtensa/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/espcoredump/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_timer/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_ipc/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/vfs/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_wifi/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_wifi/esp32s2/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_event/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_netif/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_eth/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/tcpip_adapter/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/app_trace/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/mbedtls/port/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/mbedtls/mbedtls/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/mbedtls/esp_crt_bundle/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/app_update/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/spi_flash/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/bootloader_support/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/nvs_flash/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/pthread/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_gdbstub/include", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_gdbstub/xtensa", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/esp_gdbstub/esp32s2", "-IC:/Users/pekka/.platformio/packages/framework-espidf/components/wpa_supplicant/include", "-IC:/User [... cut ...] LAVE_LINK_ENABLED", "-DESP_IDF_COMP_SPI_FLASH_ENABLED", "-DESP_IDF_COMP_ESP_ADC_CAL_ENABLED", "-target", "xtensa"]
  C:\Users\pekka\.cargo\registry\src\github.com-1ecc6299db9ec823\esp-idf-sys-0.20.4\src\include\esp-idf\bindings.h:95:10: fatal error: 'touch_sensor.h' file not found
  C:\Users\pekka\.cargo\registry\src\github.com-1ecc6299db9ec823\esp-idf-sys-0.20.4\src\include\esp-idf\bindings.h:95:10: fatal error: 'touch_sensor.h' file not found, err: true
  Error: Failed to generate bindings
```

ivmarkov, 2021-10-06:
I think I know what this is. Will look into it tmr, thanks for reporting.

ivmarkov, 2021-10-07:
Fixed:
* Do `cargo update` to get the latest 0.20.5 release of `esp-idf-sys`
* ... or just pull `rust-esp32-std-hello`

Closing. Feel free to reopen if it still does not work for you.

ivmarkov, 2021-10-07:
One more thing: Windows is **extremely sensitive** to file path lengths. Can you checkout the `rust-esp32-std-hello` project in your `C:\` root (or as close to it as possible, using very short path names) and try the build again?

pekkah, 2021-10-07:
Thanks @ivmarkov