ivmarkov/rust-esp32-std-demo #27: Linker Error: undefined reference to `bind'

cathiele, 2021-11-01:
I can not build the project with the `xtensa-esp32-espidf` target, it just works with the  `riscv32imc-esp-espidf` target. When i try to build it for classic esp32 targets i get a linker error:
```
Error: Linker /home/myuser/clones/tmp/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc failed: exit status: 1
          
STDERR OUTPUT:
          /home/myuser/clones/tmp/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/myuser/clones/tmp/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/deps/libstd-8cb85e60f741e4fa.rlib(std-8cb85e60f741e4fa.std.1ubzggxa-cgu.6.rcgu.o):(.literal._ZN3std10sys_common3net11TcpListener4bind17h0813b699c582a6dfE+0x8): undefined reference to `bind'
          collect2: error: ld returned 1 exit status
```

i also tried with "native" feature and `ESP_IDF_VERSION` set to  `master` but i still get the same linker error.

When remove the  ```test_tcp_bind```-Code the build (and flashing to the controller) works fine.

What can i do to fix this problem?

georgik, 2021-11-02:
@cathiele Thank you for reporting the problem.

Risc architecture has different requirements which are already covered by upstream version of Rust.

Please check following requirements:
- rustc >= 1.55.0
- rust stable toolchain is >= 1.55.0
- rust esp toolchain is >= 1.55.0

Please, test the following script which should fix all necessary requirements and export all necessary variables (branch test-flash):
https://github.com/esp-rs/rust-build/tree/feature/test-flash
`./test-rust-toolchain.sh`

cathiele, 2021-11-02:
Using the in README.md linked instructions from [Rust Espressif compiler fork and the Espressif LLVM Clang fork](https://github.com/esp-rs/rust) i installed the toolchain from the ```esp```-Branch as mentioned there, and not the current ```esp-1.56.0```-branch of https://github.com/esp-rs/rust . 

using ```esp-1.56.0``` all works fine. thank you @georgik for your help!

closed 2021-11-02 by cathiele