ivmarkov/rust-esp32-std-demo #29: stop to complete the build process

bss03arg, 2021-11-04:
hello

I use mac book pro14 with m1pro cpu and install the rust nightly version.
.cargo/config.toml:
target = "riscv32imc-esp-espidf"

when I run cargo build, the process blocked without any information. Could someone help me ?

cargo build
warning: Patch `getrandom v0.1.16 (https://github.com/esp-rs-compat/getrandom.git?branch=0.1#d942e06a)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling compiler_builtins v0.1.49
   Compiling core v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.103
   Compiling cc v1.0.69
   Compiling std v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/std)
   Compiling proc-macro2 v1.0.32
   Compiling unicode-xid v0.2.2
   Compiling libc v0.2.106
   Compiling memchr v2.4.1
   Compiling syn v1.0.81
   Compiling cfg-if v1.0.0
   Compiling cc v1.0.71
   Compiling serde v1.0.130
   Compiling serde_derive v1.0.130
   Compiling log v0.4.14
   Compiling autocfg v1.0.1
   Compiling fnv v1.0.7
   Compiling ryu v1.0.5
   Compiling untrusted v0.7.1
   Compiling regex-syntax v0.6.25
   Compiling glob v0.3.0
   Compiling version_check v0.9.3
   Compiling tinyvec_macros v0.1.0
   Compiling serde_json v1.0.68
   Compiling anyhow v1.0.45
   Compiling crossbeam-utils v0.8.5
   Compiling matches v0.1.9
   Compiling lazy_static v1.4.0
   Compiling once_cell v1.8.0
   Compiling percent-encoding v2.1.0
   Compiling adler v1.0.2
   Compiling unicode-width v0.1.9
   Compiling same-file v1.0.6
   Compiling gimli v0.26.1
   Compiling bitflags v1.3.2
   Compiling unicode-bidi v0.3.7
   Compiling ppv-lite86 v0.2.15
   Compiling strsim v0.8.0
   Compiling termcolor v1.1.2
   Compiling rustc-demangle v0.1.21
   Compiling vec_map v0.8.2
   Compiling humantime v2.1.0
   Compiling ansi_term v0.11.0
   Compiling bindgen v0.57.0
   Compiling itoa v0.4.8
   Compiling peeking_take_while v0.1.2
   Compiling remove_dir_all v0.5.3
   Compiling unicode-segmentation v1.8.0
   Compiling zero v0.1.2
   Compiling lazycell v1.3.0
   Compiling base64 v0.13.0
   Compiling shlex v0.1.1
   Compiling rustc-hash v1.1.0
   Compiling chunked_transfer v1.4.0
   Compiling shlex v1.1.0
   Compiling futures-core v0.3.17
   Compiling ident_case v1.0.1
   Compiling strsim v0.10.0
   Compiling az v1.1.2
   Compiling async-trait v0.1.51
   Compiling paste v1.0.5
   Compiling heapless v0.7.7
   Compiling libloading v0.7.1
   Compiling cmake v0.1.46
   Compiling miniz_oxide v0.4.4
   Compiling num-traits v0.2.14
   Compiling tinyvec v1.5.0
   Compiling unwind v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/unwind)
   Compiling nom v5.1.2
   Compiling clang-sys v1.3.0
   Compiling thread_local v1.1.3
   Compiling textwrap v0.11.0
   Compiling form_urlencoded v1.0.1
   Compiling ring v0.16.20
   Compiling backtrace v0.3.63
   Compiling walkdir v2.3.2
   Compiling xmas-elf v0.8.0
   Compiling heck v0.3.3
   Compiling addr2line v0.17.0
   Compiling unicode-normalization v0.1.19
   Compiling aho-corasick v0.7.18
   Compiling bstr v0.2.17
   Compiling object v0.27.1
   Compiling quote v1.0.10
   Compiling getrandom v0.2.3 (https://github.com/esp-rs-compat/getrandom.git#b8cb133e)
   Compiling atty v0.2.14
   Compiling which v3.1.1
   Compiling dirs-sys v0.3.6
   Compiling remove_dir_all v0.7.0
   Compiling rustc-std-workspace-core v1.99.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling regex v1.5.4
   Compiling idna v0.2.3
   Compiling rand_core v0.6.3
   Compiling clap v2.33.3
   Compiling dirs v4.0.0
   Compiling rand_chacha v0.3.1
   Compiling url v2.2.2
   Compiling globset v0.4.8
   Compiling env_logger v0.8.4
   Compiling cexpr v0.4.0
   Compiling alloc v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/alloc)
   Compiling cfg-if v0.1.10
   Compiling rand v0.8.4
   Compiling webpki v0.22.0
   Compiling sct v0.7.0
   Compiling darling_core v0.13.0
   Compiling ignore v0.4.18
   Compiling rustls v0.20.0
   Compiling webpki-roots v0.22.1
   Compiling tempfile v3.2.0
   Compiling rustc-std-workspace-alloc v1.99.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/rustc-std-workspace-alloc)
   Compiling panic_abort v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/panic_abort)
   Compiling globwalk v0.8.1
   Compiling panic_unwind v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/panic_unwind)
   Compiling std_detect v0.1.5 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/stdarch/crates/std_detect)
   Compiling hashbrown v0.11.0
   Compiling strum_macros v0.21.1
   Compiling thiserror-impl v1.0.30
   Compiling derivative v2.2.0
   Compiling ureq v2.3.0
   Compiling darling_macro v0.13.0
   Compiling strum v0.21.0
   Compiling thiserror v1.0.30
   Compiling darling v0.13.0
   Compiling proc_macro v0.0.0 (/Users/ccui/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/proc_macro)
   Compiling enumset_derive v0.5.5
   Compiling nb v1.0.0
   Compiling void v1.0.2
   Compiling cache-padded v1.1.1
   Compiling futures-io v0.3.17
   Compiling byteorder v1.4.3
   Compiling pin-project-lite v0.2.7
   Compiling parking v2.0.0
   Compiling fastrand v1.5.0
   Compiling waker-fn v1.1.0
   Compiling event-listener v2.5.1
   Compiling async-task v4.0.3
   Compiling base64 v0.12.3
   Compiling display-interface v0.4.1
   Compiling slab v0.4.5
   Compiling atomic-waker v1.0.0
   Compiling mutex-trait v0.2.0
   Compiling byte-slice-cast v0.3.5
   Compiling stable_deref_trait v1.2.0
   Compiling cty v0.2.2
   Compiling micromath v1.1.1
   Compiling nb v0.1.3
   Compiling socket2 v0.4.2 (https://github.com/esp-rs-compat/socket2#7217ebe8)
   Compiling concurrent-queue v1.2.2
   Compiling hash32 v0.2.1
   Compiling futures-lite v1.12.0
   Compiling async-lock v2.4.0
   Compiling http-auth-basic v0.1.3
   Compiling cstr_core v0.2.4
   Compiling embedded-graphics-core v0.3.3
   Compiling float-cmp v0.8.0
   Compiling polling v2.1.0 (https://github.com/esp-rs-compat/polling#260dded7)
   Compiling embedded-hal v0.2.6
   Compiling async-channel v1.6.1
   Compiling async-executor v1.4.1
   Compiling toml v0.5.8
   Compiling either v1.6.1
   Compiling no-std-net v0.5.0
   Compiling enumset v1.0.7
   Compiling async-io v1.6.0
   Compiling display-interface-i2c v0.4.0
   Compiling display-interface-spi v0.4.1
   Compiling blocking v1.0.2
   Compiling embedded-graphics v0.7.1
   Compiling ili9341 v0.5.0 (https://github.com/yuri91/ili9341-rs#c3d0d08a)
   Compiling st7789 v0.6.1
   Compiling which v4.2.2
   Compiling cargo_toml v0.9.2
   Compiling proc-macro-crate v1.1.0
   Compiling ssd1306 v0.7.0
   Compiling async-fs v1.5.0
   Compiling async-net v1.6.1
   Compiling num_enum_derive v0.5.4
   Compiling embuild v0.25.1
   Compiling smol v1.2.5 (https://github.com/esp-rs-compat/smol#b9a770e7)
   Compiling num_enum v0.5.4
   Compiling embedded-svc v0.12.0
   Compiling esp-idf-sys v0.24.3
   Compiling esp-idf-hal v0.23.2
   Compiling esp-idf-svc v0.28.1
   Compiling rust-esp32-std-demo v0.20.2 (/Users/ccui/esp/rust-esp32-std-demo)
    Building [=======================> ] 276/284: esp-idf-sys(build) 

thank you very much!

ivmarkov, 2021-11-04:
How do you know it stopped? Building esp-idf-sys takes a lot of time.e - especially the first time when it is downloading all sorts of tooling from the internet behind the scenes.

How long did you wait for it?

bss03arg, 2021-11-04:
Understood, I have just realized this, the process try to download the esp-idf. My problem is, the internet speed to github is very unstable and normally I use socket5 proxy. But if I set a proxy in console, I got some errors with this:

user@chundeMacBook-Pro-14 rust-esp32-std-demo % ALL_PROXY=socks5://127.0.0.1:1086 cargo build
warning: Patch `getrandom v0.1.16 (https://github.com/esp-rs-compat/getrandom.git?branch=0.1#d942e06a)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling esp-idf-sys v0.24.3
error: failed to run custom build command for `esp-idf-sys v0.24.3`

Caused by:
  process didn't exit successfully: `/Users/ccui/esp/rust-esp32-std-demo/target/debug/build/esp-idf-sys-a4f82c8312588d4e/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.2 -> /Users/ccui/.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=/Users/ccui/esp/esp-idf/sdkconfig.rename
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32-c3-devkitm-1; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Tool Manager: Installing platformio/toolchain-riscv-esp @ ~1.80400.0

  --- stderr
  Error: Traceback (most recent call last):
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/__main__.py", line 129, in main
      cli()  # pylint: disable=no-value-for-parameter
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1128, in __call__
      return self.main(*args, **kwargs)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1053, in main
      rv = self.invoke(ctx)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/commands/__init__.py", line 44, in invoke
      return super(PlatformioCLI, self).invoke(ctx)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1659, in invoke
      return _process_result(sub_ctx.command.invoke(sub_ctx))
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1395, in invoke
      return ctx.invoke(self.callback, **ctx.params)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 754, in invoke
      return __callback(*args, **kwargs)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
      return f(get_current_context(), *args, **kwargs)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/commands/run/command.py", line 134, in cli
      process_env(
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/commands/run/command.py", line 176, in process_env
      result = {"env": name, "duration": time(), "succeeded": ep.process()}
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/commands/run/processor.py", line 67, in process
      result = init_platform(self.options["platform"]).run(
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/platform/_run.py", line 60, in run
      self.autoinstall_runtime_packages()
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/platform/_packages.py", line 74, in autoinstall_runtime_packages
      self.pm.install(self.get_package_spec(name))
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/package/manager/_install.py", line 48, in install
      pkg = self._install(
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/package/manager/_install.py", line 99, in _install
      pkg = self.install_from_registry(spec, search_filters, silent=silent)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/package/manager/_registry.py", line 83, in install_from_registry
      package = self.fetch_registry_package(spec)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/package/manager/_registry.py", line 144, in fetch_registry_package
      result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/clients/registry.py", line 125, in get_package
      return self.fetch_json_data(
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/clients/http.py", line 147, in fetch_json_data
      response = self.send_request(method, path, **kwargs)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/util.py", line 75, in wrapper
      return func(*args, **kwargs)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/platformio/clients/http.py", line 126, in send_request
      return getattr(self._session, method)(path, **kwargs)
    File "/Users/ccui/.platformio/penv/lib/python3.9/site-packages/requests/sessions.py", line 555, in get
      return self.request('GET', url, **kwargs)
[... cut ...]
    https://docs.platformio.org/page/faq.html

  * Report this problem to the developers
    https://github.com/platformio/platformio-core/issues

  ============================================================

  Error: No such file or directory (os error 2)
 
Is it possible to choose a directory, that I have already downloaded the esp-idf from github, or disable the auto download method and choose an already downloaded esp-idf repo locally?

It's very usable, if man could choose the repo of esp-idf...

thank you very much!

ivmarkov, 2021-11-05:
Yes, it is possible, but (currently) only with the native build `cargo build --features native`.
Just do:
* `export ESP_IDF_GLOBAL_INSTALL=1` - this would fall back to the `~/.espressif` folder

Need to document the various flags the build supports...

bss03arg, 2021-11-16:
Hi,

I solve this problem. if anyone has a very slow internet to github.com, should try this method.

1. download the esp-idf repository from github locally and copy the directory to ~/.espressif/esp-idf-v4.3.1. The directory name is very important: esp-idf-v4.3.1, if you use version v4.3.1. Man can't create just a link to some esp-idf directory(shared with other projects), the build process will always create a directory.

2. configure in project dir, .cargo/config.toml
use ESP_IDF_VERSION = { value = "v4.3.1" } to choose a the esp-idf version

3. build the project with: 
export ESP_IDF_GLOBAL_INSTALL=1
cargo build --release --features native

thank you very much!

closed 2021-11-16 by bss03arg