ivmarkov/rust-esp32-std-demo #33: can't find crate for `panic_abort`

kellerkindt, 2021-11-09:
Hi there,

first of all, I really appreciate your work on rust and the esp controllers!
It'll be amazing, once this runs smoothly with the stdlib, thanks *a lot*!

So, I tried to compile  and flash this project onto a ESP32-C3-32S, which is detected via `espflash`
```
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       xx:xx:xx:xx:xx:xx
```

I cloned this project, changed the target in `.cargo/config.toml`to `riscv32imc-esp-espidf` and tried to compile it, but that does not seem to work:
```
$ RUST_ESP32_STD_DEMO_WIFI_SSID=ABC RUST_ESP32_STD_DEMO_WIFI_PASS=DEF cargo +nightly build -Z build-std

...

   Compiling embedded-svc v0.12.0
   Compiling esp-idf-sys v0.24.4
   Compiling esp-idf-hal v0.23.3
   Compiling esp-idf-svc v0.28.1
   Compiling rust-esp32-std-demo v0.20.2 (/tmp/rust-esp32-std-demo)
error[E0463]: can't find crate for `panic_abort`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `rust-esp32-std-demo` due to previous error

```

```
$ rustc --version
rustc 1.56.1 (59eed8a2a 2021-11-01)
$ rustc +nightly --version
rustc 1.58.0-nightly (495322d77 2021-11-08)
```

Sooo... what silly mistake am I doing? :)

ivmarkov, 2021-11-09:
> Sooo... what silly mistake am I doing? :)

`-Z build-std` 

Remove this from the build command line and it should work. `-Z build_std` and a few `build_std_features` are already passed to the build command via `.cargo/config.toml`. Your manual override with `-Z build_std` is overriding the `build_std_features` that need to be passed as well with empty ones, and hence your build fails.

closed 2021-11-09 by ivmarkov

ivmarkov, 2021-11-09:
If you are wondering, `-Zbuild_std` is compiling by default with a `panic_unwind` strategy, which will never be supported on the ESP32. We need `panic_abort` instead. Moreover (for now), we need the `panic_immediate_abort` behavior of the `panic_abort` crate, as without it, `panic_abort` tries to dump a backtrace before panic-ing, which is also not (yet) supported for the esp32 and hence leads to a crash while a crash (panic) is already in progress.

kellerkindt, 2021-11-09:
Uff, thats a silly one. I tried `cargo build` previously but not `cargo +nightly build` :facepalm: 

```
$ RUST_ESP32_STD_DEMO_WIFI_SSID=ABC RUST_ESP32_STD_DEMO_WIFI_PASS=DEF cargo +nightly espflash /dev/ttyUSB0
```

and 
```
$ espmonitor
```

now works as well, nice.
Any ETA for ESP32-C3 projects on stable?

 Thanks!

ivmarkov, 2021-11-09:
> Any ETA for ESP32-C3 projects on stable?

"ETA for ESP32-C3 projects on stable" = "ETA for `cargo -Z build-std` on stable".

Seriously. The only reason we are on nightly is because `-Z build-std` is a nightly-only thing. The STD support for ESP-IDF itself has hit stable since 1.56.

If you have a brilliant idea how to bring functionality similar to `-Z build-std` to stable, I'm all ears. A solution might be shipping somehow prepackaged core/libstd components for our `riscv32imc-esp-espidf` target. Since this target is Tier3, Rust upstream does not ship (nor test & compile)  these.

ivmarkov, 2021-11-09:
By the way, using the Espressif-provided `esp` toolchain is the closest you get to Rust stable, as these are branched off from the same Rust upstream commit that is used to release/branch the corresponding upstream Rust stable release.

kellerkindt, 2021-11-09:
> Seriously. The only reason we are on nightly is because `-Z build-std` is a nightly-only thing. The STD support for ESP-IDF itself has hit stable since 1.56.

Yeah, I saw that release note, which is why I am testing it now :)

> Since this target is Tier3, Rust upstream does not ship (nor test & compile) these.

My naive approach would then be to try to promote it to Tier2? ^^
Tbh, since espressif controllers are quite widespread, one could argue that the riscv version will be relevant soon as well.
But I am not familiar with the promotion process at all, so no idea how laborious this would actually be...