ivmarkov/rust-esp32-std-demo #34: Experimental flag results in bootloop for the c3

pfesenmeier, 2021-11-19:
Hello, and thank you again for making this, I'm having a blast using this.

I thought it might be helpful to give some early feedback on your 'experimental' feature flag

Running this:
```
cargo build --features experimental
```
builds successfully, but for the ESP32-C3-DevKitC-02v1.1 I am seeing this bootloop:

```
I (145) esp_image: segment 1: paddr=0003b208 vaddr=3fc8e200 size=03568h ( 13672) load
I (147) esp_image: segment 2: paddr=0003e778 vaddr=40380000 size=018a0h (  6304) load
I (152) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=d2ba0h (863136) map
I (288) esp_image: segment 4: paddr=00112bc8 vaddr=403818a0 size=0c900h ( 51456) load
I (298) esp_image: segment 5: paddr=0011f4d0 vaddr=50000000 size=00010h (    16) load
E (298) esp_image: invalid segment length 0x36ce
E (301) boot: Factory app partition is not bootable
E (306) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d153e
=> 0x403d153e: ?? ??:0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
=> 0x403ce000: ?? ??:0
load:0x403d0000,len:0x2ce0
=> 0x403d0000: ?? ??:0
entry 0x403ce000
=> 0x403ce000: ?? ??:0
I (35) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (35) boot: compile time 12:10:40
I (35) boot: chip revision: 3
I (38) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (45) boot.esp32c3: SPI Speed      : 80MHz
I (50) boot.esp32c3: SPI Mode       : DIO
I (55) boot.esp32c3: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 003f0000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 3, min. application chip revision: 0
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=2b1e0h (176608) map
I (145) esp_image: segment 1: paddr=0003b208 vaddr=3fc8e200 size=03568h ( 13672) load
I (147) esp_image: segment 2: paddr=0003e778 vaddr=40380000 size=018a0h (  6304) load
I (152) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=d2ba0h (863136) map
I (288) esp_image: segment 4: paddr=00112bc8 vaddr=403818a0 size=0c900h ( 51456) load
I (298) esp_image: segment 5: paddr=0011f4d0 vaddr=50000000 size=00010h (    16) load
E (298) esp_image: invalid segment length 0x36ce
E (301) boot: Factory app partition is not bootable
E (306) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
```

specs:
Ubuntu 18
rustc 1.58.0-nightly (cc946fcd3 2021-11-18)
espflash v1.1.0

ivmarkov, 2021-11-19:
I don't think this has anything to do with `--features experimental`. You are basically hitting this issue: https://github.com/ivmarkov/rust-esp32-std-demo/issues/14

Basically, the issue seems to randomly appear with various sizes if the final .elf executable.
I haven't got time to check what is causing it, unfortunately, but seems to be related to the final ELF executable size somehow.

ivmarkov, 2021-11-19:
You could try rebuilding with `--features experimental,native` to try if this fixes the issue for you, as suggested in this comment: https://github.com/ivmarkov/rust-esp32-std-demo/issues/14#issuecomment-950097593

pfesenmeier, 2021-11-19:
That worked! Thank you!

closed 2021-11-19 by pfesenmeier