ivmarkov/rust-esp32-std-demo #35: Stack overflow when using async with smol

derchr, 2021-11-21:
Hello,
when I try to call the test_tcp_bind_async function I get an stack overflow in task pthread:
```
I (16120) rust_esp32_std_demo: About to bind a simple echo service to port 8081 using async (smol-rs)!
***ERROR*** A stack overflow in task pthread has been detected.
Backtrace:0x400817c6:0x3ffc9558 |<-CORRUPTED
ELF file SHA256: 0000000000000000
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
```
I build with the native feature and ESP_IDF_VERSION is set to master.
Cranking up the main task stack size does not solve the problem.
Thanks for your help!

ivmarkov, 2021-11-26:
Try increasing the stack size of the custom thread here:
```
    thread::spawn(move || {
        smol::block_on(test_tcp_bind()).unwrap();
    });
```
Would be good to know your chipset as well, and whether you are using the nightly rust compiler (for esp32c3), or the `esp` one.

derchr, 2021-11-27:
> Would be good to know your chipset as well, and whether you are using the nightly rust compiler (for esp32c3), or the esp one.

Yeah, I should have added those infos. I'm using the standard ESP32 and run the `esp` rust compiler.

> Try increasing the stack size of the custom thread here

Oh, I did not know that increasing the thread stack size actually increases the stack size of the FreeRTOS task. I have set it to 4096 bytes as a test and it works now!
Thanks you for your help.

ivmarkov, 2021-11-27:
Rust STD is an almost zero cost wrapper around the OS. In this case the OS is FreeRTOS, so yeah, Rust threads are Posix threads, which in turn are FreeRTOS tasks.

I'll update the demo to set the stack size to 4096 as well.

ivmarkov, 2021-12-28:
Fixed in https://github.com/ivmarkov/rust-esp32-std-demo/commit/95ff04ee2bea669423aff0f2b929c42ecd114454

closed 2021-12-28 by ivmarkov