ivmarkov/rust-esp32-std-demo #36: signal:11, SIGSEGV: invalid memory reference

MaoniinoaM, 2021-11-26:
Hi, ivmarkov,
I try to compile this demo, and during compiling crate "esp-idf-sys", it crashs here:
`root@siemens-HP-ProBook-650-G2:/home/siemens/private/esp32C3/rust-esp32-std-demo-main# cargo build --features native
warning: unused config key `unstable.extra-link-arg` in `/home/siemens/private/esp32C3/rust-esp32-std-demo-main/.cargo/config.toml`
warning: unused config key `unstable.configurable-env` in `/home/siemens/private/esp32C3/rust-esp32-std-demo-main/.cargo/config.toml`
   Compiling esp-idf-sys v0.24.6
error: failed to run custom build command for `esp-idf-sys v0.24.6`

Caused by:
  process didn't exit successfully: `/home/siemens/private/esp32C3/rust-esp32-std-demo-main/target/debug/build/esp-idf-sys-eee2f25f0a2da066/build-script-build` (signal: 11, SIGSEGV: invalid memory reference)
`
I just change the target in .cargo/config.toml to riscv.

please help give some hints.

ivmarkov, 2021-11-26:
Are you compiling with the `esp` rust compiler, or with the nightly one? If the nightly one, try the `esp` one or the other way around.

Is this the whole build output? It is unclear whether the rust compiler itself is crashing, or the GCC, or bindgen/clang. What clang are you using?

In any case the crash is in one of these things. Might be related to your operating system as well, but difficult to say, as you have not pointed out your operating system details either.

MaoniinoaM, 2021-11-27:
Firstly, thank you for your hints very much. I use nightly toolchain. 
And I figure out the root cause: because I compile the llvm project in another Ubuntu. And I just take the compiled clang binary to the new Ubuntu, not compile the llvm project in the new ubuntu again, a bit lazy. And according to your hint, I try to compile it and re-build, it passes. 
Thank you again.

ivmarkov, 2021-11-27:
Np! Closing.

closed 2021-11-27 by ivmarkov