ivmarkov/rust-esp32-std-demo #40: SIGABRT: Failed to compile `clap v2.34.0`

thorlucas, 2021-12-02:
Cloning the example and compiling on `macOS Monterey 12.0.1` with the `esp` toolchain fails when building `clap v2.34.0`:

```
error: could not compile `clap`

Caused by:
  process didn't exit successfully: `rustc --crate-name clap --edition=2018 /Users/thorcorreia/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-2.34.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --cfg 'feature="ansi_term"' --cfg 'feature="atty"' --cfg 'feature="color"' --cfg 'feature="default"' --cfg 'feature="strsim"' --cfg 'feature="suggestions"' --cfg 'feature="vec_map"' -C metadata=ac26a8e48a625ff4 -C extra-filename=-ac26a8e48a625ff4 --out-dir /Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps -L dependency=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps --extern ansi_term=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libansi_term-7bb5c8085badc50a.rmeta --extern atty=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libatty-02f4983f38003a7d.rmeta --extern bitflags=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libbitflags-45b4e505e2497911.rmeta --extern strsim=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libstrsim-2b74c9109d8bb938.rmeta --extern textwrap=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libtextwrap-9684b827efe0b904.rmeta --extern unicode_width=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libunicode_width-d208e9921c947bc5.rmeta --extern vec_map=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libvec_map-63037abaf65b04ec.rmeta --cap-lints allow` (signal: 6, SIGABRT: process abort signal)
```

It appears that the `esp` toolchain is unable to compile `clap` for some reason.

I also mirrored the issue [here](https://github.com/esp-rs/rust-build/issues/23).

georgik, 2021-12-02:
This is known limitation of toolchain on Monterey. Build 1.57 is running, it should resolve the issue.

georgik, 2021-12-02:
@thorlucas The issue is still present in 1.57.0.0.
Recommended workaround: use Podman/Docker image https://hub.docker.com/r/espressif/idf-rust on Monterey

Any hint what could be the root-cause is appreciated.

thorlucas, 2021-12-02:
> @thorlucas The issue is still present in 1.57.0.0. Recommended workaround: use Podman/Docker image https://hub.docker.com/r/espressif/idf-rust on Monterey
> 
> Any hint what could be the root-cause is appreciated.

That's going to be a little bit difficult -- I tried running `gdb` on the rustc command but I'm having codesigning problems. I usually switch to linux when debugging but in this case the problem is macOS specific…

As posted in the other thread, the rustc pannic message is:

```
rustc(41908,0x7000083ca000) malloc: *** error for object 0x600002596000: pointer being realloc'd was not allocated
```

So we have some memory issues.

thorlucas, 2021-12-02:
> use Podman/Docker image

Sorry, eps-build says you cannot flash from docker. Any hint onto how I would go about using it? Thanks

georgik, 2021-12-03:
It's possible to flash from Podman which supports mapping of serial devices. Other option would be to use docker cp, to get file out of container to main host and flash it from there.

Meanwhile:

We found the reason of SIGABRT. The build of Rust for macOS x86_64 must contain option --set rust.jemalloc.
The new build is ready. Please test following script https://github.com/esp-rs/rust-build/tree/build/1.57:

./test-rust-toolchain.sh -t 1.57.0.1 --installation-mode reinstall
This will deploy toolchain with name esp-1.57.0.1. You can call your builds with cargo +esp-1.57.0.1 .

franxois, 2021-12-05:
Hi, can you please provide this build for macOS aarch64 apple ?

thorlucas, 2021-12-06:
@georgik Excellent, thanks! I managed to get it flashing by simply building using a shared volume in docker and flashing the `target/` outside of docker. I will test the new builder momentarily

thorlucas, 2021-12-06:
@georgik New build works great! Thanks

closed 2021-12-06 by thorlucas