ivmarkov/rust-esp32-std-demo #41: Compatible PlatformIO Core not found

alex915shen, 2021-12-09:
error: failed to run custom build command for `esp-idf-sys v0.28.2`

Caused by:
  process didn't exit successfully: `/Users/shenxiao/rust-esp32-std-demo/target/release/build/esp-idf-sys-b3c7a44c09fc8e99/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  Installer version: 1.0.2
  Platform: macOS-12.0.1
  Python version: 3.9.8 (main, Nov 10 2021, 03:55:42) 
  [Clang 13.0.0 (clang-1300.0.29.3)]
  Python path: /usr/local/opt/python@3.9/bin/python3.9
  Creating a virtual environment at /Users/shenxiao/rust-esp32-std-demo/.embuild/platformio/penv
  Updating Python package manager (PIP) in a virtual environment
  Virtual environment has been successfully created!
  Installing PlatformIO Core

  --- stderr
  Error: Compatible PlatformIO Core not found.
  Reason: PlatformIO executable not found in `/Users/shenxiao/rust-esp32-std-demo/.embuild/platformio/penv/bin`
  ERROR: Could not install packages due to an OSError: Missing dependencies for SOCKS support.

  Error: Could not install PlatformIO Core: Command '['/Users/shenxiao/rust-esp32-std-demo/.embuild/platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
  Error: Compatible PlatformIO Core not found.
  Reason: PlatformIO executable not found in `/Users/shenxiao/rust-esp32-std-demo/.embuild/platformio/penv/bin`
  Error: EOF while parsing a value at line 1 column 0

ivmarkov, 2021-12-09:
This is not a problem of `esp-idf-sys` or `rust-esp32-std-demo`, but in the (silent) installation of PlatformIO. In other words, if you just try to install PlatformIO CLI [using their official guidelines](https://platformio.org/install/cli), it will probably fail as well, with the same error. That's something you could try btw, to verify or deny the hypothesis below.

The best way to solve this is by googling the error messages that the PlatformIO installer outputs, as I did.
It *seems* this is related to you using a SOCKS proxy and/or somehow having socks proxy settings, while the Python socks proxy support is not installed. Here are some links to get you started:
* https://stackoverflow.com/questions/38794015/pythons-requests-missing-dependencies-for-socks-support-when-using-socks5-fro
* https://github.com/platformio/platformio-vscode-ide/issues/1255
* https://github.com/platformio/platformio-vscode-ide/issues/773

alex915shen, 2021-12-10:
step1：

> open ~/.bash_profile
> set `alias pip="pip3.9"`
> source ~/.bash_profile

step2:

> $ pip install --upgrade pip
> $ pip --version
> pip 21.3.1 from /usr/local/lib/python3.9/site-packages/pip (python 3.9)

step3:

> $ pip install -U platformio

step4:

> pip install pysocks

$ ALL_PROXY=socks5://127.0.0.1:1086 cargo build --release
**but still the same error.**

> error: failed to run custom build command for `esp-idf-sys v0.28.2`
> 
> Caused by:
>   process didn't exit successfully: `/Users/shenxiao/rust-esp32-std-demo/target/release/build/esp-idf-sys-b3c7a44c09fc8e99/build-script-build` (exit status: 1)
> --- stderr
>   Error: Compatible PlatformIO Core not found.
>   Reason: PlatformIO executable not found in `/Users/shenxiao/rust-esp32-std-demo/.embuild/platformio/penv/bin`
>   ERROR: Could not install packages due to an OSError: Missing dependencies for SOCKS support.
>

ivmarkov, 2021-12-10:
I'm not an expert in Python / PlatformIO, so not sure what exactly in your steps is wrong.

However: can you try installing the PlatformIO CLI as I suggested above? This way we know whether the error you are seeing indeed comes from stock PlatformIO? Also, if so, it would be easier to fix the problem first for PlatformIO CLI, and then go to esp-idf-sys / Rust again.

closed 2021-12-13 by alex915shen