ivmarkov/rust-esp32-std-demo #42: Bootloop after uploading to esp32-c3.

JakobLachermeier, 2021-12-10:
Hi, i'm getting a bootloop after building for the c3 and flashing with espflash.

I have installed rust-nightly and clang version 13.

Here are the steps i used to build and flash.
```
rustup install nightly
rustup default nightly
cargo build --target riscv32imc-esp-espidf
espflash /dev/ttyUSB0 target/riscv32imc-esp-espidf/debug/rust-esp32-std-demo
espmonitor /dev/ttyUSB0

```
Any tips on how to troubleshoot are appreciated.
This is the espmonitor output.

> ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40384396
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (35) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (35) boot: compile time 12:10:40
I (35) boot: chip revision: 3
I (38) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (45) boot.esp32c3: SPI Speed      : 80MHz
I (50) boot.esp32c3: SPI Mode       : DIO
I (55) boot.esp32c3: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 003f0000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 3, min. application chip revision: 0
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=239a0h (145824) map
I (140) esp_image: segment 1: paddr=000339c8 vaddr=3fc8e800 size=03378h ( 13176) load
I (143) esp_image: segment 2: paddr=00036d48 vaddr=40380000 size=092d0h ( 37584) load
I (153) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=ab978h (702840) map
I (260) esp_image: segment 4: paddr=000eb9a0 vaddr=403892d0 size=05334h ( 21300) load
I (264) esp_image: segment 5: paddr=000f0cdc vaddr=50000000 size=00010h (    16) load
I (270) boot: Loaded app from partition at offset 0x10000
I (272) boot: Disabling RNG early entropy source...
I (288) cpu_start: Pro cpu up.
I (301) cpu_start: Pro cpu start user code
I (301) cpu_start: cpu freq: 160000000
I (301) cpu_start: Application information:
I (304) cpu_start: Project name:     esp-idf
I (308) cpu_start: App version:      0d085d7-dirty
I (314) cpu_start: Compile time:     Dec 10 2021 18:08:35
I (320) cpu_start: ELF file SHA256:  0000000000000000...
I (326) cpu_start: ESP-IDF:          4.3.1
I (331) heap_init: Initializing. RAM available for dynamic allocation:
I (338) heap_init: At 3FC95CA0 len 0002A360 (168 KiB): DRAM
I (344) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (351) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (358) spi_flash: detected chip: generic
I (362) spi_flash: flash io: dio
E (366) spi_flash: Detected size(2048k) smaller than the size in the binary image header(4096k). Probe failed.
assertion "flash_ret == ESP_OK" failed: file "/home/jakob/temp/rust-esp32-std-demo/.embuild/platformio/packages/framework-espidf/components/esp_system/startup.c", line 329, function: do_core_init
abort() was called at PC 0x42090f5d on core 0
Core  0 register dump:
MEPC    : 0x403858a2  RA      : 0x40386018  SP      : 0x3fcde230  GP      : 0x3fc8f000  
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x3fcde294  A0      : 0x3fcde25c  A1      : 0x3fcde292  
A2      : 0x00000000  A3      : 0x3fcde289  A4      : 0x00000001  A5      : 0x3fc96000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3c0c9000  S3      : 0x3fc93038  
S4      : 0x42000020  S5      : 0x00040020  S6      : 0x00000000  S7      : 0x3fcd7000  
S8      : 0x3c0b0020  S9      : 0x00010020  S10     : 0x00000006  S11     : 0x3fcde3cc  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  
Stack memory:
3fcde230: 0x00000006 0x00010020 0x3fcde290 0x4038cd96 0x00000000 0x00040020 0x42000020 0x3fc9113c
3fcde250: 0x3fcde294 0x3fc91158 0x3fcde290 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fcde270: 0x43502074 0x34783020 0x30393032 0x20643566 0x63206e6f 0x2065726f 0x00000030 0x42090000
3fcde290: 0x00000030 0x39303234 0x64356630 0x3c0bcd00 0x00000149 0x3c0bd000 0x3c0bcdd4 0x42090f60
3fcde2b0: 0x00000000 0x3c0bce48 0x3c0bcd84 0x00000149 0x3c0c9000 0x3c0bd000 0x3fc93038 0x420293ea
3fcde2d0: 0xffff0000 0x00000146 0x3c0bcbc4 0x403802ec 0xffff0000 0x3c0b0020 0x3c0bd000 0x4202951e
3fcde2f0: 0xffff0000 0x30303030 0x30303030 0x30303030 0x30303030 0x000bb900 0x0000000c 0x420a36c6
3fcde310: 0xffff0000 0x000bb977 0x0000000c 0x40380434 0x00000000 0x000ab978 0x000bb997 0x403d0f1a
3fcde330: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcdf000
3fcde350: 0x00004d64 0x3ff1c14c 0x0000ffff 0xffffffff 0xffffffff 0x3fcde4b0 0xffffffff 0x403d15aa
3fcde370: 0x00000000 0x0000000a 0x00010000 0x003f0000 0x00010000 0x100206e9 0x403802ec 0x000000ee
3fcde390: 0x00000005 0x00000000 0x01000000 0x3c0b0020 0x000239a0 0x3fc8e800 0x00003378 0x40380000
3fcde3b0: 0x000092d0 0x42000020 0x000ab978 0x403892d0 0x00005334 0x50000000 0x00000010 0x00000000
3fcde3d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde410: 0x00000000 0x00000000 0x00000000 0x00010020 0x000339c8 0x00036d48 0x00040020 0x000eb9a0
3fcde430: 0x000f0cdc 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde450: 0x00000000 0x00000000 0x00000000 0x000e0d10 0x6aea34f5 0xe1e1cd0b 0x72e40372 0x33cdf4c7
3fcde470: 0x938e4d9d 0xe6fe6ced 0x1ef0ccf1 0x83292d06 0x00004d64 0x3ff1c14c 0x0000ffff 0x3fce0000
3fcde490: 0x00000000 0x000000e4 0x00004d70 0x403ce07c 0x4cbb5b54 0x712931aa 0x00000000 0xffffffff
3fcde4b0: 0x00000000 0x00000000 0x00010000 0x003f0000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde530: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde550: 0x600c5200 0x600c5200 0x8082dbb8 0x4004a296 0xba1481c2 0xec8c6203 0x3fcde668 0x00000000
3fcde570: 0x34333936 0x00000000 0x00010000 0x2f0203e9 0x403ce000 0x403d0000 0x00002ce0 0x000000ee
3fcde590: 0x00000005 0x00000000 0x01000000 0x3c004d70 0x7533885e 0xcedd6070 0x3b6231b0 0x144aa432
3fcde5b0: 0xc32e18bd 0xf4abe6ff 0xbbaffc5d 0xf3047dd7 0x4c97a429 0x00000000 0x00000000 0x00000000
3fcde5d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcde700 0x00000101 0x00000002
3fcde5f0: 0xcedd6070 0x3b6231b0 0x144aa432 0xc32e18bd 0xf4abe6ff 0xbbaffc5d 0xf3047dd7 0x4c97a429
3fcde610: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
ELF file SHA256: 0000000000000000
Rebooting...

ivmarkov, 2021-12-10:
How much flash does your esp32c3 board have?

This looks suspicious:
E (366) spi_flash: Detected size(2048k) smaller than the size in the binary image header(4096k). Probe failed.

JakobLachermeier, 2021-12-12:
Seems i have bought the wrong devboard.

OlivierLemoine, 2021-12-12:
Hi, I actually have the same problem. My ESP32C3 has 2048k flash. I find it strange that a simple release build weight more than 8Mb.

ivmarkov, 2021-12-13:
> Hi, I actually have the same problem. My ESP32C3 has 2048k flash. I find it strange that a simple release build weight more than 8Mb.

What is actually flashed is neither 8mb nor 4mb (4096k), but much smaller. Only the code portions of the elf file is being flashed.

The error reported above is likely related to the 2nd stage bootloader being for 4mb and not for 2mb and is not related to the binary ary being flashed.

ivmarkov, 2021-12-13:
@jessebraham what do you think? ^^^

javiser, 2022-01-14:
I had the same problem as described by @JakobLachermeier. I guess you are using the [ESP-C3-32S-Kit](https://www.waveshare.com/esp-c3-32s-kit.htm) right? I solved it by using the solution proposed in [this issue](https://github.com/fkohlgrueber/esp32c3-idf-led-example/issues/1) from https://github.com/fkohlgrueber/esp32c3-idf-led-example. What I basically did:
- I followed the instructions in https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/index.html#get-started-get-esp-idf to install ESP-IDF in my machine, including building the "hello world" example in step 5. By doing that you get a `bootloader.bin` file which I will need later.
- I also need a partition table. I used this one, but how to get it (it is board specific I guess) was a little confusing for me and is described in the issue anyway:
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        1M,
```
- Then I build the example and finally flash it using additional arguments for the bootloader and the partition table: `espflash /dev/ttyUSB0 target/riscv32imc-esp-espidf/debug/rust-esp32-std-demo --bootloader bootloader.bin --partition-table partitions.csv` (probably the new files should be in a separate folder named `resources` or similar)

Side note: you can speed up the process of flashing the board by adding `--speed 460800` to the above command. You get a warning but the process goes from 1 minute to 25 seconds... I observed that this is the baud rate used automatically in the hello world example.

zhongwei, 2022-02-05:
I have the same problem. I solved this problem by unplugging and reinserting the USB.

sgrimee, 2022-03-10:
I have the same problem with a Waveshare ESP-C3-32S-Kit. I have tried re-instering the USB, and also @javiser method to build the bootloader from idf hello world project (that flashes fine with idf.py) but could not get it to work either.

```
rust-esp32-std-demo ➜  cargo espflash --release --monitor
Serial port: /dev/tty.usbserial-210
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        2MB
Features:          WiFi
MAC address:       7c:df:a1:b9:3c:f0
    Finished release [optimized] target(s) in 0.58s
[00:00:01] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:01:05] ########################################     560/560     segment 0x10000

Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�
 �ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (30) boot: compile time 12:10:40
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (44) boot.esp32c3: SPI Speed      : 80MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=27e10h (163344) map
I (138) esp_image: segment 1: paddr=00037e38 vaddr=3fc8f200 size=03528h ( 13608) load
I (140) esp_image: segment 2: paddr=0003b368 vaddr=40380000 size=04cb0h ( 19632) load
I (147) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=c7578h (816504) map
I (274) esp_image: segment 4: paddr=001075a0 vaddr=40384cb0 size=0a3e8h ( 41960) load
I (282) esp_image: segment 5: paddr=00111990 vaddr=50000000 size=00010h (    16) load
I (287) boot: Loaded app from partition at offset 0x10000
I (287) boot: Disabling RNG early entropy source...
I (303) cpu_start: Pro cpu up.
I (315) cpu_start: Pro cpu start user code
I (316) cpu_start: cpu freq: 160000000
I (316) cpu_start: Application information:
I (318) cpu_start: Project name:     esp-idf
I (323) cpu_start: App version:      b5c88c1-dirty
I (329) cpu_start: Compile time:     Mar 10 2022 09:45:52
I (335) cpu_start: ELF file SHA256:  0000000000000000...
I (341) cpu_start: ESP-IDF:          4.3.2
I (346) heap_init: Initializing. RAM available for dynamic allocation:
I (353) heap_init: At 3FC96960 len 000296A0 (165 KiB): DRAM
I (359) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (366) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (372) spi_flash: detected chip: generic
I (377) spi_flash: flash io: dio
[... 28 lines cut ...]
3fcde320: 0x00000000 0x000c7578 0x000d7597 0x403d0f1a 0x00000000 0x00000000 0x00000000 0x00000000
3fcde340: 0x00000000 0x00000000 0x00000000 0x3fcdf000 0x00004d64 0x3ff1c14c 0x0000ffff 0xffffffff
3fcde360: 0xffffffff 0x3fcde4b0 0xffffffff 0x403d15aa 0x00000000 0x0000000a 0x00010000 0x003f0000
3fcde380: 0x00010000 0x100206e9 0x403809dc 0x000000ee 0x00000005 0x00000000 0x01000000 0x3c0d0020
3fcde3a0: 0x00027e10 0x3fc8f200 0x00003528 0x40380000 0x00004cb0 0x42000020 0x000c7578 0x40384cb0
3fcde3c0: 0x0000a3e8 0x50000000 0x00000010 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00010020
3fcde420: 0x00037e38 0x0003b368 0x00040020 0x001075a0 0x00111990 0x00000000 0x00000000 0x00000000
3fcde440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x001019d0
3fcde460: 0xb58ffc8c 0x06a152d2 0x66bbbe35 0xec6d55b4 0xc0b1ffb1 0x718b6671 0xaabc4e3d 0x41968f65
3fcde480: 0x00004d64 0x3ff1c14c 0x0000ffff 0x3fce0000 0x00000000 0x000000e4 0x00004d70 0x403ce07c
3fcde4a0: 0x8e4d3993 0xd8208221 0x00000000 0xffffffff 0x00000000 0x00000000 0x00010000 0x003f0000
3fcde4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde540: 0x00000000 0x00000000 0x00000000 0x00000000 0x600c5200 0x600c5200 0x8082dbb8 0x4004a296
3fcde560: 0xb294a2fc 0x20098570 0x3fcde668 0x00000000 0xffffff63 0x00000000 0x00010000 0x2f0203e9
3fcde580: 0x403ce000 0x403d0000 0x00002ce0 0x000000ee 0x00000005 0x00000000 0x01000000 0x3c004d70

ELF file SHA256: 0000000000000000

Rebooting...
```

ivmarkov, 2022-03-10:
The moment you call this: `cargo espflash --release --monitor` everything you've done to workaround the problem evaporates and you are back at square 0, as you've just re-flashed a standard 4mb bootloader and a standard partition table...

ivmarkov, 2023-10-19:
This had been resolved in the meantime.

closed 2023-10-19 by ivmarkov