ivmarkov/rust-esp32-std-demo #46: build error on windows 10

jjpark78, 2021-12-22:
```
 ...
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_aead\chacha20poly1305\sodium\aead_chacha20poly1305.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_aead\xchacha20poly1305\sodium\aead_xchacha20poly1305.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_auth\crypto_auth.o
  Archiving .pio\build\debug\esp-idf\jsmn\libjsmn.a
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_auth\hmacsha256\auth_hmacsha256.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_auth\hmacsha512\auth_hmacsha512.o
  Indexing .pio\build\debug\esp-idf\jsmn\libjsmn.a
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_auth\hmacsha512256\auth_hmacsha512256.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\crypto_box.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\crypto_box_easy.o
  Archiving .pio\build\debug\esp-idf\hal\libhal.a
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\crypto_box_seal.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\curve25519xchacha20poly1305\box_curve25519xchacha20poly1305.o
  Indexing .pio\build\debug\esp-idf\hal\libhal.a
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\curve25519xchacha20poly1305\box_seal_curve25519xchacha20poly1305.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\curve25519xsalsa20poly1305\box_curve25519xsalsa20poly1305.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\ed25519\core_ed25519.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\ed25519\core_ristretto255.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\ed25519\ref10\ed25519_ref10.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\hchacha20\core_hchacha20.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\hsalsa20\core_hsalsa20.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\hsalsa20\ref2\core_hsalsa20_ref2.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_core\salsa\ref\core_salsa_ref.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_generichash\blake2b\generichash_blake2.o
  Compiling .pio\build\debug\libsodium\libsodium\src\libsodium\crypto_generichash\blake2b\ref\blake2b-compress-avx2.o

  --- stderr
  Assembler messages:
  Fatal error: can't create .pio/build/debug/libsodium/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.o: No such file or directory
  Assembler messages:
  Fatal error: can't create .pio/build/debug/libsodium/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_curve25519xchacha20poly1305.o: No such file or directory
  *** [.pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\curve25519xchacha20poly1305\box_seal_curve25519xchacha20poly1305.o] Error 1
  *** [.pio\build\debug\libsodium\libsodium\src\libsodium\crypto_box\curve25519xchacha20poly1305\box_curve25519xchacha20poly1305.o] Error 1
  ========================= [FAILED] Took 91.26 seconds =========================
  Environment    Status    Duration
  -------------  --------  ------------
  debug          FAILED    00:01:31.258
  ==================== 1 failed, 0 succeeded in 00:01:31.258 ====================
  ```

i cannot build this example on windows

ivmarkov, 2021-12-22:
Clone the repo as close to the drive root as possible and then try again, I.e. `c:\rust-esp32-std-demo` and then try again.

georgik, 2021-12-22:
@jjpark78 This is one of limitations PATH length on Windows.
More details: https://github.com/esp-rs/rust/discussions/88#discussioncomment-1581167

Failed libsodium build indicates problem with long path which is not handled correctly by CCache. Option is to disable CCache for the build, but you'll hit another long path problem during link time. https://github.com/espressif/idf-installer/issues/103

Please use very short path on Windows or use WSL2 to build the application on Linux.

ivmarkov, 2021-12-25:
@jjpark78 Did the workaround suggested above work for you?

ivmarkov, 2022-01-29:
Closing due to inactivity.

closed 2022-01-29 by ivmarkov