ivmarkov/rust-esp32-std-demo #47: Linker error when building for ESP32-S2

ryanc-me, 2021-12-24:
hello! i'm getting a linker error when building for the ESP32-S2 variant.

the error is very similar to #24, except it's complaining about `undefined reference to '__sync_lock_test_and_set_4'`

full error:
```
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: "ldproxy" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.17cr2e1tk2gyufs7.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.1aknpudcas07lwc6.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.1bj0sql7mmysp4de.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.1q0ojaklqqblxjuf.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.1r9ru3ef2465pvpx.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.23acug9zg82x7bl5.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2axqel57hdn7uj9c.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2c8jp6f41du9qdzi.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2f1hi8ovuikd28ng.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2ovug2lkaj25o4bk.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2paz3i5ba9js58f0.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2qktm9zbg0tmnxq7.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2x5efin4zphwilfo.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.2xhhbgpr6axugxb6.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.30ich6uw49x9t1x6.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3covqqhl7fmhwcgv.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3dr3m7kw2xywy3cs.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3hl8m4k9wuzbolxa.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3rvvmfm99ortokqz.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3tixuhkehb1clp33.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3us1alpgkrx04shw.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3wjcb6vix0rnmgqg.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.3ze5scq96sfupnw8.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4cgnub8dklrqm15w.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4d8g9askh2a4lv3q.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4eg9cn7xsp35kd7.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4fx3n3jlwph6my0a.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4fydkrzen7lljvxm.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4httt2x9g2e8d9gf.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4jrle0b8u8uk9kyw.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4kotpi782xkj1fi3.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4mpguesbrc19sy0l.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4osqz3gxc66sz3vh.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.4trm4mg256jfubm6.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.55oivwcs1ocif15g.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.5c6kdi5tqd0xm4k0.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.5cfbmykbkfxpy6i0.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.9jbx8a8pps2avph.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.hy5warvoxwudqix.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.i8vibpv0jwxa519.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.u7stq80i1yhk8q8.rcgu.o" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_demo-3ffb2665e5b3cce5.1y4h1p3yhsfi1qqh.rcgu.o" "-Wl,--as-needed" "-L" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps" "-L" "/opt/rust-projects/rust-esp32-std-demo/target/debug/deps" "-L" "/root/.rustup/toolchains/esp/lib/rustlib/xtensa-esp32s2-espidf/lib" "-Wl,-Bstatic" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libst7789-19f2bfd7723c973f.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libheapless-4dc57c360c0082a0.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libstable_deref_trait-7e55de1c60254032.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libhash32-8a7a7c1275e7625c.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libssd1306-8ed83883a9f71d52.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libdisplay_interface_i2c-ab7b5c395cc8072f.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libili9341-00927b2092c2c029.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libembedded_graphics-8b0b997827a870e8.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libbyteorder-0f3b1eb6be6dc405.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libfloat_cmp-bd674329cf6c4a90.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libnum_traits-65f9ff03cf0ede60.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libmicromath-a3ce339df72f17df.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libembedded_graphics_core-91d906bc7d7fa5ab.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libaz-57ef482c00efda35.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libdisplay_interface_spi-e96ea413f0398628.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libbyte_slice_cast-4d4c2f684b9987b8.rlib" "/opt/rust-projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libdisplay_interfa [... cut ...]
 * installed `esp-idf` via https://docs.espressif.com/projects/esp-idf/en/latest/esp32/
 * installed `esp-rs/rust` via https://github.com/esp-rs/rust-build
 * trying to build the vanilla version of this project - no changes

for what it's worth, i'm getting the same error when using `cargo build` on the [`espressif/idf-rust-examples`](https://github.com/esp-rs/rust-build#rust-with-podman-or-docker) docker image also (which, again, is using Ubuntu 20.04 as a base). maybe some library is missing in the minimal Ubuntu docker base?

thanks!

ivmarkov, 2021-12-25:
As strange as it sounds, it is a dupe of #45, which I'm about to fix soon (certain patches that have to be applied to ESP-IDF 3.4.1 are actually not applied; one of these is support for S2 atomics that brings this function that you see the linker complaining about).

closed 2021-12-25 by ivmarkov

ryanc-me, 2021-12-26:
thanks Ivan! I did see your work & PRs over at ESP-IDF.

for what it's worth, the [esp-idf-template](https://github.com/esp-rs/esp-idf-template) _does_ work, if you choose to install ESP-IDF `master` branch. I guess your fixes are included in master. I couldn't get rust-esp32-std-demo working though, even if I uncommented `ESP_IDF_VERSION = { value = "master" }` in the Cargo config.

isw there another way I can force embuild to install ESP-IDF from master?

ivmarkov, 2021-12-26:
> thanks Ivan! I did see your work & PRs over at ESP-IDF.
> 
> for what it's worth, the [esp-idf-template](https://github.com/esp-rs/esp-idf-template) _does_ work, if you choose to install ESP-IDF `master` branch. 

But even then you have to build with `cargo build --features native`. See below.

> I guess your fixes are included in master.

No. The fixes are related to how you *build* the thing. If you use the default `pio` builder (as in `cargo build --features pio` or `cargo build`), then you are affected because the PlatformIO builder is used, and that one cannot apply the patches which are necessary for V4.3.1 (the only branch supported by PlatformIO).

>  I couldn't get rust-esp32-std-demo working though, even if I uncommented `ESP_IDF_VERSION = { value = "master" }` in the Cargo config.

Read [this](https://github.com/esp-rs/esp-idf-sys). `ESP_IDF_VERSION` is only honored by the non-default native builder (`cargo build --features native`). 
 
> isw there another way I can force embuild to install ESP-IDF from master?

See above. Branches other than v4.3.1 are only supported by the native builder.

ryanc-me, 2021-12-26:
Right, that makes sense! I've done a bit more reading and am more up to speed now. Sorry to bother you with simple questions.

For what it's worth though, `esp-idf-template` was working _without_ the `--features native` flag. Maybe it was an envvar or cached crate. But anyway, this project is compiling nicely using the native builder and ESP-IDF `master` branch. Thanks!