ivmarkov/rust-esp32-std-demo #51: Cargo build failed to parse the `edition` key

simon0356, 2022-01-13:
Hello

i'm trying to build the demo project, 
I got the following error when i running "cargo build"
```
warning: Patch `polling v2.1.0 (https://github.com/esp-rs-compat/polling#260dded7)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
error: failed to download `esp-idf-sys v0.29.1`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `/root/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.29.1/Cargo.toml`

Caused by:
  failed to parse the `edition` key

Caused by:
  supported edition values are `2015` or `2018`, but `2021` is unknown
```

I'm not a rust and cargo guru but i thook it maybe came from the following change here : https://github.com/esp-rs/esp-idf-sys/commit/acb86d2b47c33a52e059cb5a0de02159532f9957#diff-2e9d962a08321605940b5a657135052fbcef87b5e360662bb527c96d9a615542

Edition of the cargo.toml change from 2018 to 2021 

Can you help me ? 

Thanks 
Simon

ivmarkov, 2022-01-13:
You have to use latest rust nightly (works for esp32c3 only) or the Rust toolchain provided by Espressif. The demo readme contains information how to do both.

ivmarkov, 2022-01-14:
@simon0356 Did you try with rust compiler as explained above ^^^?

simon0356, 2022-01-14:
Yes it compile like a charm when i follow your instruction.
I have not flashed the ESP yet.

Sorry, i was following instruction from a deprecated article  (https://kerkour.com/rust-on-esp32/ and https://kerkour.com/compile-rust-for-esp32-xtensa-on-raspberry-pi-aarch64/ )

Thanks :)

closed 2022-01-14 by ivmarkov