ivmarkov/rust-esp32-std-demo #53: Build issue on macOS Monterey 12.1

JeanBlignaut, 2022-01-27:
export ESP_IDF_VERSION=master; cargo build --features native                              
   Compiling esp-idf-sys v0.30.1
error: failed to run custom build command for `esp-idf-sys v0.30.1`

Caused by:
  process didn't exit successfully: `/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/target/debug/build/esp-idf-sys-0a3cbb5d657b21a9/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools:$PATH
  Installing ninja@1.10.2
  Downloading ninja-1.10.2-osx.tar.gz to /Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/dist/ninja-1.10.2-osx.tar.gz.tmp
  Downloading ninja-1.10.2-osx.tar.gz to /Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/dist/ninja-1.10.2-osx.tar.gz.tmp
  Downloading ninja-1.10.2-osx.tar.gz to /Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/dist/ninja-1.10.2-osx.tar.gz.tmp

  --- stderr
  fatal: No names found, cannot describe anything.
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
  WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-osx.tar.gz to /Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/dist/ninja-1.10.2-osx.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
  WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-osx.tar.gz to /Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/dist/ninja-1.10.2-osx.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1123)>
  WARNING: Failed to download https://dl.espressif.com/dl/ninja-1.10.2-osx.tar.gz to /Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/dist/ninja-1.10.2-osx.tar.gz.tmp
  ERROR: Failed to download, and retry count has expired
  Traceback (most recent call last):
    File "/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools/idf_tools.py", line 1989, in <module>
      main(sys.argv[1:])
    File "/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools/idf_tools.py", line 1981, in main
      action_func(args)
    File "/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools/idf_tools.py", line 1489, in action_install
      tool_obj.download(tool_version)
    File "/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools/idf_tools.py", line 721, in download
      raise DownloadError()
  __main__.DownloadError
  Error: Could not install esp-idf

  Caused by:
      command '"/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env/bin/python" "/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools/idf_tools.py" "--idf-path" "/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master" "install" "ninja" "xtensa-esp32-elf" "esp32ulp-elf"' exited with non-zero status code 1

JeanBlignaut, 2022-01-27:
After looking around for a bit I came across this: https://github.com/espressif/esp-idf/issues/4775

Following these steps and downloading missing files manualy and copying them to  .embuild/espressif/dist/ as per the errors - I now get the following errors:

export ESP_IDF_VERSION=master; cargo build --features native               
   Compiling esp-idf-sys v0.30.1
error: failed to run custom build command for `esp-idf-sys v0.30.1`

Caused by:
  process didn't exit successfully: `/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/target/debug/build/esp-idf-sys-0a3cbb5d657b21a9/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools:$PATH
  Skipping ninja@1.10.2 (already installed)
  Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
  Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools:$PATH
  Skipping cmake@3.20.3 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/.embuild/espressif/esp-idf-master/tools:$PATH
  cargo:rerun-if-changed=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/oneiric/projects-rust-xtensa/rust-esp32-std-demo/sdkconfig.defaults.esp32s2

  --- stderr
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  CMake Error at /var/folders/_b/rgq318p92w965jrf_d_zprdh0000gn/T/.tmpAYqrs4:1 (include):
    include could not find requested file:

      /tools/cmake/utilities.cmake

  Error: command '"cmake" "-P" "/var/folders/_b/rgq318p92w965jrf_d_zprdh0000gn/T/.tmpAYqrs4"' exited with non-zero status code 1

  Caused by:
      CMake Error at /var/folders/_b/rgq318p92w965jrf_d_zprdh0000gn/T/.tmpAYqrs4:1 (include):
        include could not find requested file:
      
          /tools/cmake/utilities.cmake

JeanBlignaut, 2022-01-27:
ps. this is my first real attempt at building with this tool chain so I'm not too sure what I might be missing...

ivmarkov, 2022-01-28:
Building against ESP-IDF master - especially at the moment, when they do all sorts of breaking changes - is a risky business.

I just fixed the above issue, and in a matter of < 2 hours, another did pop up, due to another commit of theirs.

Please wait until #56 is fixed, and then please build against V4.3.2 (default for PIO and native build), or against v4.4 (just released, requires native build).

ivmarkov, 2022-01-29:
Building against ESP-IDF master (why would you want to do that anyway, this is a moving target?) works now. Issue a `cargo update` to update the `esp-idf-sys` dependency to v0.30.3 then `cargo clean; cargo build --features native` and it should be OK.

closed 2022-01-29 by ivmarkov

JeanBlignaut, 2022-01-31:
I tried 4.3.2 and 4.4 they also had issues