ivmarkov/rust-esp32-std-demo #54: Fail to build for esp32s2, Ubuntu 21.10

sumhoy, 2022-01-28:
Good day.
i tried to build esp32s2 but failed, however esp32 build was successful.

#target = "xtensa-esp32-espidf"      <-- build successfully
#target = "xtensa-esp32s2-espidf"  <-- fail to run cargo build

Ran command below and got esp-idf-hal-0.32.4 compilation error.
>> cargo build -v 

Errors : 
error[E0412]: cannot find type `T` in this scope
   --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/ulp.rs:284:46
    |
284 |     pub unsafe fn write_var(&self, dst: *mut T, value: T) -> Result<(), EspError> {
    |                                              ^ not found in this scope

error[E0412]: cannot find type `T` in this scope
   --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/ulp.rs:284:56
    |
284 |     pub unsafe fn write_var(&self, dst: *mut T, value: T) -> Result<(), EspError> {
    |                                                        ^ not found in this scope

error[E0412]: cannot find type `T` in this scope
   --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/ulp.rs:292:49
    |
292 |     pub unsafe fn swap_var(&mut self, ptr: *mut T, value: T) -> Result<T, EspError> {
    |                                                 ^ not found in this scope

error[E0412]: cannot find type `T` in this scope
   --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/ulp.rs:292:59
    |
292 |     pub unsafe fn swap_var(&mut self, ptr: *mut T, value: T) -> Result<T, EspError> {
    |                                                           ^ not found in this scope

error[E0412]: cannot find type `T` in this scope
   --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/ulp.rs:292:72
    |
265 | impl ULP {
    |     - help: you might be missing a type parameter: `<T>`
...
292 |     pub unsafe fn swap_var(&mut self, ptr: *mut T, value: T) -> Result<T, EspError> {
    |                                                                        ^ not found in this scope

error[E0308]: mismatched types
   --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/ulp.rs:108:33
    |
108 |     const TIMER_REG: *mut u32 = RTC_CNTL_ULP_CP_TIMER_REG;
    |                                 ^^^^^^^^^^^^^^^^^^^^^^^^^ expected *-ptr, found `u32`
    |
    = note: expected raw pointer `*mut u32`
                      found type `u32`

Some errors have detailed explanations: E0308, E0412.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `esp-idf-hal` due to 6 previous errors

Caused by:
  process didn't exit successfully: `rustc --crate-name esp_idf_hal --edition=2018 /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.4/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=z -C embed-bitcode=no -C debuginfo=2 -C debug-assertions=on --cfg 'feature="default"' --cfg 'feature="embedded-svc"' --cfg 'feature="embedded-svc-mutex"' --cfg 'feature="esp-idf-sys"' --cfg 'feature="std"' -C metadata=b9c1c774e622c597 -C extra-filename=-b9c1c774e622c597 --out-dir /home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps --target xtensa-esp32s2-espidf -C linker=ldproxy -L dependency=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps -L dependency=/home/sumhoy/git/rust-esp32-std-demo/target/debug/deps --extern 'noprelude:alloc=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/liballoc-d3bd9574fb98762c.rmeta' --extern 'noprelude:compiler_builtins=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libcompiler_builtins-e9c7863a56d27070.rmeta' --extern 'noprelude:core=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libcore-7b520c3c1bd1e795.rmeta' --extern embedded_hal_0_2=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libembedded_hal-4f7977672458e652.rmeta --extern embedded_hal=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libembedded_hal-987ef8b0ef4413c5.rmeta --extern embedded_svc=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libembedded_svc-1051e55eca01894f.rmeta --extern esp_idf_sys=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libesp_idf_sys-4409fa89559930e7.rmeta --extern nb=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libnb-27839f3857a2c3f7.rmeta --extern 'noprelude:panic_abort=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libpanic_abort-20d28edbda8e5014.rmeta' --extern 'noprelude:panic_unwind=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libpanic_unwind-3f21ce81f2ce614e.rmeta' --extern 'noprelude:proc_macro=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libproc_macro-b0e2e1fa8bbfa545.rmeta' --extern 'noprelude:std=/home/sumhoy/git/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/deps/libstd-6ca87025b32ca845.rmeta' -Z unstable-options --cap-lints allow --cfg esp_idf_idf_cmake --cfg esp_idf_idf_target_arch_xtensa --cfg 'esp_idf_idf_target="esp32s2"' --cfg esp_idf_idf_target_esp32s2 --cfg esp_idf_app_build_type_app_2ndboot --cfg esp_idf_app_build_generate_binaries --cfg esp_idf_app_build_bootloader --cfg esp_idf_app_build_use_flash_sections --cfg esp_idf_app_compile_time_date --cfg esp_idf_bootloader_compiler_optimization_size --cfg esp_idf_bootloader_log_level_info --cfg esp_idf_bootloader_vddsdio_boost_1_9v --cfg esp_idf_bootloader_wdt_enable --cfg esp_idf_secure_boot_supports_rsa --cfg esp_idf_secure_target_has_secure_rom_dl_mode --cfg esp_idf_esptoolpy_flashmode_dio --cfg esp_idf_esptoolpy_flashfreq_80m --cfg esp_idf_esptoolpy_flashsize_2mb --cfg esp_idf_esptoolpy_flashsize_detect --cfg esp_idf_esptoolpy_before_reset --cfg esp_idf_esptoolpy_after_reset --cfg esp_idf_esptoolpy_monitor_baud_115200b --cfg esp_idf_partition_table_single_app --cfg esp_idf_partition_table_md5 --cfg esp_idf_compiler_optimization_default --cfg esp_idf_compiler_optimization_assertions_enable --cfg esp_idf_compiler_stack_check_mode_none --cfg esp_idf_apptrace_dest_none --cfg esp_idf_apptrace_lock_enable --cfg esp_idf_bt_ctrl_ble_adv_report_flow_ctrl_supp --cfg esp_idf_bt_ctrl_ble_scan_dupl --cfg esp_idf_bt_nimble_use_esp_timer --cfg esp_idf_coap_mbedtls_psk --cfg esp_idf_adc_disable_dac --cfg esp_idf_spi_master_isr_in_iram --cfg esp_idf_spi_slave_isr_in_iram --cfg esp_idf_esp_tls_using_mbedtls --cfg esp_idf_esp_tls_use_ds_peripheral --cfg esp_idf_esp_tls_insecure --cfg esp_idf_esp_tls_skip_server_cert_verify --cfg esp_idf_esp32s2_default_cpu_freq_160 --cfg esp_idf_esp32s2_instruction_cache_8kb --cfg esp_idf_esp32s2_instruction_cache_line_32b --cfg esp_idf_esp32s2_data_cache_0kb --cfg esp_idf_esp32s2_data_cache_line_32b --cfg esp_idf_esp32s2_universal_mac_addresses_two --cfg esp_idf_esp32s2_ulp_coproc_enabled --cfg esp_idf_esp32s2_ulp_coproc_riscv --cfg esp_idf_esp32s2_debug_ocdaware --cfg esp_idf_esp32s2_brownout_det --cfg esp_idf_esp32s2_brownout_det_lvl_sel_7 --cfg esp_idf_esp32s2_time_syscall_use_rtc_frc1 --cfg esp_idf_esp32s2_rtc_clk_src_int_rc --cfg esp_idf_esp_err_to_name_lookup --cfg esp_idf_esp_console_uart_default --cfg esp_idf_esp_console_uart --cfg esp_idf_esp_console_multiple_uart --cfg esp_idf_esp_int_wdt --cfg esp_idf_esp_task_wdt --cfg esp_idf_esp_task_wdt_check_idle_task_cpu0 --cfg esp_idf_esp_mac_addr_universe_wifi_sta --cfg esp_idf_esp_mac_addr_universe_wifi_ap --cfg esp_idf_eth_enabled --cfg esp_idf_eth_use_spi_ethernet --cfg esp_idf_eth_spi_ethernet_dm9051 --cfg esp_idf_eth_spi_ethernet_w5500 --cfg esp_idf_esp_event_post_from_isr --cfg esp_idf_esp_event_post_from_iram_isr --cfg esp_idf_esp_http_client_enable_https --cfg esp_idf_httpd_err_resp_no_delay --cfg esp_idf_esp_netif_tcpip_lwip --cfg esp_idf_esp_netif_tcpip_adapter_compatible_layer --cfg esp_idf_esp_system_panic_print_reboot --cfg esp_idf_esp_system_single_core_mode --cfg esp_idf_esp_system_rtc_fast_mem_as_heap_depcheck --cfg esp_idf_esp_system_allow_rtc_fast_mem_as_heap --cfg esp_idf_esp_system_pd_flash --cfg esp_idf_esp_system_memprot_feature --cfg esp_idf_esp_system_memprot_feature_lock --cfg esp_idf_esp_time_funcs_use_rtc_timer --cfg esp_idf_esp_time_funcs_use_esp_timer --cfg esp_idf_ [... cut ...] or="3"' --cfg 'esp_idf_version_patch="1"' --cfg esp_idf_comp_pthread_enabled --cfg esp_idf_comp_nvs_flash_enabled --cfg esp_idf_comp_esp_tls_enabled --cfg esp_idf_comp_esp_http_client_enabled --cfg esp_idf_comp_esp_http_server_enabled --cfg esp_idf_comp_espcoredump_enabled --cfg esp_idf_comp_app_update_enabled --cfg esp_idf_comp_esp_serial_slave_link_enabled --cfg esp_idf_comp_spi_flash_enabled --cfg esp_idf_comp_esp_adc_cal_enabled --cfg esp_idf_comp_mqtt_enabled --cfg esp32s2` (exit status: 1)

ivmarkov, 2022-01-28:
Yes, I know that and I'm working on fixing it. Will keep you posted.

sumhoy, 2022-01-29:
Thanks Ivan.
Let me know if I can help in any way, am happy to do so and thanks for your effort.

ivmarkov, 2022-01-29:
Fixed. As usual, git pull the latest demo, and/or do `cargo update` before clean and rebuild.

closed 2022-01-29 by ivmarkov