ivmarkov/rust-esp32-std-demo #56: patch does not apply

jcdc404, 2022-01-28:
I really don't know what to do with this error message:
```
cargo build --release

warning: unused config key `unstable.configurable-env` in `/home/newuser/MHA/ESP32/.cargo/config.toml`
warning: unused config key `unstable.extra-link-arg` in `/home/newuser/MHA/ESP32/.cargo/config.toml`
warning: unused manifest key: dependencies.esp-idf-sys.feature
   Compiling esp-idf-sys v0.30.1
error: failed to run custom build command for `esp-idf-sys v0.30.1`

Caused by:
  process didn't exit successfully: `/home/newuser/MHA/ESP32/target/release/build/esp-idf-sys-54c05dc1c94cef1b/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  Found compatible PlatformIO Core 5.2.4 -> /home/newuser/MHA/ESP32/.embuild/platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/missing_riscv_atomics_fix.diff
  cargo:rerun-if-changed=/home/newuser/MHA/ESP32/sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing release (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  os.chdir('/home/newuser/MHA/ESP32/.embuild/platformio/packages/framework-espidf')
  git init
  Initialized empty Git repository in /home/newuser/MHA/ESP32/.embuild/platformio/packages/framework-espidf/.git/
  os.chdir('/home/newuser/MHA/ESP32/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5d4b8a37800777f1/out/esp-idf')
  os.chdir('/home/newuser/MHA/ESP32/.embuild/platformio/packages/framework-espidf')
  git apply /home/newuser/MHA/ESP32/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5d4b8a37800777f1/out/esp-idf/patches/ping_setsockopt_fix.diff
  os.chdir('/home/newuser/MHA/ESP32/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5d4b8a37800777f1/out/esp-idf')

  --- stderr
  error: patch failed: components/lwip/apps/ping/ping_sock.c:274
  error: components/lwip/apps/ping/ping_sock.c: patch does not apply
  *** Error 1
  ========================== [FAILED] Took 0.28 seconds ==========================
  Environment    Status    Duration
  -------------  --------  ------------
  release        FAILED    00:00:00.277
  ==================== 1 failed, 0 succeeded in 00:00:00.277 ====================
  Error: No such file or directory (os error 2)

ivmarkov, 2022-01-28:
That's because PlatformIO just released (as in a few hours ago!) a new version of their "espressif32" platform that is based on V4.3.2. Unfortunately, one of the patches we maintain against V4.3 does not apply cleanly against V4.3.2.

A fix for that will be released later today/tomorrow.

ivmarkov, 2022-01-29:
Fixed. As usual, git pull the latest demo, and/or do `cargo update` before clean and rebuild.

closed 2022-01-29 by ivmarkov