ivmarkov/rust-esp32-std-demo #57: A stack overflow in task sys_evt with dev build

omninonsense, 2022-01-29:
Hello!

Thanks for setting up and maintaining this project. It's truly a godsend!

I have a small issue to report when using the dev build. It seems to flash successfully, but then it fails somewhere during WiFi initialisation (I think at least, I'm a fresh novice to ESPs).

```
cargo espflash /dev/ttyUSB0
Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32 (revision 1)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       4c:eb:d6:74:de:b0
    Finished dev [optimized + debuginfo] target(s) in 0.11s
[00:00:01] ########################################      16/16      segment 0x1000
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:55] ########################################     540/540     segment 0x10000

Flashing has completed!
```

When I connect to it with the monitor, it eventually gets to this output:

```
I (9645) esp_idf_svc::wifi: Got IP event: 0
I (9645) esp_idf_svc::wifi: IP event 0 handled, set status: Status(Started(Connected(Done(ClientSettings { ip: 192.168.1.113, subnet: Subnet { gateway: 192.168.1.1, mask: Mask(24) }, dns: None, secondary_dns: None }))), Started(Done))
RROR*** A stack overflow in task sys_evt has been detected.

Backtrace:0x40081946:0x3ffbf500 0x40088efd:0x3ffbf520 0x4008cc91:0x3ffbf540 0x4008b02d:0x3ffbf5c0 0x40088ff4:0x3ffbf5e0 0x40088fa6:0x00000001 |<-CORRUPTED
ELF file SHA256: 0000000000000000
```

Also, the `install-rust-toolchain.sh` script installed `xtensa-esp32-elf-llvm13_0_0-esp-13.0.0-20211203-linux-amd64` and I told it to use the [`1.58.0.0` version of the toolchain](https://github.com/esp-rs/rust-build/releases/tag/v1.58.0.0).

I don't know if this is enough information for you to work with. So, let me know if you need something else, and I'll be more than happy to provide it.

---

When doing the `--release` build, everything works. I can give you the output of that as well, but since it works as expected, I assume it's not necessary.

I was scratching my head and trying out different things since I've only started playing with this a couple of hours ago. So, if this is something common/expected, maybe it could be added to the README to lessen the frustration for future novices.

Thanks!

bernii, 2022-02-01:
I can confirm experiencing exactly the same, see below. In my case `--release` does not help though :(
```
I (7479) esp_idf_svc::wifi: Setting STA configuration: ClientConfiguration { ssid: "**REDACTED**", bssid: None, auth_method: WPA2Personal, password: "**REDACTED**", channel: Some(6), ip_conf: Some(DHCP(DHCPClientSettings { hostname: None })) }
I (7499) esp_idf_svc::wifi: Setting STA interface configuration: InterfaceConfiguration { key: "WIFI_STA_DEF", description: "sta", route_priority: 100, ip_configuration: Client(DHCP(DHCPClientSettings { hostname: None })), interface_stack: Sta }
I (7519) esp_idf_svc::wifi: STA IP configuration done
I (7519) esp_idf_svc::wifi: STA configuration done
I (7529) esp_idf_svc::wifi: Setting AP configuration: AccessPointConfiguration { ssid: "aptest", ssid_hidden: false, channel: 6, secondary_channel: None, protocols: EnumSet(P802D11B | P802D11BG | P802D11BGN), auth_method: None, password: "", max_connections: 255, ip_conf: Some(RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }) }
I (7569) esp_idf_svc::wifi: Setting AP interface configuration: InterfaceConfiguration { key: "WIFI_AP_DEF", description: "ap", route_priority: 10, ip_configuration: Router(RouterConfiguration { subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dhcp_enabled: true, dns: Some(8.8.8.8), secondary_dns: Some(8.8.4.4) }), interface_stack: Ap }
I (7599) esp_idf_svc::wifi: AP IP configuration done
I (7599) esp_idf_svc::wifi: AP configuration done
I (7609) esp_idf_svc::wifi: Starting with status: Status(Starting, Starting)
I (7619) esp_idf_svc::wifi: Status is of operating type, starting
I (7629) wifi:mode : sta (f0:08:d1:61:52:14) + softAP (f0:08:d1:61:52:15)
I (7629) wifi:enable tsf
I (7629) wifi:Total power save buffer number: 16
I (7639) wifi:Init max length of beacon: 752/752
I (7639) wifi:Init max length of beacon: 752/752
I (7649) esp_idf_svc::wifi: Got wifi event: 2
I (7649) esp_idf_svc::wifi: Reconnecting
I (7659) esp_idf_svc::wifi: Wifi event 2 handled, set status: Status(Started(Connecting), Starting)
I (7669) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,0>, prof:6
I (8449) wifi:state: init -> auth (b0)
I (8459) esp_idf_svc::wifi: Got wifi event: 12
I (8459) esp_idf_svc::wifi: Wifi event 12 handled, set status: Status(Started(Connecting), Started(Done))
I (8459) esp_idf_svc::wifi: Start requested
I (8469) esp_idf_svc::wifi: About to wait 10s for status
I (8499) wifi:state: auth -> assoc (0)
I (8509) wifi:state: assoc -> run (10)
I (8649) wifi:connected with s1, aid = 6, channel 6, BW20, bssid = 34:2c:c4:b6:44:06
I (8649) wifi:security: WPA2-PSK, phy: bgn, rssi: -88
I (8649) wifi:pm start, type: 1

I (8649) esp_idf_svc::wifi: Got wifi event: 4
I (8659) esp_idf_svc::wifi: Wifi event 4 handled, set status: Status(Started(Connected(Waiting)), Started(Done))
W (8729) wifi:<ba-add>idx:0 (ifx:0, 34:2c:c4:b6:44:06), tid:0, ssn:2, winSize:64
I (8779) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (9659) esp_idf_svc::wifi: Got IP event: 0
I (9659) esp_idf_svc::wifi: IP event 0 handled, set status: Status(Started(Connected(Done(ClientSet
***ERROR*** A stack overflow in task sys_evt has been detected.

Backtrace:0x40081946:0x3ffbf4d0 0x40088efd:0x3ffbf4f0 0x4008cc91:0x3ffbf510 0x4008b02d:0x3ffbf590 0x40088ff4:0x3ffbf5b0 0x40088fa6:0x00000001 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
I (427) cpu_start: Pro cpu up.
I (427) cpu_start: Starting app cpu, entry point is 0x40081628
I (412) cpu_start: App cpu up.
I (442) cpu_start: Pro cpu start user code
I (442) cpu_start: cpu freq: 160000000
I (442) cpu_start: Application information:
I (446) cpu_start: Project name:     esp-idf
I (451) cpu_start: App version:      950d438-dirty
I (457) cpu_start: Compile time:     Feb  1 2022 17:27:24
I (463) cpu_start: ELF file SHA256:  0000000000000000...
I (469) cpu_start: ESP-IDF:          4.3.2
I (474) heap_init: Initializing. RAM available for dynamic allocation:
I (481) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (487) heap_init: At 3FFB7F18 len 000280E8 (160 KiB): DRAM
I (493) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (499) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (506) heap_init: At 400961A4 len 00009E5C (39 KiB): IRAM
I (513) spi_flash: detected chip: gd
I (516) spi_flash: flash io: dio
W (520) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (535) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello from Rust!
More complex print ["foo", "bar"]
Result: 0, 1
Rust main thread: Thread { id: ThreadId(1), name: Some("main"), .. }
```

I'm using the `esp-1.58.0.0` branch of `github.com/esp-rs/rust` that I've built myself (including clang)

ivmarkov, 2022-02-02:
Update: putting `CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096` in `sdkconfig.defaults` might fix it.

bernii, 2022-02-09:
it seems to have fixed it (with latest `main` branch pulled)! thank you :)

brianmay, 2022-02-12:
I am getting this error too. However:

* I am using `--release`.
* I am using `CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096`.
* I am on the latest main release.

ivmarkov, 2022-02-13:
`cargo clean` and then try again.

brianmay, 2022-02-13:
@ivmarkov Thanks for that hint, seems to work now.