ivmarkov/rust-esp32-std-demo #60: edition 2021 error on build

samandeggs, 2022-02-07:
Hey - unsure if I'm just missing something obvious with the whole process, however I have followed through the rust, rust-build and this repo's instructions twice now, and still don't appear to be missing anything, but I am repeatadly getting the following error during `cargo build`:

```
Compiling embedded-svc v0.16.7
error: edition 2021 is unstable and only available with -Z unstable-options
error: could not compile `embuild`
```

There are other errors, which I'm presuming are red herrings, this error itself could be also - as this is a `std` build, yet it's appearing to want unstable items - a little out of my depth here any advice would be appreciated, however if it's obviously not your software, then please ignore.

Cheers.

ivmarkov, 2022-02-07:
You have an old rudtc compiler on your path.
 Please verify with `rustc --version`. Probably coming from a package in your OS and not from Rustup. Please remove that.

samandeggs, 2022-02-07:
```
> rustc --version
rustc 1.58.1 (db9d1b20b 2022-01-20)
```
Seems like a fairly new version though? I can still have a look around though - thanks so much for your response.

ivmarkov, 2022-02-07:
If the Rustc compiler version above ^^^ was **really** used for compiling `embedded-svc` you should have **not** received "edition 2021 is unstable and only available with -Z unstable-options", because Rust 2021 was stabilized in rustc V1.56.0.

So I still believe there is something wrong with your Rust setup.

ivmarkov, 2022-02-07:
Might be an old cargo lurking around. Why don't you remove all `.rustup` and `.cargo` directories from your home dir, check afterwards that you **don't** have `rustc`, and re-try the whole installation, starting from `rustup`.

samandeggs, 2022-02-07:
Alright, shall do - cheers for the info, you've helped a lot. Will let you know how I get on!

samandeggs, 2022-02-08:
All good - sorted it - was following too closely to other documentation and not just straight up rust-build + this repo. Cleared everything out and worked a charm. Cheers for your input.

closed 2022-02-08 by samandeggs