ivmarkov/rust-esp32-std-demo #61: Build esp32s2 error

sumhoy, 2022-02-08:
Hi Ivan,

I was compiling , cargo build , with target xtensa-esp32s2-espidf, and got the error below.  Would you mind having a look ?
Thank you very much.

   Compiling embedded-svc v0.16.7
error[E0061]: this function takes 1 argument but 0 arguments were supplied
     --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.9/src/interrupt.rs:25:9
      |
25    |         vPortEnterCritical();
      |         ^^^^^^^^^^^^^^^^^^-- supplied 0 arguments
      |         |
      |         expected 1 argument
      |
note: function defined here
     --> /home/sumhoy/git/projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/build/esp-idf-sys-6277f74ff3fcd334/out/bindings.rs:22236:12
      |
22236 |     pub fn vPortEnterCritical(mux: *mut portMUX_TYPE);
      |            ^^^^^^^^^^^^^^^^^^

error[E0061]: this function takes 1 argument but 0 arguments were supplied
     --> /home/sumhoy/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.32.9/src/interrupt.rs:44:9
      |
44    |         vPortExitCritical();
      |         ^^^^^^^^^^^^^^^^^-- supplied 0 arguments
      |         |
      |         expected 1 argument
      |
note: function defined here
     --> /home/sumhoy/git/projects/rust-esp32-std-demo/target/xtensa-esp32s2-espidf/debug/build/esp-idf-sys-6277f74ff3fcd334/out/bindings.rs:22239:12
      |
22239 |     pub fn vPortExitCritical(mux: *mut portMUX_TYPE);
      |            ^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0061`.
error: could not compile `esp-idf-hal` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...

ivmarkov, 2022-02-08:
Sure. If [this](https://github.com/esp-rs/esp-idf-hal/commit/ca508c8487ab2c142ce9fb5e7800f21e39495568) passes CI I'll republish the crate and let you know and then you can do `cargo update` and then re-try the build.

sumhoy, 2022-02-08:
did cargo update, all working fine thank you !

closed 2022-02-08 by sumhoy