ivmarkov/rust-esp32-std-demo #63: Failure to build for target riscv32imc-esp-espidf

yoganandc, 2022-02-14:
rust version = `rustc 1.60.0-nightly (1e12aef3f 2022-02-13)` (nightly toolchain)
target = `riscv32imc-esp-espidf`

might be related to this? https://github.com/esp-rs/embedded-svc/commit/3e822e789ee03a302977b1e9d581fba8bb145b65

```
yogi@yogi-mac rust-esp32-std-demo % cargo build                            
   Compiling esp-idf-sys v0.30.5
   Compiling embedded-svc v0.16.8
error[E0034]: multiple applicable items in scope
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/utils/nonblocking/event_bus.rs:214:24
    |
214 |             CV::Mutex::new(SubscriptionState {
    |                        ^^^ multiple `new` found
    |
note: candidate #1 is defined in the trait `mutex::Mutex`
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/mutex.rs:21:5
    |
21  |     fn new(data: Self::Data) -> Self;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: candidate #2 is defined in the trait `mutex::Condvar`
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/mutex.rs:33:5
    |
33  |     fn new() -> Self;
    |     ^^^^^^^^^^^^^^^^^
help: disambiguate the associated function for candidate #1
    |
214 |             <_ as mutex::Mutex>::new(SubscriptionState {
    |             ~~~~~~~~~~~~~~~~~~~~~
help: disambiguate the associated function for candidate #2
    |
214 |             <_ as mutex::Condvar>::new(SubscriptionState {
    |             ~~~~~~~~~~~~~~~~~~~~~~~

error[E0609]: no field `0` on type `Arc<_>`
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/utils/nonblocking/event_bus.rs:226:51
    |
226 |                 let (mut state, condvar) = (state.0.lock(), &state.1);
    |                                                   ^ unknown field

error[E0034]: multiple applicable items in scope
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/utils/nonblocking/mqtt/client.rs:174:37
    |
174 |                 payload: CV::Mutex::new(Payload {
    |                                     ^^^ multiple `new` found
    |
note: candidate #1 is defined in the trait `mutex::Mutex`
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/mutex.rs:21:5
    |
21  |     fn new(data: Self::Data) -> Self;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: candidate #2 is defined in the trait `mutex::Condvar`
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/mutex.rs:33:5
    |
33  |     fn new() -> Self;
    |     ^^^^^^^^^^^^^^^^^
help: disambiguate the associated function for candidate #1
    |
174 |                 payload: <_ as mutex::Mutex>::new(Payload {
    |                          ~~~~~~~~~~~~~~~~~~~~~
help: disambiguate the associated function for candidate #2
    |
174 |                 payload: <_ as mutex::Condvar>::new(Payload {
    |                          ~~~~~~~~~~~~~~~~~~~~~~~

error[E0284]: type annotations needed: cannot satisfy `<CV as mutex::Condvar>::Mutex<_> == _`
   --> /Users/yogi/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-svc-0.16.8/src/utils/nonblocking/mqtt/client.rs:194:55
    |
194 |             payload = self.connection_state.processed.wait(payload);
    |                                                       ^^^^ cannot satisfy `<CV as mutex::Condvar>::Mutex<_> == _`
```

yoganandc, 2022-02-14:
looked at CI and saw that it passed with `channel = "nightly-2022-02-08"`. I did
```
rustup install nightly-2022-02-08
rustup component add rust-src --toolchain nightly-2022-02-08-aarch64-apple-darwin
rustup default nightly-2022-02-08
```
verified that everything builds with that version.

ivmarkov, 2022-02-14:
I'll fix soon, thanks for reporting.

ivmarkov, 2022-02-20:
Seems they fixed something as CI works again. Maybe this one: https://github.com/rust-lang/rust/pull/94088

closed 2022-02-20 by ivmarkov