ivmarkov/rust-esp32-std-demo #65: clippy errors

brianmay, 2022-02-20:
Just curious what is wrong with clippy in the CI?

https://github.com/ivmarkov/rust-esp32-std-demo/runs/5257226106?check_suite_focus=true

It seems to be complaining about errors in embedded-svc.

But my understanding is that as we use the `--no-deps` option we shouldn't be checking for errors in dependencies.

Also if I run the same command locally it works.

ivmarkov, 2022-02-20:
Issues with latest Rust nightly and GATs.

ivmarkov, 2022-02-20:
Looking at it since an hour or so, but not sure what is going on. Might be a regression in the compiler itself, as the error looks bogus.

ivmarkov, 2022-02-20:
Apparently they fixed something because since this morning, we get clean CIs again.

Perhaps [this one](https://github.com/rust-lang/rust/pull/94088).

closed 2022-02-20 by ivmarkov