ivmarkov/rust-esp32-std-demo #67: linking with `ldproxy` failed?

S3j5b0, 2022-02-28:
Hi, I'm running ubuntu, and i've tried to build this project.

ldproxy is installed, the env variables mentioned should be set, and `rustup default esp` is run.

When I try to build, I end up failing like this:

```
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: "ldproxy" "/home/ask/Git/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/deps/rust_esp32_std_demo-809052f0f64fb6cb.rust_esp32_std_demo.2bb45e5c-cgu.0.rcgu.o" "/home/ask/Git/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/deps/rust_esp32_std_demo-809052f0f64fb6cb.rust_esp32_std_demo.2bb45e

....

wrap=__register_frame_info_table_bases" "-Wl,--wrap=__register_frame_table" "-Wl,--wrap=longjmp" "-Wl,--wrap=mbedtls_mpi_exp_mod" "-Wno-frame-address" "-fno-lto" "-fno-rtti" "-mlongcalls" "-Og" "-g2" "-ggdb2"
  = note: Running ldproxy
          Error: Linker /home/ask/Git/rust-esp32-std-demo/.embuild/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/ask/Git/rust-esp32-std-demo/.embuild/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/ask/Git/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/deps/libstd-eec2b8affb5436c9.rlib(std-eec2b8affb5436c9.std.ede01b72-cgu.1.rcgu.o):(.literal._ZN3std4time7Instant3now17hd5915cce09384ec8E+0x10): undefined reference to `__sync_val_compare_and_swap_8'
          collect2: error: ld returned 1 exit status
          
          
  = help: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `rust-esp32-std-demo` due to previous error

```

Do you have any idea why this might happen? it seems like I have the crate, but maybe the version is wrong or something?

ivmarkov, 2022-02-28:
Is this with the latest 1.59 compiler?
* If yes, I need to fix something, thanks for reporting. In the meantime you can work-around the issue by enabling the `release/4.4` branch in `.cargo/config.toml` inside the demo crate and then building with `cargo build --features native`.
* If not, please let me know what compiler version you are using.

atoav, 2022-03-07:
I have the same issue, I checked, my rustc-version is `rustc 1.59.0-dev`.

ivmarkov, 2022-03-07:
For now, a workaround would be to use the native build as described above.

S3j5b0, 2022-03-10:
Hi, I tried the fix as described, it build for a while, and then this happens:
```
$ cargo build --features native
   Compiling esp-idf-sys v0.30.6
   Compiling esp-idf-hal v0.33.1
   Compiling esp-idf-svc v0.37.2
   Compiling rust-esp32-std-demo v0.23.0 (/home/ask/Git/rust-esp32-std-demo)
error[E0412]: cannot find type `TcpStream` in this scope
   --> src/main.rs:690:47
    |
690 |             async fn echo(stream: smol::Async<TcpStream>) -> smol::io::Result<()> {
    |                                               ^^^^^^^^^ not found in this scope
    |
help: consider importing one of these items
    |
673 |     use crate::TcpStream;
    |
673 |     use std::net::TcpStream;
    |

error[E0412]: cannot find type `TcpListener` in this scope
   --> src/main.rs:696:42
    |
696 |             let listener = smol::Async::<TcpListener>::bind(([0, 0, 0, 0], 8081))?;
    |                                          ^^^^^^^^^^^ not found in this scope
    |
help: consider importing one of these items
    |
673 |     use crate::TcpListener;
    |
673 |     use std::net::TcpListener;
    |

error[E0433]: failed to resolve: use of undeclared crate or module `thread`
   --> src/main.rs:717:17
    |
717 |         thread::Builder::new().stack_size(4096).spawn(move || {
    |                 ^^^^^^^ not found in `thread`
    |
help: consider importing one of these items
    |
673 |     use crate::thread::Builder;
    |
673 |     use std::thread::Builder;
    |

Some errors have detailed explanations: E0412, E0433.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `rust-esp32-std-demo` due to 3 previous errors
ask@Garsy:~/Git/rust-esp32-std-demo$ 
```

mike1703, 2022-03-10:
I also stumbled over this while fixing #68 and fixed it by importing the necessary crates into the mod.
See: https://github.com/ivmarkov/rust-esp32-std-demo/commit/3bafe22f8e6bf834c70de690cf6dc90863fce727#diff-42cb6807ad74b3e201c5a7ca98b911c5fa08380e942be6e4ac5807f8377f87fcR673 

Are you using the latest code?

ivmarkov, 2022-03-10:
It is just a compilation error (missing imports). Might be related to what @mike1703 is mentioning even though "worked for a while" sounds as if you've also made some code changes yourself and it no longer compiles?

Perhaps just fix them damn imports, as the compiler itself suggests? :)

S3j5b0, 2022-03-11:
Ah yes I pulled again and now it works - with the fixes a described above.
Thanks a lot for the help, you guys are super helpfull

closed 2022-03-16 by S3j5b0

Traijan1, 2022-04-02:
> Is this with the latest 1.59 compiler?
>
> * If yes, I need to fix something, thanks for reporting. In the meantime you can work-around the issue by enabling the `release/4.4` branch in [...]

Hello, I just wanted to ask how do I do this? I'm not that experienced with the config.toml but wanted to start on the ESP32 with Rust! Thanks for the help!