ivmarkov/rust-esp32-std-demo #68: example code is not working with qemu networking

mike1703, 2022-03-01:
I have issues with running the example code on qemu and have working networking.

one issue is that code apparently was copied from the wifi section [here](https://github.com/ivmarkov/rust-esp32-std-demo/blob/7ae9cb319f307f58108a06f2cffdcebb6e395762/src/main.rs#L1272) and produces this error

```
error[E0425]: cannot find value `wifi` in this scope
    --> src/main.rs:1272:5
     |
1272 |     wifi.wait_status_with_timeout(Duration::from_secs(10), |status| !status.is_transitional())
     |     ^^^^ not found in this scope
```
this can be fixed by including 
`use embedded_svc::eth::TransitionalState;`
and replacing `wifi` with `eth`

But then the code seems to have problems in receiving an IP when starting with qemu.sh (therefor started with `qemu-system-xtensa -nographic -machine esp32 -nic user,model=open_eth,id=lo0,hostfwd=tcp:127.0.0.1:7888-:80 -drive ...`).
```
...
I (10439) system_api: Base MAC address is not set
I (10459) system_api: read default base MAC address from EFUSE
I (10499) esp_idf_svc::eth: Driver initialized
I (10519) esp_idf_svc::eth: Event handlers registered
I (10519) esp_idf_svc::eth: Initialization complete
I (10519) rust_esp32: Eth created
I (10539) esp_idf_svc::eth: Setting configuration: Client(DHCP(DHCPClientSettings { hostname: None }))
I (10539) esp_idf_svc::eth: Stopping
W (10549) esp_eth: driver not started yet
I (10549) esp_idf_svc::eth: Stop requested
I (10569) esp_idf_svc::eth: Setting client interface configuration: InterfaceConfiguration { key: "ETH_CL_DEF", description: "eth", route_priority: 60, ip_configuration: Client(DHCP(DHCPClientSettings { hostname: None })), interface_stack: Eth }
I (10589) esp_eth.netif.glue: 00:00:00:00:00:03
I (10589) esp_eth.netif.glue: ethernet attached to netif
I (10599) esp_idf_svc::eth: IP configuration done
I (10599) esp_idf_svc::eth: Starting with status: Starting
I (10599) esp_idf_svc::eth: Status is of operating type, starting
I (10719) esp_idf_svc::eth: Start requested
I (10739) esp_idf_svc::eth: ETH netif status: EspNetif(EspNetifStack(PrivateData), 0x3ffc2494), index: 1, name: lo0, ifkey: ETH_CL_DEF
I (10749) esp_idf_svc::eth: Got eth event: Started(0x3ffbf884) 
I (10749) esp_idf_svc::eth: Eth event Started(0x3ffbf884) handled, set status: Starting
I (10749) esp_idf_svc::eth: Got eth event: Connected(0x3ffbf884) 
I (10759) esp_idf_svc::eth: Eth event Connected(0x3ffbf884) handled, set status: Started(Connected(Waiting))
I (10759) esp_idf_svc::eth: Configuration set
I (10769) rust_esp32: Eth configuration set, about to get status
I (10769) esp_idf_svc::eth: About to wait 10s for status
I (16489) esp_idf_svc::eth: Timeout while waiting for status
I (16509) esp_idf_svc::nvs: Dropped
I (16519) esp_idf_svc::eventloop: Dropped
I (16519) esp_idf_svc::netif: Dropped
Error: Unexpected Eth status: Started(Connected(Waiting))
```
As the code itself had to be patched it does not look like it was tested yet.
Is there an issue with the code itself, with the qemu command (network options) or potentially something wrong with my system? 

### Versions
ESP-IDF v4.3.2:
```
I (3365) cpu_start: ESP-IDF:          v4.3.2-dirty
```
Environment:
```
$ qemu-system-xtensa --version
QEMU emulator version 6.2.0 (v4.2.0-rc1-18806-g440ff71648)
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers
$ cargo version
cargo 1.59.0-dev
$ rustc --version
rustc 1.59.0-dev
```

mike1703, 2022-03-01:
I can get better results If I'm using the following line in .cargo/config.toml
`ESP_IDF_VERSION = { value = "release/v4.4" }`

```
...
I (8911) system_api: Base MAC address is not set
I (8921) system_api: read default base MAC address from EFUSE
I (8971) esp_idf_svc::eth: Driver initialized
I (8981) esp_idf_svc::eth: Event handlers registered
I (8981) esp_idf_svc::eth: Initialization complete
I (8981) rust_esp32: Eth created
I (8991) esp_idf_svc::eth: Setting configuration: Client(DHCP(DHCPClientSettings { hostname: None }))
I (9001) esp_idf_svc::eth: Stopping
E (9001) esp_eth: esp_eth_stop(298): driver not started yet
I (9001) esp_idf_svc::eth: Stop requested
I (9021) esp_idf_svc::eth: Setting client interface configuration: InterfaceConfiguration { key: "ETH_CL_DEF", description: "eth", route_priority: 60, ip_configuration: Client(DHCP(DHCPClientSettings { hostname: None })), interface_stack: Eth }
I (9061) esp_eth.netif.netif_glue: 00:00:00:00:00:03
I (9061) esp_eth.netif.netif_glue: ethernet attached to netif
I (9071) esp_idf_svc::eth: IP configuration done
I (9071) esp_idf_svc::eth: Starting with status: Starting
I (9081) esp_idf_svc::eth: Status is of operating type, starting
I (9231) esp_idf_svc::eth: Start requested
I (9271) esp_idf_svc::eth: ETH netif status: EspNetif(EspNetifStack(PrivateData), 0x3ffc1114), index: 2, name: en1, ifkey: ETH_CL_DEF
I (9281) esp_idf_svc::eth: Got eth event: Started(0x3ffc0d0c) 
I (9281) esp_idf_svc::eth: Eth event Started(0x3ffc0d0c) handled, set status: Starting
I (9331) esp_idf_svc::eth: Got eth event: Connected(0x3ffc0d0c) 
I (9331) esp_idf_svc::eth: Eth event Connected(0x3ffc0d0c) handled, set status: Started(Connected(Waiting))
I (9331) esp_idf_svc::eth: Configuration set
I (9341) rust_esp32: Eth configuration set, about to get status
I (9341) esp_idf_svc::eth: About to wait 10s for status
I (10061) esp_idf_svc::eth: Got IP event: DhcpIpAssigned(DhcpIpAssignment { netif_handle: 0x3ffc1114, ip_settings: ClientSettings { ip: 10.0.2.15, subnet: Subnet { gateway: 10.0.2.2, mask: Mask(24) }, dns: None, secondary_dns: None }, ip_changed: true })
I (10071) esp_idf_svc::eth: IP event DhcpIpAssigned(DhcpIpAssignment { netif_handle: 0x3ffc1114, ip_settings: ClientSettings { ip: 10.0.2.15, subnet: Subnet { gateway: 10.0.2.2, mask: Mask(24) }, dns: None, secondary_dns: None }, ip_changed: true }) handled, set status: Started(Connected(Done(Some(ClientSettings { ip: 10.0.2.15, subnet: Subnet { gateway: 10.0.2.2, mask: Mask(24) }, dns: None, secondary_dns: None }))))
I (10081) esp_idf_svc::eth: Waiting for status done - success
I (10081) esp_idf_svc::eth: Providing status: Started(Connected(Done(Some(ClientSettings { ip: 10.0.2.15, subnet: Subnet { gateway: 10.0.2.2, mask: Mask(24) }, dns: None, secondary_dns: None }))))
I (10091) rust_esp32: Eth connected
I (10091) rust_esp32: About to do some pings for ClientSettings { ip: 10.0.2.15, subnet: Subnet { gateway: 10.0.2.2, mask: Mask(24) }, dns: None, secondary_dns: None }
...
```

ivmarkov, 2022-03-01:
Sorry about the copy-pasted code. This occurred with the last commit, as the API of wifi (and eth) changed, and I had to update it, yet I neither compiled, nor tested with eth enabled.

I would gladly accept a PR for this.

As for the second issue, as you have noticed, QEMU only works with V4.4 (used to be master) and *should* work with 5.0 (= master) but I haven't tested the latter. As the instructions mention:
`export ESP_IDF_VERSION=master; cargo build --features native,qemu`

I'll keep this issue open until the copy-paste is fixed & merged. And will probably update the README to also mention version `release/4.4`.

closed 2022-03-02 by ivmarkov