ivmarkov/rust-esp32-std-demo #70: build native fails in docker

Moran296, 2022-03-12:
Being new to rust and also docker, I have been struggling to build a project for quite some time now.
I am trying to use docker to build this project. I will be describing the problems along the way.

Since for some reason esp is not the default toolchain when running the docker env and since #67, I have uncommented the line `ESP_IDF_VERSION = { value = "release/v4.4" }` in config.toml, and my docker command is as follows:

`docker run -it -v $PWD:/project -w /project espressif/idf-rust cargo +esp build --features "native"`

This is the closest I got to building for now but I fail with:

`` 
Compiling esp-idf-sys v0.30.6
error: failed to run custom build command for "esp-idf-sys v0.30.6"

Caused by:
  process didn't exit successfully: `/project/target/debug/build/esp-idf-sys-1f8da36d2cb1c9ed/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  PATH=/project/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH

  --- stderr
  Error: Could not install esp-idf

  Caused by:
      command '"python3" "/project/.embuild/espressif/esp-idf-release/v4.4/tools/idf_tools.py" "--idf-path" "/project/.embuild/espressif/esp-idf-release/v4.4" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1
` `

I also had sometimes `fatal: No names found, cannot describe anything.` errors, but going into .embuild/espressif/esp-idf-release/v4.4 and running `sudo git git fetch --depth=500` seemed to stop them..

Running the last command without --quiet I get
`ERROR: This script was called from a virtual environment, can not create a virtual environment again`

`which python3`:
/opt/esp/python_env/idf5.0_py3.8_env/bin/python3

tried to run the last command with /bin/python3 instead of python3 and it seemed to do what it was supposed to do but cargo build fails still with the same error.

Will be glad for any assistence

lexologe, 2022-03-28:
Hello, I am not using Docker but I have the same issue. I tried to remove the .embuild and target directory and set ESP_IDF_VERSION to release/v4.4 or master but both failed.

brianmay, 2022-03-29:
I created this experimental Dockerfile that builds everything from scratch from clean docker environment. No caching, so not particularly  fast.

https://github.com/ivmarkov/rust-esp32-std-demo/compare/main...brianmay:docker?expand=1

Build with:

```
docker build -t rust-esp32-std-demo .
```

At present it fails with a linker error: `undefined reference to `__sync_val_compare_and_swap_8'`. I suspect I might be missing a library or something.

No idea if this helps or not...

brianmay, 2022-03-30:
Remembered I forgot to select IDF 4.4. With that change, can now build in a Docker container.

Moran296, 2022-04-25:
I could not compile or burn using docker. Also tried brianmay's solution.
Finally I could compile and burn using https://github.com/esp-rs/rust-build native install instructions.

closed 2022-05-07 by Moran296