ivmarkov/rust-esp32-std-demo #72: Reading efuse values?

S3j5b0, 2022-03-16:
Hi; i have a use case where I need some key material to be hardcoded onto the devices. For that purpose,  using the efuse component seemed like the best option.

So I found the esptool, and espfuse tools, that I can use to read the data and write to it through the command line.

But I can't find too much on how to read the efuse from rust. I found [this thing(https://docs.rs/esp32/0.6.0/esp32/efuse/index.html), but the docs are exremely sparse, and there is no examples to the code.

The only thing I really can find about it, is this line, in your sample output on the readme:
```

Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4761) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
```

Is there any resources that you know of that one can learn to read these keys?

brianmay, 2022-03-16:
In my code I have:

```
let mut mac: [u8; 6] = [0; 6];
unsafe {
    let ptr = &mut mac as *mut u8;
    esp_efuse_mac_get_default(ptr);
 }
let client_id = format!("robotica-remote-rust_{}", hex::encode(mac));
```

I don't think there is any safe API to access this yet.

S3j5b0, 2022-03-17:
ah thanks a lot! Which crate has this function, and do you know if it has one for reading the key blocks, adn not the mac value?

ivmarkov, 2022-03-17:
`esp-idf-sys` which is raw unsafe bindings to ESP IDF.
Therefore check the ESP IDF documentation / API for esp efuse: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html

S3j5b0, 2022-03-17:
cool thx. 
But the efuse manager api seems to be c. Is there no way to do this in pure rust?

My thought was that I could use espefuse.py to place some keys on my devices and then read from them in rust, so they can be used in my application. Is this not possilbe?

ivmarkov, 2022-03-17:
Please check esp-idf-sys and read on the Bindgen utility and what raw unsafe bindings really mean in Rust - this will help you get started. Essentially I'm telling you that most ESP IDF C APIs you can call from rust already as they are exposed as unsafe Rust APIs in esp-idf-sys.

closed 2022-03-17 by ivmarkov