ivmarkov/rust-esp32-std-demo #73: Question:why use different pins from document

buhe, 2022-03-18:
Hi, 
[image]
I use esp32 devkit and waveshare 1.54in e-paper, i see doc gpio14 is spi clk pin and gpio13 is mosi ,but code is sclk: gpio::Gpio13<gpio::Unknown>, sdo: gpio::Gpio14<gpio::Unknown>,. why?

thanks

buhe

buhe, 2022-04-04:
thanks @simon0356 reply

The code is definitely correct. I just have a question, look at that picture. The picture says gpio14 is sck and the code is mosi, so I want to ask why.

ivmarkov, 2022-04-04:
> thanks @simon0356 reply
> 
> The code is definitely correct. I just have a question, look at that picture. The picture says gpio14 is sck and the code is mosi, so I want to ask why.

What the picture is telling you is what e.g. gpio14 is *usually* used for. You can use SPI1/SPI2 with any pin configuration. It is just that some configurations are slightly more efficient as then the pin routing goes via IOMUX (Direct IO) and not via the GPIO matrix.

Consult p45 in the ESP32 Technical Reference: https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

buhe, 2022-04-05:
Excuse me, how does this project map the GPIO matrix, thanks.