ivmarkov/rust-esp32-std-demo #76: Compilation error, esp-idf-sys

simeondmr, 2022-04-21:
Hello,

first of all thanks for this project.
I'm using Linux Debian and I have this error during the compilation:

```
cargo build
   Compiling esp-idf-sys v0.31.0
error: failed to run custom build command for `esp-idf-sys v0.31.0`

```

If it can help, I followed the instructions in https://github.com/esp-rs/rust-build for the Xtensa support.

Please could you help me to solve this issue?

Thanks

ivmarkov, 2022-04-21:
Please provide the full build output.

simeondmr, 2022-04-21:
```
   Compiling esp-idf-sys v0.31.0
error: failed to run custom build command for `esp-idf-sys v0.31.0`

Caused by:
  process didn't exit successfully: `/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/release/build/esp-idf-sys-73798fc8e1fa16ed/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH
  Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH
  Skipping cmake@3.20.3 (already installed)
  IDF_PYTHON_ENV_PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out/sdkconfig
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out/gen-sdkconfig.defaults...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- App "libespidf" version: 1
  -- Adding linker script /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/components/esp_system/ld/esp32/sections.ld.in
[... cut ...] ING_ENABLED", "-target", "xtensa"]
  thread 'main' panicked at 'Unable to find libclang: "couldn't find any valid shared libraries matching: ['libclang.so', 'libclang-*.so', 'libclang.so.*', 'libclang-*.so.*'], set the `LIBCLANG_PATH` environment variable to a path where one of these files can be found (invalid: [])"', /home/dmrsim/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/lib.rs:2144:31
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

ivmarkov, 2022-04-22:
You don't have the xtensa clang on your path. Did you follow the xtensa toolchain installation instructions s carefully? At the end of the installation, it output things that you should keep on your '$path'.

alexeden, 2022-04-23:
Adding the stuff to `$PATH` worked for me, thank you!  I had what I presume was the same underlying problem but was seeing a slightly different error message. Posting here in case anyone else runs into it:

```
--- stderr
fatal: No names found, cannot describe anything.
Output: "/Users/alexeden/code/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-521cf171d7f8b4f3/out/bindings.rs"
Bindgen builder flags: ["/Users/alexeden/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.0/src/include/....(redacted)... "-DESP_IDF_COMP_ESP_WIFI_PROVISIONING_ENABLED", "-target", "xtensa"]
error: unknown target triple 'xtensa', please use -triple or -arch
thread 'main' panicked at 'libclang error; possible causes include:
- Invalid flag syntax
- Unrecognized flags
- Invalid flag arguments
- File I/O errors
- Host vs. target architecture mismatch
If you encounter an error missing from this list, please file an issue or a PR!', /Users/alexeden/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/ir/context.rs:538:15
```

simeondmr, 2022-04-23:
Now I have this error, could you help me?

```
cargo build --release
   Compiling esp-idf-sys v0.31.0
error: failed to run custom build command for `esp-idf-sys v0.31.0`

Caused by:
  process didn't exit successfully: `/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/release/build/esp-idf-sys-73798fc8e1fa16ed/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH
  Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH
  Skipping cmake@3.20.3 (already installed)
  IDF_PYTHON_ENV_PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.20.3/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/.embuild/espressif/esp-idf-release/v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out/sdkconfig
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out/gen-sdkconfig.defaults...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- App "libespidf" version: 1
  -- Adding linker script /home/dmrsim/Documenti/rust-embedded/rust-esp32-std-demo/target/xtensa-esp32-espidf/release/build/esp-idf-sys-5ebc1c3370eee0af/out/build/esp-idf/esp_system/ld/memory.ld
[... cut ...]
     2: core::panicking::panic_display
     3: core::panicking::panic_str
     4: core::option::expect_failed
     5: core::option::Option<T>::expect
     6: bindgen::ir::context::BindgenContext::new
     7: bindgen::Bindings::generate
     8: bindgen::Builder::generate
     9: embuild::bindgen::run_for_file
    10: embuild::bindgen::run
    11: build_script_build::main
    12: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

closed 2022-04-23 by ivmarkov