ivmarkov/rust-esp32-std-demo #78: Workspace setup issue

callmephilip, 2022-04-21:
@ivmarkov I am not sure if this is the right place to ask this question - please let me know if there is a better place for this

I am trying to reorganize the original project as a workspace in order to isolate part of the code in a library to improve test flow as described in [this article](https://ferrous-systems.com/blog/test-embedded-app/). I created a [PR](https://github.com/ivmarkov/rust-esp32-std-demo/pull/77) to illustrate the approach.

The project builds and flashes correctly, however I get the following exception when running it on the device:

```
***ERROR*** A stack overflow in task sys_evt has been detected.

Backtrace:0x40082c0a:0x3ffbd6600x40087e15:0x3ffbd680 0x4008afaa:0x3ffbd6a0 0x400898a1:0x3ffbd720 0x40087f14:0x3ffbd750 0x40087ec6:0x00000080  |<-CORRUPTED

ELF file SHA256: 0000000000000000
```

I am fairly new to embedded development and I found myself stuck at this point. Any suggestion/pointer would be highly appreciated. 

Including full log below

```
Chip type:         ESP32 (revision 1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       30:83:98:71:28:24
[00:00:01] ########################################      16/16      segment 0x1000
[00:00:00] ########################################       1/1       segment 0x8000
[00:01:20] ########################################     775/775     segment 0x10000

Resetting device... done
0, SPIWP:0xee
clk_drv:0x00,q_drets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
I (589) cpu_start: Pro cpu up.
I (589) cpu_start: Starting app cpu, entry point is 0x40082684
I (0) cpu_start: App cpu up.
I (603) cpu_start: Pro cpu start user code
I (603) cpu_start: cpu freq: 160000000
I (603) cpu_start: Application information:
I (607) cpu_start: Project name:     libespidf
I (613) cpu_start: App version:      1
I (617) cpu_start: Compile time:     Apr 21 2022 12:52:43
I (623) cpu_start: ELF file SHA256:  0000000000000000...
I (629) cpu_start: ESP-IDF:          4b2098b
I (634) heap_init: Initializing. RAM available for dynamic allocation:
I (641) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (647) heap_init: At 3FFB7860 len 000287A0 (161 KiB): DRAM
I (653) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (660) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (666) heap_init: At 400937B0 len 0000C850 (50 KiB): IRAM
I (674) spi_flash: detected chip: generic
I (677) spi_flash: flash io: dio
W (681) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (695) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello from Rust!
More complex print ["foo", "bar"]
Result: 0, 1
Rust main thread: Thread { id: ThreadId(1), name: Some("main"), .. }
Main TLS before change: 13
Main TLS after change: 42
This is thread number 0, Thread { id: ThreadId(2), name: None, .. }
[... 99 lines cut ...]
I (7761) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,0>, prof:6
I (8511) wifi:state: init -> auth (b0)
I (8511) esp_idf_svc::wifi: Got wifi event: ApStarted
I (8511) esp_idf_svc::wifi: STA event ApStarted handled, set status: Status(Started(Connecting), Started(Done))
I (8521) wifi:state: auth -> assoc (0)
I (8521) esp_idf_svc::wifi: Configuration set
I (8531) wifi:state: assoc -> run (10)
I (8531) rust_esp32_std_demo: Wifi configuration set, about to get status
I (8541) esp_idf_svc::wifi: About to wait 20s for status
I (8551) wifi:connected with acs, aid = 1, channel 6, BW20, bssid = 78:45:58:4d:49:11
I (8551) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (8561) wifi:pm start, type: 1
I (8561) esp_idf_svc::wifi: Got wifi event: StaConnected
I (8561) esp_idf_svc::wifi: STA event StaConnected handled, set status: Status(Started(Connected(Waiting)), Started(Done))
I (8581) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (8611) wifi:<ba-add>idx:0 (ifx:0, 78:45:58:4d:49:11), tid:6, ssn:1, winSize:64
I (9321) esp_idf_svc::wifi: Got STA IP event: DhcpIpAssigned(DhcpIpAssignment { netif_handle: 0x3ffc5968, ip_settings: ClientSettings { ip: 192.168.0.217, subnet: Subnet { gateway: 192.168.0.1, mask: Mask(24) }, dns: None, secondary_dns: None }, ip_changed: true })
I (9331) esp_idf_svc::wifi: STA IP event DhcpIpAssigned(DhcpIpAssignment { netif_handle: 0x3ffc5968, ip_settings: ClientSettings { ip: 192.168.0.217, subnet: S
***ERROR*** A stack overflow in task sys_evt has been detected.

Backtrace:0x40082c0a:0x3ffbd6600x40087e15:0x3ffbd680 0x4008afaa:0x3ffbd6a0 0x400898a1:0x3ffbd720 0x40087f14:0x3ffbd750 0x40087ec6:0x00000080  |<-CORRUPTED

ELF file SHA256: 0000000000000000
Rebooting...
```

ivmarkov, 2022-04-21:
It crashes because you need to move all sdkconfig.defaults* files into the root of your workspace. Otherwise they are not picked up and the tasks run with too little stack size.

ivmarkov, 2022-04-21:
... and clean + rebuild after that.

callmephilip, 2022-04-21:
@ivmarkov thank you very much for the quick response! moving configs did the trick. thank you. here's a link to the updated code for posterity - https://github.com/bakery/rust-esp32-std-demo/tree/workspace-setup

closed 2022-04-21 by callmephilip