ivmarkov/rust-esp32-std-demo #81: Enable proper TLS

svenstaro, 2022-04-24:
It seems like the [original issue](https://github.com/espressif/esp-idf/issues/7631) for why TLS was disabled in the first place has been resolved with esp-idf 4.4 (which has been released). Is it now feasible to turn on TLS with proper verification and certificates?

ivmarkov, 2022-04-24:
Yes, I believe so. I'll turn it on for my own purposes likely in ~ 2 months from now (overwhelmed with other stuff). If you need it sooner, I'll gladly accept a PR!

svenstaro, 2022-04-24:
Already tried and it's "easy" to turn on but there's this blocker currently: https://github.com/esp-rs/esp-idf-svc/issues/67

ivmarkov, 2022-04-24:
Ok so it is not exposed by esp-idf-svc, sure. Vut is it also not exposed by esp-idf-sys?

svenstaro, 2022-04-24:
It's not exposed as far as I can see. Furthermore, since it's marked `extern` and it's inline asm I think there's no good way to expose it since it's needed at link time. This is currently a bit over my head.

ivmarkov, 2022-04-25:
Dupe of https://github.com/esp-rs/esp-idf-svc/issues/67

closed 2022-04-25 by ivmarkov