ivmarkov/rust-esp32-std-demo #86: Backtrace esp-idf 4.3.1+

jan-br, 2022-05-03:
Hi,
So I've noticed that the std example states [here](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L131), it is no longer possible to generate backtraces.
And I'm really wondering why that is the case. 
If someone could deliver technical details about the problems with that topic, I'd gladly try to help fix it.

ivmarkov, 2022-05-08:
Actually the above comment - last time we checked - is incorrect. Have you teried uncommenting it?

jan-br, 2022-05-08:
I've tried to get backtraces working for a while now. Didn't succeed so far, but I'll take a closer look at it again

jan-br, 2022-05-08:
Okay so apparantly it is working better than I thought it was.

I created a panic right after the `env::set_var("RUST_BACKTRACE", "1");`
The result seems to be promising, but well that's the output.

```
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', src/main.rs:135:28
stack backtrace:

abort() was called at PC 0x4010b0ee [abort_return<int>:/home/jan/CLionProjects/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/cxx/cxx_exception_stubs.cpp:33] on core 0
Backtrace:0x4008251a [panic_abort:/home/jan/CLionProjects/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/panic.c:402]:0x3ffb60100x40084d09 [esp_system_abort:/home/jan/CLionProjects/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/esp_system.c:128]:0x3ffb6030 0x400898e2 [abort:/home/jan/CLionProjects/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/newlib/abort.c:46]:0x3ffb6050 0x4010b0ee [abort_return<int>:/home/jan/CLionProjects/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/cxx/cxx_exception_stubs.cpp:33]:0x3ffb60c0 0x400f23cf [std::backtrace_rs::backtrace::libunwind::trace:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93]:0x3ffb60e0 0x400dc3fb [std::sys_common::backtrace::_print_fmt:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:67]:0x3ffb6110 0x40104e9d [core::fmt::write:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:1168]:0x3ffb6180 0x400de782 [std::io::Write::write_fmt:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/io/mod.rs:1660]:0x3ffb61d0 0x400dc329 [std::sys_common::backtrace::_print:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:49]:0x3ffb6220 0x400e66d6 [std::panicking::default_hook::{{closure}}:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:211]:0x3ffb6270 0x400e658f [std::panicking::default_hook:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:228]:0x3ffb62e0 0x400e6884 [std::panicking::rust_panic_with_hook:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:606]:0x3ffb6350 0x400dc82f [std::panicking::begin_panic_handler::{{closure}}:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:??]:0x3ffb63b0 0x400dc489 [std::sys_common::backtrace::__rust_end_short_backtrace:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:139]:0x3ffb63e0 0x400e6768 [rust_begin_unwind:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:498]:0x3ffb6410 0x40105993 [core::panicking::panic_fmt:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:116]:0x3ffb6440 0x40105937 [core::panicking::panic:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:48]:0x3ffb6470 0x400d5075 [rust_esp32_std_demo::main:/home/jan/CLionProjects/rust-esp32-std-demo/src/main.rs:??]:0x3ffb64b0 0x4011a1e7 [core::ops::function::FnOnce::call_once:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ops/function.rs:227]:0x3ffb6560 0x400d5454 [std::rt::lang_start::{{closure}}:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/rt.rs:145]:0x3ffb6580 0x400f52e1 [core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ops/function.rs:259]:0x3ffb65a0 0x400d5442 [std::rt::lang_start:/home/jan/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/rt.rs:144]:0x3ffb65c0 0x400d52c5 [main:??:??]:0x3ffb65f0 0x400d61a7 [app_main:/home/jan/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.5/src/start.rs:46]:0x3ffb6610 0x4011b5c4 [main_task:/home/jan/CLionProjects/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/port_common.c:129]:0x3ffb6630 
ELF file SHA256: 0000000000000000
```

while I would expect a normal full backtrace to be beautifully formatted like this:

```
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', src/main.rs:13:28
stack backtrace:
   0:     0x55af53f8358d - std::backtrace_rs::backtrace::libunwind::trace::h7c06355936cefbfa
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
   1:     0x55af53f8358d - std::backtrace_rs::backtrace::trace_unsynchronized::h0d29350836df727e
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x55af53f8358d - std::sys_common::backtrace::_print_fmt::h7cc76caea8ef80df
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/std/src/sys_common/backtrace.rs:66:5
   3:     0x55af53f8358d - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h41dad21719ef8ab8
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/std/src/sys_common/backtrace.rs:45:22
   4:     0x55af53f9ef6c - core::fmt::write::h6c4cfebe9aad777b
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/core/src/fmt/mod.rs:1190:17
[... cut ...]
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/std/src/rt.rs:128:20
  42:     0x55af53e8c980 - std::rt::lang_start::h48e03617adc8bba4
                               at /rustc/f103b2969b0088953873dc1ac92eb3387c753596/library/std/src/rt.rs:144:17
  43:     0x55af53e89d1c - main
  44:     0x7f90c0eaf310 - __libc_start_call_main
  45:     0x7f90c0eaf3c1 - __libc_start_main@GLIBC_2.2.5
  46:     0x55af53e877a5 - _start
  47:                0x0 - <unknown>

Process finished with exit code 101
```

ivmarkov, 2022-05-10:
Not sure what you are trying to achieve:

1. if you just want **decoded** (as in containing file names, line numbers and function names) backtraces from **panics**, you don't need to enable the `RUST_BACKTRACE` flag at all. You just need a monitor, which is capable of mapping program counters output from the panic backtrace into filename+line number+function name triples. On embedded firmware, this decoding is not available out of the box, because it needs the original ELF binary and its debugging symbols, and these things are not burned onto the chip flash as they take too much space. Instead, you have to use a monitor which is capable of doing this decoding. For example, both `espflash` and `espmonitor` (the latter - with the `--bin` option pointing to the ELF file). Just mind [this bug](https://github.com/esp-rs/espmonitor/issues/39), which is opened since forever and which results in a bit unreadable output from `espmonitor` specifically

2. If you want to capture and print backtraces with normal Rust errors (as in anyhow errors; NOT panics), then and only then you need the discussed flag.

jan-br, 2022-05-10:
Ah thats interesting, so the formatting of the backtrace is handled by the monitor and not by the process itself? Didn't know that.
Doesn't seem very intuitive for me.
But if I understand you correctly, it also means the formatting should work out of the box if debug symbols are available in the binary. And they should be available in my case, at least I am testing with `debug=true` in my profile flags.
I got an esp32 with 16MB flash, so I got some headroom

ivmarkov, 2022-05-10:
> I am actually trying to catch panics and format them like a default rust program without any configuration would do it, not just some results. I think the proper rust term for this process is unwinding.

Forget about `panic-unwind` on embedded. It will never, ever work for esp32 with its limited flash size. Treat panics as hard failures that restart the whole firmware. Embedded uses panic-abort, not panic-unwind. Why do you need panic-unwind on embedded anyway, where restarting the whole firmware takes a second?

> Ah thats interesting, so the formatting of the backtrace is handled by the monitor and not by the process itself? Didn't know that. Doesn't seem very intuitive for me. Or do you mean the monitor can optionally do that, when the process doesn't out of the box? Since I've seen normally printed rust panics in normal terminals without any smart monitor features.

Again: there is just no place for the symbol tables with the debug info in the actual firmware. Even with 16 MB you are pushing it. Just no way. So it is always the monitor which decodes it, as the monitor which runs on your PC has access to the original elf binary that contrains the debug info. What is flashed on  the actual embedded firmware is only a portion of the elf -the one that contains the code and only the code basically. And the firmware outputs a panic backtrace that contains program counters. The monitor then greps them and decodes based on lookups inside the elf which stays on your PC and it has access to.

> But if I understand you correctly, it also means the formatting should work out of the box if debug symbols are available in the binary. And they should be available in my case, at least I am testing with `debug=true` in my profile flags. I got an esp32 with 16MB flash, so I got some headroom

No you do t have a headroom, even with 16 mb. :)

closed 2022-05-10 by ivmarkov