ivmarkov/rust-esp32-std-demo #87: Cannot build the latest demo inside espressif/idf-rust-examples Docker container

callmephilip, 2022-05-10:
Running 

```bash
docker run -it espressif/idf-rust-examples
```

Then inside docker

```bash
cd rust-esp32-std-demo
git pull origin main
export RUST_ESP32_STD_DEMO_WIFI_SSID=<ssid>
export RUST_ESP32_STD_DEMO_WIFI_PASS=<ssid>
cargo +esp build
```

Results in

```
Compiling esp-idf-sys v0.31.5
The following warnings were emitted during compilation:

warning: esp-idf version (5.0.0) not officially supported by `esp-idf-sys`. Supported versions are 'master', 'release/v4.4', 'release/v4.3', 'v4.4(.X)', 'v4.3.2'.

error: failed to run custom build command for `esp-idf-sys v0.31.5`

Caused by:
  process didn't exit successfully: `/opt/rust-esp32-std-demo/target/debug/build/esp-idf-sys-c1bd466bd2ec9680/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  cargo:warning=esp-idf version (5.0.0) not officially supported by `esp-idf-sys`. Supported versions are 'master', 'release/v4.4', 'release/v4.3', 'v4.4(.X)', 'v4.3.2'.
  cargo:rerun-if-changed=/opt/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/opt/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/opt/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/opt/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/opt/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8d0c4c9d291d3560/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/opt/esp/idf/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/opt/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8d0c4c9d291d3560/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  -- Configuring incomplete, errors occurred!
  See also "/opt/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8d0c4c9d291d3560/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Using activated esp-idf v5.0.0 environment at '/opt/esp/idf'
  ERROR: /opt/rust-esp32-std-demo/.embuild/espressif/python_env/idf5.0_py3.8_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
  CMake Error at /opt/esp/idf/tools/cmake/build.cmake:286 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /opt/esp/idf/tools/cmake/build.cmake:415 (__build_check_python)
    CMakeLists.txt:6 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /opt/cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

SergioGasquez, 2022-05-10:
Yes, just tried with our [latest builds](https://hub.docker.com/r/sergiogasquez/esp-rs-env), and it's building fine. Here are the steps:
1. Choose the desired tag, [here is a description of every tag](https://github.com/SergioGasquez/esp-rs-container#esp-rs-container)
   - I would recommend using `esp32`, `esp32c3`, `esp32s2` or `esp32s3` if you are only planning to use one target, and `espidf_v4.4` if you want to build the project for any target.
1. `docker run -it sergiogasquez/esp-rs-env:<tag>`
1. Then inside the docker
   1. `git clone https://github.com/ivmarkov/rust-esp32-std-demo`
   2. `cd rust-esp32-std-demo/`
   3. `source /home/esp/export-rust.sh` 
      > This step wont be required one this action completes: https://github.com/SergioGasquez/esp-rs-container/actions/runs/2301195699.
      I'll update this comment once its finished
   4. `cargo +esp build`

SergioGasquez, 2022-05-10:
The github action completed suscesfully, just tested it again and I can confirm that is working fine!

callmephilip, 2022-05-11:
nice one @SergioGasquez! I can confirm that `sergiogasquez/esp-rs-env:esp32` works!

SergioGasquez, 2022-07-05:
Can we close this issue? Using tags from [espressif/idf-rust](https://hub.docker.com/r/espressif/idf-rust) also work.

closed 2022-07-05 by ivmarkov