ivmarkov/rust-esp32-std-demo #98: VSCode and WSL go crazy when running the example

Squirrelcoding, 2022-06-25:
Ok, this issue is quite weird. I'm not sure if its a problem with my computer or the code, but when I run the example in WSL inside of VSCode, this happens:
- The terminal freezes
- VSCode disconnects from WSL
- My computer fans go crazy
- Probably unrelated, but my WiFi speed slows down a lot whenever this happens
I have no idea what is causing this but it has to be *something*!

## Steps to reproduce (assuming everything is installed)
- Make sure ESP32 has all permisisons
- Flash this repo onto the ESP32
- Run `espmonitor /dev/ttyUSB0`

Everything is fine until `I (13509) rust_esp32_std_demo: About to open a TCP connection to 1.1.1.1 port 80` is printed out, and everything goes downhill...
It seems as if `espmonitor` freezes on [this line of code](https://github.com/ivmarkov/rust-esp32-std-demo/blob/2918d360cf84dd3663dbd46996025e2ed8fb2c06/src/main.rs#L464)! I was wondering two things:

- Can this be reproduced successfully?
- Is there a solution? I can't find one at the moment!

A response would be helpful!

ivmarkov, 2022-06-26:
And what happens when you comment out that line of code?

Squirrelcoding, 2022-06-26:
@ivmarkov if I comment out `test_tcp()`, the next function to freeze is `test_mqtt_client()`, and [it](https://github.com/ivmarkov/rust-esp32-std-demo/blob/2918d360cf84dd3663dbd46996025e2ed8fb2c06/src/main.rs#L646) freezes right before connecting. 
---
I was playing around with the demo as I was writing this, and shut down my PCs hotspot while the serial monitor was still running, and a couple things happened:
- The original issue doesn't happen (yay!)
- The demo's output continues as normal until line 319 is reached
- The loop on [line 319](https://github.com/ivmarkov/rust-esp32-std-demo/blob/2918d360cf84dd3663dbd46996025e2ed8fb2c06/src/main.rs#L319) continues forever, due there not being a connection.

I suspect that the issue is that the logs aren't being outputted for whatever reason.

After more investigating while writing this, I used putty to run the demo, and pretty much

Squirrelcoding, 2022-06-26:
The issue is NOT with the demo, its the fact that running WSL and a hotspot at the same time in Windows is causing issues, which is already a known issue with WSL, so I'm going to go ahead and close this.

closed 2022-06-26 by Squirrelcoding