ivmarkov/rust-esp32-std-demo #99: sdkconfig files ignored

karlri, 2022-06-30:
I tried building on Windows and the stack always overflows after drawing text with feature ttgo. I tried `cargo clean; cargo +esp build --release` to no avail. `CONFIG_ESP_MAIN_TASK_STACK_SIZE=20000` is set in sdkconfig.defaults and I tried duplicating it in sdkconfig.defaults.esp32, again clean and build but still no luck, still overflows, I also tried doubling that value, same result. Eventually I suspected that it's ignored.

I tested if the files actually get picked up by checking the stack size at runtime and as suspected, it's too small:
```
if esp_idf_sys::CONFIG_MAIN_TASK_STACK_SIZE < 20000 {
    error!("stack too small: {} bail!", esp_idf_sys::CONFIG_MAIN_TASK_STACK_SIZE);
    return Ok(());
}
```
Prints the idf-sdk default value:
`E (296) rust_esp32_std_demo: stack too small: 3584 bail!`

The only non-standard conf I have is that I've set the following env var:
`$Env:CARGO_TARGET_DIR="C:/espt/"`
Because It's the only way I could get the paths short enogh to not get CMake build errors about path length. Perhaps that could be the cause. Any ideas?

karlri, 2022-06-30:
I tried copying the sdkconfig files to C:/ and indeed that works, so I suppose they are assumed to be in target/.. I'll leave this open as the problem with path lengths causes this in the first place. Perhaps the problem with path length could be fixed somehow or perhaps it's possible to modify the build process to sdkconfig files are searched for in the project root rathet than target/..

xMAC94x, 2022-09-06:
Hi i can reproduce this bug on linux too, is there any workaround to set the MAIN_TASK_STACK_SIZE on linux ?

ivmarkov, 2022-09-06:
> Hi i can reproduce this bug on linux too, is there any workaround to set the MAIN_TASK_STACK_SIZE on linux ?

Yes, just place your `sdkconfig.defaults` file in the **parent** directory of your `target` directory. Unless you've overridden the target directory with `CARGO_TARGET_DIR` as the original poster did, the parent of your `target` directory is the binary crate directory itself.

Stenodyon, 2022-10-23:
I'd like to add that if your binary is part of a cargo workspace, the `sdkconfig.defaults` file has to be in the root directory of the *workspace* and not of the binary (found that out through trial and error).

ivmarkov, 2023-10-19:
> I'd like to add that if your binary is part of a cargo workspace, the `sdkconfig.defaults` file has to be in the root directory of the _workspace_ and not of the binary (found that out through trial and error).

That's true and that's usually the culprit.

closed 2023-10-19 by ivmarkov

kellerkindt, 2024-01-07:
Just for forgetful people digging up old projects which do not compile due to "ignored sdkconfig.defaults": Don't forget, that in `.cargo/config.toml` one can set funny things, such as:
```
[env]
ESP_IDF_VERSION = { value = "release/v4.4" }
ESP_IDF_TOOLS_INSTALL_DIR = {value = "global"}
ESP_IDF_SDKCONFIG_DEFAULTS = { value = "/path/to/sdkconfig.defaults" }
# or
# ESP_IDF_SDKCONFIG_DEFAULTS = { value = "sdkconfig.defaults", relative = true }

```