ivmarkov/rust-esp32-std-demo #100: HTTP server issues

flyaruu, 2022-07-02:
Hi, got the demo working without trouble, but the httpd server seems flakey, I seem to have two issues:

- I can call from Firefox, but not from curl as it returns ('curl: (1) Unsupported HTTP version in response'), I've tried HTTP/1.0 and HTTP/1.1, but it does not seem to help.
- The httpd server seems to die after a few (10-20) requests. If I start the server, open the http://<ip> url in Firefox, I get the 'Hello from Rust!' message, but after a 10-20 requests I get a 'The connection was reset' message. The logs say:
```
> W (280507) httpd: httpd_accept_conn: error in accept (23)
> W (280507) httpd: httpd_server: error accepting new connection
```
It only recovers after a reboot.

I see a seemingly related post: https://www.esp32.com/viewtopic.php?t=10179 (but not in Rust)

I've followed the suggested solutions (purging LRU, reducing # of sessions and sockets, to no avail:
```
    let mut conf: esp_idf_svc::http::server::Configuration = Default::default();
    conf.lru_purge_enable = true;
    conf.max_open_sockets = 1;
    conf.max_sessions = 1;
    let mut server = esp_idf_svc::http::server::EspHttpServer::new(&conf)?;
```
 
Any ideas?

ivmarkov, 2022-07-19:
I'll close this as another user has identified and fixed the issue in `esp-idf-svc` main/master branch.
Not released yet though.

The user identified another issue unfortunately, related to how response headers are processed. This one has to wait for the upcoming HTTPD server rework, which will significantly change the shape of the traits.

closed 2022-07-19 by ivmarkov

flyaruu, 2022-07-19:
Great news, thanks for following up, I'll make sure to follow that repo as well.

flyaruu, 2022-08-07:
I see that the 'running out of sockets' problem has been fixed, but I can not connect from curl or Postman at all. Firefox seems to work. I'll create a new issue for that.