ivmarkov/rust-esp32-std-demo #101: Build fails for `esp32s2`

SergioGasquez, 2022-07-05:
Demo is failing for `esp32s2`. `cargo +esp build --target xtensa-esp32s2-espidf --release` results in:
```bash
error[E0425]: cannot find function, tuple struct or tuple variant `ToStd` in this scope
    --> src/main.rs:1245:13
     |
1245 |             ToStd(req.into_reader()?).read_to_end(&mut body)?;
     |             ^^^^^ not found in this scope

error[E0599]: no method named `into_reader` found for struct `esp_idf_svc::http::server::EspHttpRequest` in the current scope
    --> src/main.rs:1245:23
     |
1245 |             ToStd(req.into_reader()?).read_to_end(&mut body)?;
     |                       ^^^^^^^^^^^ method not found in `esp_idf_svc::http::server::EspHttpRequest<'_>`

Some errors have detailed explanations: E0425, E0599.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `rust-esp32-std-demo` due to 2 previous errors
```

darconeous, 2022-08-02:
I'm seeing the same thing.

ivmarkov, 2022-08-03:
@darconeous Can you try again, by pulling latest from github?

SergioGasquez, 2022-08-04:
I just did some tests with updated master branch (including https://github.com/ivmarkov/rust-esp32-std-demo/commit/3656007036519049127594d74e3ffc231a21aafc) and the issue still persists

ivmarkov, 2022-08-04:
Now fixed for real!: https://github.com/ivmarkov/rust-esp32-std-demo/issues/101

closed 2022-08-04 by ivmarkov

SergioGasquez, 2022-08-04:
I can confirm that it's now working! Thanks @ivmarkov