ivmarkov/rust-esp32-std-demo #110: HTTP headers seem wrong

flyaruu, 2022-08-07:
I'm running the demo on a c3 chip with no modifications.
Connecting to the controller over http seems to fail when using curl and Postman, however it does work from Firefox.

E.g.:
```
curl -v http://10.11.12.233       
*   Trying 10.11.12.233:80...
* Connected to 10.11.12.233 (10.11.12.233) port 80 (#0)
> GET / HTTP/1.1
> Host: 10.11.12.233
> User-Agent: curl/7.79.1
> Accept: */*
> 
* Unsupported HTTP version in response
* Closing connection 0
```
Postman fails with similar error:
```
GET http://10.11.12.233/
Error: Parse Error: Invalid response status
Request Headers
User-Agent: PostmanRuntime/7.29.2
Accept: */*
Host: 10.11.12.233
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
```
If I connect with telnet I see something like this:
```
$>telnet 10.11.12.233 80
Trying 10.11.12.233...
Connected to 10.11.12.233.
Escape character is '^]'.
GET / HTTP/1.1

HTTP/1.1 T???????M
Content-Type: text/html
Transfer-Encoding: chunked

10
Hello from Rust!
0

```
As you can see it does return a body, but the return header seems weird. See my previous issue [here](https://github.com/ivmarkov/rust-esp32-std-demo/issues/100)

palmtoy, 2022-08-08:
Branch: main;  Commit code: 354e0b324b1ebbf6ccf2b9f48e18c7e6fed20067

Same issue with ESP32C3. It does work from Chrome ( version: 104.* ).
Because of the weird response header even the Chrome can't get the correct status code, such as 500 or 403.  Every response status code is 200.
For example, when the server prints "Boo, something happened!", the Chrome still gets code 200, it should be 500 instead.

There is way to walk around, go back to "old" HTTP server creating method ↓
#[cfg(feature = "experimental")]
fn httpd(mutex: Arc<(Mutex<Option<u32>>, Condvar)>) -> Result<idf::Server> {
PS: remember to delete the word "not" after cfg in { #[cfg(not(feature = "experimental"))] }

flyaruu, 2022-08-12:
upgrading esp-idf-svc to 0.42.3 works! Headers seem good, curl works as well.
Thanks! This issue can be closed.

closed 2022-08-12 by ivmarkov