ivmarkov/rust-esp32-std-demo #112: Joining the threads crashes

Venoox, 2022-09-01:
It flashes successfully and starts but then something goes wrong, what could be the cause?

```
Hello from Rust!
More complex print ["foo", "bar"]
Result: 0, 1
Rust main thread: Thread { id: ThreadId(1), name: Some("main"), .. }
Main TLS before change: 13
Main TLS after change: 42
This is thread number 1, Thread { id: ThreadId(3), name: None, .. }
This is thread number 0, Thread { id: ThreadId(2), name: None, .. }
Inner TLS: 0
Inner TLS: 1
This is thread number 2, Thread { id: ThreadId(4), name: None, .. }
This is thread number 3, Thread { id: ThreadId(5), name: None, .. }
Inner TLS: 3
This is thread number 4, Thread { id: ThreadId(6), name: None, .. }
Inner TLS: 4
Inner TLS: 2
About to join the threads. If ESP-IDF was patched successfully, joining will NOT crash
Main TLS after threads: 42
E (10829) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10829) task_wdt:  - IDLE (CPU 0)
E (10829) task_wdt: Tasks currently running:
E (10829) task_wdt: CPU 0: main
E (10829) task_wdt: CPU 1: IDLE
E (10829) task_wdt: Print CPU 0 (current core) backtrace
Backtrace: 0x4016FCC6:0x3FFB0FD0 0x40082CE1:0x3FFB0FF0 0x4008A3C5:0x3FFBE270 0x40167913:0x3FFBE290 0x4013446D:0x3FFBE2B0 0x40131A14:0x3FFBE2F0 0x400D4602:0x3FFBE310 0x401C8E63:0x3FFBE790 0x400D7D74:0x3FFBE7B0 0x4011CBD1:0x3FFBE7D0 0x400D6639:0x3FFBE7F0 0x400E83AF:0x3FFBE820 0x401CFAAB:0x3FFBE840
E (10829) task_wdt: Print CPU 1 backtrace
Backtrace: 0x40084C19:0x3FFB15D0 0x40082CE1:0x3FFB15F0 0x4000BFED:0x3FFBF660 0x4008B0A6:0x3FFBF670 0x4016FF2F:0x3FFBF690 0x4016FF3B:0x3FFBF6C0 0x40148FDA:0x3FFBF6E0 0x400892D8:0x3FFBF700

```

PTD110, 2022-09-05:
I am facing the same issue.
It's the `thread::sleep(Duration::from_secs(2));` on line 438 that is the culprit.
It has been discussed [here in esp-rs project](https://github.com/esp-rs/rust/issues/137)

ivmarkov, 2023-10-19:
This had been resolved in the meantime.

closed 2023-10-19 by ivmarkov