ivmarkov/rust-esp32-std-demo #116: partitions.csv not being picked up.

axos88, 2022-09-16:
sdkconfig.defaults is missing:

```
CONFIG_PARTITION_TABLE_SINGLE_APP=
CONFIG_PARTITION_TABLE_CUSTOM=y
```

But even then the build is looking for the partitions.csv in the wrong place

axos88, 2022-09-16:
I needed to add the following as well:

```
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="../../../../../../partitions.csv"
```

pyaillet, 2022-09-22:
`espflash` documentation [here](https://github.com/esp-rs/espflash/blob/master/cargo-espflash/README.md#package-metadata) states that you can either pass this as a parameter to the command or add a configuration in `Cargo.toml`

axos88, 2022-09-24:
The problem is that the example project has this file in the root directory, but it's not actually used, which is misleading. Either it would need to be added to the README.md, or these config options need to be added, so a newcomer is not confused by their edits to the partition table file not being picked up.

pyaillet, 2022-09-24:
Yes, that's why I proposed a PR #117 adding the configuration to `Cargo.toml` :)

axos88, 2022-09-25:
Sorry, I did not see that!