ivmarkov/rust-esp32-std-demo #122: mac m1 can not build in docker, cause by embedded-io-0.30

xiaoheng14, 2022-10-23:
root@c3c726dd90c0:/app/rust-esp32-std-demo# cargo update
    Updating git repository `https://github.com/esp-rs-compat/getrandom.git`
    Updating git repository `https://github.com/esp-rs-compat/polling`
    Updating git repository `https://github.com/esp-rs-compat/smol`
    Updating git repository `https://github.com/esp-rs-compat/socket2`
    Updating crates.io index
    Updating git repository `https://github.com/yuri91/ili9341-rs`
root@c3c726dd90c0:/app/rust-esp32-std-demo# cargo build --release
   Compiling compiler_builtins v0.1.82
   Compiling core v0.0.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.131
   Compiling cc v1.0.73
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/std)
   Compiling proc-macro2 v1.0.47
   Compiling quote v1.0.21
   Compiling unwind v0.0.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/unwind)
   Compiling unicode-ident v1.0.5
   Compiling syn v1.0.103
   Compiling libc v0.2.135
   Compiling autocfg v1.1.0
   Compiling serde v1.0.147
   Compiling serde_derive v1.0.147
   Compiling log v0.4.17
   Compiling cfg-if v1.0.0
   Compiling rustversion v1.0.9
   Compiling serde_json v1.0.87
   Compiling fnv v1.0.7
   Compiling backtrace v0.3.66
   Compiling thiserror v1.0.37
   Compiling anyhow v1.0.66
   Compiling once_cell v1.15.0
   Compiling semver v1.0.14
   Compiling aho-corasick v0.7.19
   Compiling regex-syntax v0.6.27
   Compiling futures-core v0.3.25
   Compiling gimli v0.26.2
   Compiling thiserror-impl v1.0.37
   Compiling adler v1.0.2
   Compiling bitflags v1.3.2
   Compiling miniz_oxide v0.5.4
   Compiling addr2line v0.17.0
   Compiling regex v1.6.0
   Compiling object v0.29.0
   Compiling rustc-std-workspace-core v1.99.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/alloc)
   Compiling cfg-if v0.1.10
   Compiling adler v0.2.3
   Compiling rustc-demangle v0.1.21
   Compiling rustc-std-workspace-alloc v1.99.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-alloc)
   Compiling gimli v0.25.0
   Compiling hashbrown v0.12.3
   Compiling std_detect v0.1.5 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/stdarch/crates/std_detect)
   Compiling miniz_oxide v0.4.0
   Compiling object v0.26.2
   Compiling panic_abort v0.0.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/panic_abort)
   Compiling panic_unwind v0.0.0 (/root/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust/library/panic_unwind)
   Compiling addr2line v0.16.0
   Compiling ryu v1.0.11
   Compiling glob v0.3.0
   Compiling heck v0.4.0
   Compiling itoa v1.0.4
   Compiling clang-sys v1.4.0
   Compiling strum_macros v0.24.3
   Compiling dirs-sys v0.3.7
   Compiling indexmap v1.9.1
   Compiling crossbeam-utils v0.8.12
   Compiling shlex v1.1.0
   Compiling dirs v4.0.0
   Compiling bstr v0.2.17
[... 295 lines cut ...]
error[E0310]: the parameter type `T` may not live long enough
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.0/src/asynch.rs:267:28
    |
267 |     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>>
    |                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ ...so that the type `T` will meet its required lifetime bounds
    |
help: consider adding an explicit lifetime bound...
    |
257 | impl<T: ?Sized + Write + 'static> Write for alloc::boxed::Box<T> {
    |                        +++++++++

error: non-defining opaque type use in defining scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.0/src/asynch.rs:286:9
    |
286 | /         async move {
287 | |             self.extend_from_slice(buf);
288 | |             Ok(buf.len())
289 | |         }
    | |_________^ lifetime `'a` is part of concrete type but not used in parameter list of the `impl Trait` type alias

Some errors have detailed explanations: E0310, E0491.
For more information about an error, try `rustc --explain E0310`.
error: could not compile `embedded-io` due to 21 previous errors
warning: build failed, waiting for other jobs to finish...
root@c3c726dd90c0:/app/rust-esp32-std-demo#

xiaoheng14, 2022-10-23:
esp32c3,and target is riscv32imc-esp-espidf

xiaoheng14, 2022-10-24:
> having the same issue, target is `riscv32imc-esp-espidf`

me too ... not only in my mac, windows use wsl also has this problem, A few days ago we both can build the project, but today occur this problem and can't find the way to resolve it; have you solved?

nascent-chris, 2022-10-24:
I have not yet, no.  When I have time I will try to investigate further

ivmarkov, 2022-10-24:
Switching to older Rust Nightly should fix the issue.

xiaoheng14, 2022-10-24:
> I have not yet, no. When I have time I will try to investigate further

> Switching to older Rust Nightly should fix the issue.

ok ,i will try it later, thanks!

nascent-chris, 2022-10-24:
> > I have not yet, no. When I have time I will try to investigate further
> 
> thanks for discussing!

没问题 :slightly_smiling_face:  -- I just tried an older version of nightly (2022-10-01) and it builds.  Thank you @ivmarkov for the suggestion

so you have it handy, here's the command I used to switch to that specific version of nightly

```bash
rustup override set nightly-2022-10-01
```

xiaoheng14, 2022-10-24:
> > > I have not yet, no. When I have time I will try to investigate further
> > 
> > 
> > thanks for discussing!
> 
> 没问题 🙂 -- I just tried an older version of nightly (2022-10-01) and it builds. Thank you @ivmarkov for the suggestion
> 
> so you have it handy, here's the command I used to switch to that specific version of nightly
> 
> ```shell
> rustup override set nightly-2022-10-01
> ```

666, it works!!!

xiaoheng14, 2022-10-24:
> > I have not yet, no. When I have time I will try to investigate further
> 
> > Switching to older Rust Nightly should fix the issue.
> 
> ok ,i will try it later, thanks!

thanks for your help, it works!!!

closed 2022-10-24 by xiaoheng14