ivmarkov/rust-esp32-std-demo #127: Not a bug. Just more like rust questions.

mangelajo, 2022-11-09:
First, I know this is not the right place to drop this type of question.  I am wondering if there is a chat or mail list I could join where other people are also working with rust on ESP32 platforms.

I have been playing around with the esp-rs crates, and your demo, and struggling, mostly because I'm still in the process of learning Rust (lots of Asm/C/C++/golang/python experience, some ADA of which Rust reminds me of a bit)

My question:

As part of a project I am doing just for fun, I am hitting some issues encapsulating a reference to my display:

https://github.com/mangelajo/mobil-monitor/blob/main/src/peripherals/display.rs#L23

That so far works, but it's a bit awful :-), I wonder if there's a better way to encapsulate the reference to my display from "Display", probably should be called UI instead. 

Things go south when I try to encapsulate the SPI Interface under a shared_bus (my board has a resistive touchscreen controller under the same SPI bus). Because shared_bus holds some private structures inside, and then I cannot describe the type in full, so I thought I could keep a boxed referenced to the traits, which got me here:

```
pub struct Display2 {
    display: Box<dyn DrawTarget<Color=dyn From<Rgb565>, Error=Error>>,
}
```

```
39  |     display: Box<dyn DrawTarget<Color=dyn From<Rgb565>, Error=Error>>,
    |                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `embedded_graphics::draw_target::DrawTarget` cannot be made into an object
    |
```

Best regards, 
Miguel Ángel.

mangelajo, 2022-11-10:
I had some suggestions in https://app.element.io/#/room/#esp-rs:matrix.org to try esp-hal-idf master and Peripheral/PinDriver, and incoming support for multiple SPI devices on a bus.

ivmarkov, 2023-10-19:
Here's the chat: https://matrix.to/#/#esp-rs:matrix.org

closed 2023-10-19 by ivmarkov